crankyoldgit/IRremoteESP8266 #191: SONY .sendraw

kiralikbeyin, 2017-05-02:
@crankyoldgit 
I cant send raw data for sony maybe related with #118 ?

```
unsigned int  rawData[31] = {2500,550, 1300,550, 1300,500, 700,550, 750,550, 1300,550, 700,500, 750,500, 750,500, 700,550, 750,500, 750,550, 1300,500, 1300,550, 700,500, 700};  //  

tried 
     irsend.sendRaw(tempbuf, templen, 40);
      irsend.sendRaw(tempbuf, templen, 40);
      irsend.sendRaw(tempbuf, templen, 40);

and tried single
      irsend.sendRaw(tempbuf, templen, 38);
```
**sendraw for Samsung and nec works well.**

more data:
```
Encoding  : SONY

Code      :  (15 bits)
Timing[31]: 
     +2500, - 550     +1300, - 550     +1300, - 500     + 700, - 550
     + 750, - 550     +1300, - 550     + 700, - 500     + 750, - 500
     + 750, - 500     + 700, - 550     + 750, - 500     + 750, - 550
     +1300, - 500     +1300, - 550     + 700, - 500     + 700
unsigned int  rawData[31] = {2500,550, 1300,550, 1300,500, 700,550, 750,550, 1300,550, 700,500, 750,500, 750,500, 700,550, 750,500, 750,550, 1300,500, 1300,550, 700,500, 700};  //  
unsigned int  data = 0x640C;
*******************
Serial print of tempbuf:
2500, 550, 1300, 550, 1300, 500, 700, 550, 750, 550, 1300, 550, 700, 500, 750, 500, 750, 500, 700, 550, 750, 500, 750, 550, 1300, 500, 1300, 550, 700, 500, 700, 
Length : 31
********************
```

crankyoldgit, 2017-05-03:
Hey, still having issues with Sony eh?
One of the differences with `sendSony(0x640C, 15, 0);` and your `irsend.sendRaw(tempbuf, templen, 40);` is that the sendSony() routine handles the inter-message gap, where sendRaw() typically doesn't.
Since Sony requires three repeats to sense a command, the lack of a gap is probably what is causing you issue.

You can simulate sendSony()'s behaviour in a way by adding a delay() after each invocation of the equiv. sendRaw() call, or add an extra value to your tempbuf array. They would both emulate a 'space()' for the required time to perform the inter-message gap.

e.g.
```
     // gap via delay()
     irsend.sendRaw(tempbuf, templen, 40);
     delay(45);
     irsend.sendRaw(tempbuf, templen, 40);
     delay(45);
     irsend.sendRaw(tempbuf, templen, 40);
     delay(45);
```
or
unsigned int  tempbuf[**32**] = {2500,550, 1300,550, 1300,500, 700,550, 750,550, 1300,550, 700,500, 750,500, 750,500, 700,550, 750,500, 750,550, 1300,500, 1300,550, 700,500, 700,**45000**};  //
unsigned int templen = **32**;
irsend.sendRaw(tempbuf, templen, 40);
irsend.sendRaw(tempbuf, templen, 40);
irsend.sendRaw(tempbuf, templen, 40);

But, I really recommend using the supported encoder `sendSony(0x640C, 15, 2);` which does it all for you.

crankyoldgit, 2017-05-03:
I should add sendSony() calculates the correct gap, the delay(45) & extra 45000 in the raw data (a.k.a. space(45000)) are just approximations.

kiralikbeyin, 2017-05-03:
Ok i understand now thanks. I will not use sendraw for Sony.
Is there any more protocol which makes problem with sendraw function?

crankyoldgit, 2017-05-03:
It's not that there is a problem with sendRaw(), it's that you need to know what is going on in _some_ cases. sendRaw() will do pretty much anything, but it's not smart. You need to be the smarts for it. :-)

Any protocol with a _repeat_ mechanism is likely to cause you an issue if you use sendRaw() and **don't** add some inter-message gaps. etc.

If you look at the code, you'll find a number of sendBlah() procedures deal with repeats. e.g. JVC, Sony, NEC, Sherwood, ...

crankyoldgit, 2017-05-03:
Marking this issue closed for now. Feel free to re-open if you don't think it should be.

closed 2017-05-03 by crankyoldgit

kiralikbeyin, 2017-05-08:
note:  code block must be in the loop, if not it is not working.

```
  if (irgir)
  {
    irgir = false;

    irsend.sendRaw(tempbuf, templen, 40);
    irsend.sendRaw(tempbuf, templen, 40);
    irsend.sendRaw(tempbuf, templen, 40);
  }
```

crankyoldgit, 2017-05-08:
Can you explain? Because that makes no sense to me.

kiralikbeyin, 2017-05-09:
#198