crankyoldgit/IRremoteESP8266 #198: sendsony again

kiralikbeyin, 2017-05-06:
@crankyoldgit #118 

I tried:        irsend.sendSony(irlong , irbit.toInt(), 2);

Latest release didnt send VOL+ irdata:0x240c
BUT VOL- irdata:0x640c works fine

I also tried
````
      
      irsend.sendSony(irlong , irbit.toInt());
      irsend.sendSony(irlong , irbit.toInt());
      delay(45);
      irsend.sendSony(irlong , irbit.toInt());
      delay(45);
      irsend.sendSony(irlong , irbit.toInt());
      irsend.sendSony(irlong , irbit.toInt());

      delay(300);
 
      irsend.sendSony(irlong , irbit.toInt());
      delay(45);
      irsend.sendSony(irlong , irbit.toInt());
      delay(45);
      irsend.sendSony(irlong , irbit.toInt());
      irsend.sendSony(irlong , irbit.toInt());
````
First i press my test button twice and VOL+ works, if i press again in 2-3 seconds VOL+ works again...
There is no problem with VOL-
VOL-(25612) bigger value than VOL+(9228) http://www.binaryhexconverter.com/hex-to-decimal-converter

I also tried some combinations of these +50 -50 
#define SONY_ONE_MARK	1300  // 1250 Experiments suggest +50 to spec is better.
#define SONY_ZERO_MARK	650 // 650 Experiments suggest +50 to spec is better.

crankyoldgit, 2017-05-07:
I've re-read #118 and this, and I'm at a loss as to why it's not working. If it sends 0x640c fine, it should send 0x240c fine too.

As I don't have the Sony equipment (or your exact hardware) I'm going to have to get you to try a number of things and collect lots and lots of data to try to figure out what could be wrong and how I can fix it/help you.

Unfortunately, that means we need to confirm & check all the basics (like hardware/circuit etc) and try to isolate if its a hardware (sony device) issue, or software etc. I case we've missed something along the way.

Can I get you to follow [these steps](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) please? Particularly the advanced, and reporting an issue steps. It can give you some ideas for things to try/confirm/check and collect data for.

Also, can you have your program print out irlong, and irbit.toInt() too, so we can confirm those values etc.

Let's stick with a _standard_ call of `irsend.sendSony(0x240CUL, 15, 2);` when dumping / collecting data etc as that's what's causing you grief and **should** work. I am assuming it's the Sony 15 bit variant you are/need to send? or is it the 12 or 20 bit one?

In short, we need to capture (i.e raw dumps etc) what is different between you pressing the remote button, and the ESP sending the code. (it could also be a physical signal strength issue. i.e. circuit power etc, but less likely as you already have it working for other signals)

Also, bonus marks etc, if you want to try out/test the [v2.0-dev branch](https://github.com/markszabo/IRremoteESP8266/tree/v2.0-dev). It has some signal timing improvements that could also be the root of your problem. Just don't expect v2.0 to work etc. It's still under active development and only partially tested etc.

kiralikbeyin, 2017-05-07:
OK i focused on with definitions 
now short code VOL+(9228)  and long code VOL-(25612) works fine with these:

#define SONY_ONE_MARK	1300  // 1250 Experiments suggest +50 to spec is better.
#define SONY_ZERO_MARK	550 // 650 Experiments suggest +50 to spec is better.

single code : irsend.sendSony(irlong , irbit.toInt(), 2);

If i change SONY_ONE_MARK to 1200 or 1250 long code didn't work
If i change SONY_ZERO_MARK to 600 or 650 short code didn't work

All defs:
#define SONY_HDR_MARK	2400
#define SONY_HDR_SPACE	600
#define SONY_ONE_MARK	1300  // 1250 Experiments suggest +50 to spec is better.
#define SONY_ZERO_MARK	550 // 650 Experiments suggest +50 to spec is better.
#define SONY_RPT_LENGTH 45000
#define SONY_DOUBLE_SPACE_USECS  500 

Note that you have double def:
#define RAWBUF 100 // Length of raw duration buffer

Is there any special reason using hex value for unsigned long data in 
void sendBlah(unsigned long data, int nbits);

crankyoldgit, 2017-05-07:
Thanks for spotting/reporting the duplicate RAWBUF define.

The spec seems to say a one mark should be 1200, and a zero mark should be 600. (http://www.sbprojects.com/knowledge/ir/sirc.php)

Can you clarify what you mean by working? e.g. is it the ESP sending to the Sony device?

What version of the library are you using? when it works?
Can you try it on the v2.0-dev tree? and can you collect/capture what the ESP device is sending and with what settings you used to send etc?

I'd surprised and starting to wonder if the device or the remote is slightly out of spec. I'd really like confirmation from other sony users before we change those parameters further.

crankyoldgit, 2017-05-07:
Oh, as for hex vs. decimal. No, there is no difference. Hex is just shorter to write and formats better on a screen, and easier to convert visually to bit sizes by humans.

kiralikbeyin, 2017-05-07:
OK i checked https://github.com/z3t0/Arduino-IRremote/blob/f0aa48a4d134999f94b636dc0fa9be45cc093fbe/ir_Sony.cpp Everything is looking normal with Arduino lib and IRremoteESP8266 lib
But atmel and esp chips are not the same , working freq and voltage etc...

I think i made a mistake, i am using 160Mhz CPU freq
(work esp sending to Sony device)
#define SONY_ONE_MARK	1300  //1300 ok 1250 Experiments suggest +50 to spec is better.
#define SONY_ZERO_MARK	**550** //550 ok 650 Experiments suggest +50 to spec is better.
(160CPU works fine)

This config works fine with 80CPU :
#define SONY_ONE_MARK	1300  //1300 ok 1250 Experiments suggest +50 to spec is better.
#define SONY_ZERO_MARK	**650** //550 ok 650 Experiments suggest +50 to spec is better.

Everything is deeply related with timing so my upload prop CPU freq is wrong (people use generally 80Mhz)

I am using 1.2 latest release didn't try v2.0 dev but i will give info when i try.

crankyoldgit, 2017-05-07:
Wow that's interesting. I wouldn't have thought of that. It really does stress the 'use the examples' when debugging point.
Interestingly, in v2.0-dev I just added something to handle different cpu freqs as an issue came up due to cpu instruction time and delays.
I urge you to try the v2.0 branch, and use the IRsend.calibrate() function (once) before you send. i.e. after IRsend.begin() when running at 160MHz.
It will try to auto detect some subtle timing disparities and calculate some internal timing offsets to produce far better signals.
It defaults without calibration, to values for a 80Mhz CPU. All of that is new in v2.0.

kiralikbeyin, 2017-05-07:
```
Encoding  : SONY
Code      : 240C (15 bits)
Timing[31]: 
     +2500, - 550     + 700, - 550     +1300, - 550     + 700, - 500
     + 750, - 500     +1300, - 550     + 700, - 500     + 700, - 500
     + 750, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1300, - 500     +1350, - 550     + 700, - 550     + 700
uint16_t  rawData[31] = {2500,550, 700,550, 1300,550, 700,500, 750,500, 1300,550, 700,500, 700,500, 750,550, 700,550, 700,550, 700,550, 1300,500, 1350,550, 700,550, 700};  // SONY 240C
uint32_t  address = 0x120630;
uint32_t  command = 0x12;
uint64_t  data = 0x240C;

Encoding  : SONY
Code      : 240C (15 bits)
Timing[31]: 
     +2500, - 550     + 700, - 500     +1300, - 500     + 750, - 550
     + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 500     + 750, - 500     + 750, - 550
     +1300, - 550     +1300, - 550     + 700, - 550     + 700
uint16_t  rawData[31] = {2500,550, 700,500, 1300,500, 750,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 700,500, 750,500, 750,550, 1300,550, 1300,550, 700,550, 700};  // SONY 240C
uint32_t  address = 0x120630;
uint32_t  command = 0x12;
uint64_t  data = 0x240C;

Encoding  : SONY
Code      : 240C (15 bits)
Timing[31]: 
     +2500, - 500     + 700, - 550     +1300, - 550     + 700, - 550
     + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 500     + 750, - 550     + 700, - 550
     +1300, - 550     +1300, - 550     + 700, - 550     + 700
uint16_t  rawData[31] = {2500,500, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 700,500, 750,550, 700,550, 1300,550, 1300,550, 700,550, 700};  // SONY 240C
uint32_t  address = 0x120630;
uint32_t  command = 0x12;
uint64_t  data = 0x240C;

Encoding  : SONY
Code      : 240C (15 bits)
Timing[63]: 
     +2500, - 550     + 700, - 550     +1300, - 550     + 700, - 550
     + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 500     + 750, - 550     + 700, - 550
     +1300, - 550     +1300, - 550     + 700, - 550     + 700, -23150
     +2500, - 550     + 700, - 550     +1300, - 550     + 700, - 550
     + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1300, - 550     +1300, - 550     + 700, - 550     + 700
uint16_t  rawData[63] = {2500,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 700,500, 750,550, 700,550, 1300,550, 1300,550, 700,550, 700,23150, 2500,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 700,550, 700,550, 700,550, 1300,550, 1300,550, 700,550, 700};  // SONY 240C
uint32_t  address = 0x120630;
uint32_t  command = 0x12;
uint64_t  data = 0x240C;

Encoding  : SONY
Code      : 240C (15 bits)
Timing[63]: 
     +2500, - 550     + 700, - 550     +1300, - 550     + 700, - 550
     + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1300, - 550     +1300, - 550     + 700, - 550     + 700, -23150
     +2500, - 550     + 700, - 550     +1300, - 550     + 700, - 550
     + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1300, - 550     +1300, - 550     + 700, - 550     + 700
uint16_t  rawData[63] = {2500,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 700,550, 700,550, 700,550, 1300,550, 1300,550, 700,550, 700,23150, 2500,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 700,550, 700,550, 700,550, 1300,550, 1300,550, 700,550, 700};  // SONY 240C
uint32_t  address = 0x120630;
uint32_t  command = 0x12;
uint64_t  data = 0x240C;

Encoding  : SONY
[... 167 lines cut ...]
  irsend.calibrate() ;
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
  irrecv.enableIRIn();  // Start the receiver
}

void loop() {
  // Check if the IR code has been received.
  //  if (irrecv.decode(&results, &save)) {
  //    dumpInfo(&results);           // Output the results
  //    dumpRaw(&results);            // Output the results in RAW format
  //    dumpCode(&results);           // Output the results as source code
  //    Serial.println("");           // Blank line between entries
  //  }

  Serial.println("Sony");
  irsend.sendSony(0x240c, 15, 2);
  delay(1000);
  irsend.sendSony(0x640c, 15, 2);
  delay(1000);

}
```

dev 2.0 160 cpu and 80cpu, didn't changed any defs... all works fine
i added and removed   irsend.calibrate() ;  and irsend.calibrate(40) ;  nothing changed, sonysend worked fine.

rahuljawale, 2017-05-07:
Could be related. I have Sony TV and Sony DVD home theater system. I
rebuilt my IRController code using 1.2 and my systems have problem
identifying IR code. There are no hardware changes and i am using 3x Vishay
TSAL6200 IR LEDs. I will try to downgrade and try again.

crankyoldgit, 2017-05-07:
I'm wondering if when I increased the times in the previous issue #118 that was due to running at 160MHz. @kiralikbeyin Would have that have been the case?

@kiralikbeyin Just to confirm, it all seems to be 'working' i.e. controlling the device correctly, under v2.0?

Interestingly, I notice the `address =` values are horribly wrong. That's something I haven't tested. I'll take a look at that.

crankyoldgit, 2017-05-07:
Oh, the calibrate() probably won't change things you'll notice using the library's tools. You'll really need to set up an oscilloscope  or similar to notice the timing differences. But, they do make it closer to the spec, and maybe closer to what the _real_ device need/require etc.

kiralikbeyin, 2017-05-07:
@crankyoldgit v2.0 works perfect no modification.
It was 160Mhz in #118 but it doesn't (1250-650) work with 1.2 latest release

I re tested 1.2 release 
This config works fine with 80CPU :
#define SONY_ONE_MARK	1300  
#define SONY_ZERO_MARK	**650** 

This config works fine with 160CPU :
#define SONY_ONE_MARK	1300  
#define SONY_ZERO_MARK	**550** 

@rahuljawale I'm using 2 x TSAL6200 as described here : https://github.com/markszabo/IRremoteESP8266/wiki
Receiver is :
[image]

crankyoldgit, 2017-05-08:
Nice. Thanks for the feedback and data. I'm of two minds to if we do another minor release prior to v2.0 to try to address that, or just wait till probably the end of the month when v2.0 will go out for (pre-)release.
I'm tempted to make a minor release for the 80Mhz value, but leave 160MHz value alone for now.

@kiralikbeyin Can I please get you to try 1200 & 600 for the appropriate values under v2.0 and see if your device still accepts those commands? As they are the 'specification' values, I'd like to return to those values under v2.0 **IF** they work under v2.0.

crankyoldgit, 2017-05-08:
Also, just for the record, I just compared the v1.2 & v2.0 code for sendSony(), they are effectively the same, except for v2.0 uses a 33% duty cycle where as v1.2 uses a 50% duty cycle. There are of course other factors in v2.0 which make it better for more accurate timing and signal production.
So, this clearly isn't a problem with the logic of sendSony().

kiralikbeyin, 2017-05-08:
```
#define SONY_HDR_MARK          2400U
#define SONY_SPACE              600U
#define SONY_ONE_MARK    1200U  // Experiments suggest +50 is better.
#define SONY_ZERO_MARK    600U  // Experiments suggest +50 is better.
#define SONY_RPT_LENGTH       45000U
#define SONY_MIN_GAP          10000U
```
this config 1200-600 worked fine with v2.0

i wanted to use sendRaw under v2.0 but i get error:
```
      irsend.sendRaw(tempbuf, templen, 40);
```
no matching function for call to 'IRsend::sendRaw(unsigned int [32], unsigned int&, int)'

kiralikbeyin, 2017-05-08:
I found the main problem!

I was always sending out of the loop (function goes to another function...)
Absolute solution is sending it from the main loop 

1200 and 600 works perfectly with v1.2

Note that v2.0 worked fine with 1200-600 and 1250-650 configs because it was sending from the loop.

@crankyoldgit I was wrong because nec / samsung / lg protocols were working out of the loop but sony made problem out of the loop...

Example sending with v1.2 1200-600

```
loop{
 if (irgir)
  {
    irgir = false;

//    irsend.sendRaw(tempbuf, templen, 40);
//    irsend.sendRaw(tempbuf, templen, 40);
//    irsend.sendRaw(tempbuf, templen, 40);

    if (irpro == "NEC") {
      irsend.sendNEC(irlong , irbit.toInt());
      //if (yaz) Serial.println("NEC GÖNDERDİM");
    }
    else if (irpro == "SONY") {
      irsend.sendSony(irlong , irbit.toInt(), 2);
    //  irsend.sendSony(irlong , irbit.toInt(), 2);

      Serial.println(irlong);
    }
    else if (irpro == "SAMSUNG") {
      irsend.sendSAMSUNG(irlong , irbit.toInt());
      //if (yaz) Serial.println("SAMSUNG GÖNDERDİM");
    }
    else if (irpro == "LG") {
      irsend.sendLG(irlong , irbit.toInt());
    }
    else if (irpro == "RC5") {
      irsend.sendRC5(irlong , irbit.toInt());
    }
    else if (irpro == "RC6") {
      irsend.sendRC6(irlong , irbit.toInt());
    }

    else if (irpro == "WHYNTER") {
      irsend.sendWhynter(irlong, irbit.toInt());
    }
    else if (irpro == "JVC") {
      //  void sendJVC(unsigned long data, int nbits, unsigned int repeat=0);
      //      irsend.sendJVC(irlong, irbit.toInt());
    }
    else if (irpro == "DISH") {
      //  irsend.sendDISH(irlong, irbit.toInt(), 4);
    }
    else {
      // irsend.sendRaw(tempbuf, templen, 38);
      //
    }
    
  } 
}
```

crankyoldgit, 2017-05-08:
I'm fairly sure it has nothing to do with ICACHE_FLASH_ATTR.
Can you explain what you mean by 'out of the loop'? It should work pretty much anywhere. As long as the IRsend object is created, and a irsend.begin() has been performed.

Do you have a link to your full code for your 'non-working' version so I can see why it wasn't working?

kiralikbeyin, 2017-05-09:
My code is very complicated, i am using some callbacks etc...

i am sure if there is ICACHE_FLASH in somewhere you must use a flag which works in the loop like a bool function. I couldnt find its explanation but i read it last summer from @igrr
````
loop{
 if (irgir)
  {
    irgir = false;
...
````
I think Sony (3 repeat) worked slower (made some time gaps) than other protocols (samsung-lg-nec) and we thought sony send function has problem.

Solution is simple - use a flag and send your ir in the main loop.

crankyoldgit, 2017-05-10:
I still don't think I understand, but I'm going to assume it is an issue with your code etc.
I'll stress again, I'm fairly sure ICACHE_FLASH doesn't have anything to do with it.
sendSony() should work outside of the loop{} stanza.

crankyoldgit, 2017-05-20:
Friendly ping for more details. Or should we just close this as you got it working?

kiralikbeyin, 2017-05-20:
Callbacks are making delays so its better to use the code directly in the loop with a flag. Thanks.

closed 2017-05-20 by kiralikbeyin