crankyoldgit/IRremoteESP8266 #203: Sending with 5v question

kiralikbeyin, 2017-05-08:
https://github.com/markszabo/IRremoteESP8266/wiki
If i put led to 5v, esp and device range is more longer.

How can i use 5v without hurting gpio? Adding extra components?

darshkpatel, 2017-05-08:
Try something like this

kiralikbeyin, 2017-05-08:
@darshkpatel no link

crankyoldgit, 2017-05-08:
If you are driving the IR LED via a transistor, then your ESP should be fine. i.e. Using a circuit similar to https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending

I drive one of my circuits using that diagram, but instead of connecting to a '3V3' pin, I connect it to the 'Vin' pin, and I'm powering the board via a decent 5V micro usb power supply. So, I'm over driving the IR LED to 5V, and I don't use any resistors. So it gets the full current the usb power supply can give and that the IR LED can use. It only does it for fractions of a second, so doesn't do _much_ harm, but does result in a brighter signal IMHO.

Doing so may shorten the life of the IR LED, but not by a heck of a lot.
#include <stddisclaimer.h>
"If you use your equipment out of spec. (over drive the LED) you do so at your own risk etc."

In that diagram, the ESP is signalling the transistor over 3.3V. That typically is enough to turn on a transistor. The transistor then allows the 5V to flow straight to the LED, effectively bypassing the ESP. So the ESP is not effected by the 5 Volts.

kiralikbeyin, 2017-05-08:
Very helpful thank you!

closed 2017-05-08 by kiralikbeyin