crankyoldgit/IRremoteESP8266 #204: sendNec help

kiralikbeyin, 2017-05-11:
My decode from remote:
```
Encoding  : NEC
Code      : C3E0C874 (32 bits)
Timing[211]: 
     +8950, -4400     + 600, -1650     + 600, -1700     + 600, - 500
     + 600, - 500     + 600, - 500     + 650, - 450     + 650, -1650
     + 650, -1600     + 650, -1600     + 650, -1650     + 600, -1650
     + 600, - 500     + 650, - 500     + 600, - 500     + 600, - 500
     + 650, - 500     + 600, -1650     + 600, -1650     + 650, - 450
     + 650, - 500     + 600, -1650     + 600, - 500     + 650, - 500
     + 600, - 500     + 650, - 450     + 650, -1650     + 600, -1650
     + 600, -1650     + 600, - 500     + 650, -1650     + 600, - 500
     + 650, - 500     + 600, - 500     + 600, - 500     + 600, - 500
     + 650, - 500     + 600, - 500     + 600, -1650     + 650, - 500
     + 600, -1650     + 600, - 550     + 600, - 500     + 600, - 500
     + 650, - 450     + 650, - 500     + 600, - 500     + 600, - 500
     + 650, - 500     + 600, - 500     + 650, - 500     + 550, - 550
     + 600, - 500     + 650, - 450     + 650, - 500     + 600, - 500
     + 600, - 500     + 650, - 500     + 600, - 500     + 650, - 500
     + 600, - 500     + 600, - 500     + 600, - 500     + 650, - 450
     + 650, - 550     + 600, - 500     + 600, - 500     + 600, - 500
     + 600, - 500     + 650, - 500     + 600, - 500     + 600, - 500
     + 650, - 500     + 600, - 500     + 600, - 550     + 600, - 500
     + 600, - 500     + 650, - 450     + 650, -1600     + 650, - 500
     + 650, - 500     + 600, - 500     + 600, - 500     + 600, - 500
     + 650, - 500     + 600, - 500     + 600, - 550     + 550, - 550
     + 600, - 500     + 650, -1600     + 650, - 500     + 600, -1650
     + 600, - 500     + 650, - 500     + 600, - 500     + 600, - 500
     + 650, - 500     + 600, - 500     + 600, - 500     + 650, - 500
     + 600, - 500     + 600, -1650     + 650, - 450     + 650, -1600
     + 650, -1650     + 650
unsigned int  rawData[211] = {8950,4400, 600,1650, 600,1700, 600,500, 600,500, 600,500, 650,450, 650,1650, 650,1600, 650,1600, 650,1650, 600,1650, 600,500, 650,500, 600,500, 600,500, 650,500, 600,1650, 600,1650, 650,450, 650,500, 600,1650, 600,500, 650,500, 600,500, 650,450, 650,1650, 600,1650, 600,1650, 600,500, 650,1650, 600,500, 650,500, 600,500, 600,500, 600,500, 650,500, 600,500, 600,1650, 650,500, 600,1650, 600,550, 600,500, 600,500, 650,450, 650,500, 600,500, 600,500, 650,500, 600,500, 650,500, 550,550, 600,500, 650,450, 650,500, 600,500, 600,500, 650,500, 600,500, 650,500, 600,500, 600,500, 600,500, 650,450, 650,550, 600,500, 600,500, 600,500, 600,500, 650,500, 600,500, 600,500, 650,500, 600,500, 600,550, 600,500, 600,500, 650,450, 650,1600, 650,500, 650,500, 600,500, 600,500, 600,500, 650,500, 600,500, 600,550, 550,550, 600,500, 650,1600, 650,500, 600,1650, 600,500, 650,500, 600,500, 600,500, 650,500, 600,500, 600,500, 650,500, 600,500, 600,1650, 650,450, 650,1600, 650,1650, 650};  // NEC C3E0C874
unsigned int  data = 0xC3E0C874;
````

sendNec from esp i get with other esp(decode)
so it is not sending 0xC3E0C874;
````
Encoding  : NEC
Code      : C3E0E0E8 (32 bits)
Timing[67]: 
     +8950, -4500     + 550, -1650     + 600, -1650     + 550, - 550
     + 600, - 500     + 600, - 550     + 550, - 550     + 600, -1650
     + 550, -1650     + 600, -1650     + 600, -1650     + 550, -1700
     + 550, - 550     + 600, - 550     + 550, - 550     + 600, - 500
     + 600, - 550     + 550, -1650     + 600, -1650     + 600, -1650
     + 550, - 550     + 600, - 500     + 600, - 500     + 600, - 550
     + 550, - 550     + 600, -1650     + 550, -1650     + 600, -1650
     + 600, - 500     + 650, -1600     + 600, - 500     + 600, - 550
     + 550, - 550     + 600
unsigned int  rawData[67] = {8950,4500, 550,1650, 600,1650, 550,550, 600,500, 600,550, 550,550, 600,1650, 550,1650, 600,1650, 600,1650, 550,1700, 550,550, 600,550, 550,550, 600,500, 600,550, 550,1650, 600,1650, 600,1650, 550,550, 600,500, 600,500, 600,550, 550,550, 600,1650, 550,1650, 600,1650, 600,500, 650,1600, 600,500, 600,550, 550,550, 600};  // NEC C3E0E0E8
unsigned int  data = 0xC3E0E0E8;
```

note: sendRaw works fine

crankyoldgit, 2017-05-11:
Usual questions: What version of the library are you using?
What commands are you sending with? etc. etc.

Your first decode is NOT a real/typical NEC protocol. It's a bad decode/match to NEC.
NEC is typically 32 bits (like your second decode) (67 - 3)/2 == 32 bits

The nr. of bits in your first decode is (211 - 3) / 2 == 104 bits

Your first decode is NEC-like in format as in it obeys most of the rules of NEC's protocols in timing etc, but NOT in length/nr. of bits.

I'm guessing your remote is a air conditioner or similar remote. They typically have bit lengths that long.
v2.0 should correctly not decode it as an NEC protocol. Sadly, even though v2.0 handles 64 bit message codes, it's not going to help you with 104 bits. You'll have to use sendGC or sendRaw to send it, or write your own send/decode() routines.

crankyoldgit, 2017-05-28:
Friendly ping. Any progress? If you haven't, please try the new [v2.0 release candidate](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.0-RC0).

JZ-SmartThings, 2017-05-29:
On a regular NodeMCU and on a D1 Mini, the 1.2.0 version definitely killed my regular sendNEC and I had to start using sendRaw for things that worked before with earlier library versions. I'll try to use the new 2.0.0 RC0 as time permits and post back with results.

Few questions/points that you may ask me to open a separate issue/enhancement for but I'd kick myself for not asking given this opportunity.

1. The dump sketch seems to spit out increments of 50 between values. I see lots of 550 600 650 1550 1600 1650, etc. However, when I look at samples out on the web, its using 9024,4512 at the head and values like 564 1692, etc. I had to use those exact values otherwise the little HDMI switcher which used to work with older library with sendNEC, did not work with the 50 increments. After adjusting to exact values that I mentioned above, I got success 100% of the time. Would you consider updating the dump sketch to use these more precise values at least when you think you sensed the NEC standard.

1. If possible, please see [my post](https://community.smartthings.com/t/esp8266-nodemcu-arduino-based-tv-remote/50161) 3rd paragraph from top, where I've changed your library a long time ago at 3 lines of code. Where you send HIGH, I sent LOW and vice-versa... this allowed me to use 3 IR LED's and send the ground signal to my LED's rather than positive. Would you consider this enhancement? Possibly add an optional flag to the library invoker **IRsend irsend(D3, optional parameter here)** so that I can use the full 5 volts rather than 3.3? If you do this, I would just change up my code a bit and stop distributing an older library of yours. Which I'm sure neither of us are a fan of :)

Thank you for your hard work as it easily does exactly what any good nerd can expect. Much appreciated!!!

crankyoldgit, 2017-05-29:
If you can, before you change libraries, I'd love a dump of what the allegedly 'broken' sendNEC in v1.2 is sending. Nothing really changed with it, and it works with my devices, so I'm quite confused.

The '50' increments is a hangover from the old arduino IR remote library. IIUC they have a much slower CPU and can't get accurate enough resolution. Moving to a finer resolution is something I want to tackle once v2.0 is officially out the front door. Please log a feature request/issue for that so I remember to do it. ;-) But you really do need to test it with v2.0. A lot of systemic timing issues have been addressed.

I can probably add something to send an inverted signal, but man, that's kind of dangerous w.r.t. possibly killing the IR LEDs if someone wires it up 'wrong'.

TBH you are much better, safer (for the IR LED and for the ESP8266 device to not have the 5V go anywhere near it. i.e. **use a transistor** rather than relying on the voltage drop from the resistor and the series of LEDs to the GPIO pin of the ESP8266. Get the resistor wrong, or wrong number of LEDs and you'd get 5Vs into the ESP8266 and probably fry it.

TL;DR: Don't do it your way, do it [this way](https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending) instead, but connect the LED to Vin (or 5V), instead of 3v3. Note: This is what I do.

Another plus side of using a transistor to gate the flow of the 5V and current is it should use less power.

Anyway, again, log an issue/feature request for an inverted signal for GPIO and I'll add it.

JZ-SmartThings, 2017-05-29:
THANK YOU!

The explanation of increments of 50 makes perfect sense given the history.

I'll add the new inverted signal request. Don't be too shy with that :) this library is for low level folks who know what they're doing. Just to be transparent, my lame 2 LED and no resistor configuration has bee working for a year w/o burning out the TSAL6100, not even the 6400's. However, I totally get your point and will try the NPN transistor as you mentioned. Keep in mind any resistance, even 20 ohms kills my 13 foot distance down to 7-8 feet. Hence the no resistor at risk of burning it. BTW, the new inverted feature needs no sample sketch or overwhelming documentation to decrease any accidents.

I already switched libraries but will switch back tomorrow to give you a dump of what happens with sendNEC. I also saw per the samples that the include reference changed, no biggie, did that but then got the error below, even though I see sendRaw in your code, friggin weird but this is what 2.0.0 RC0 does for me and hence un-testable at the moment other than single device which is not my code full of these sendRaw commands:

```
c:\test\IRSender.ino: In function 'void handleIr()':
IRSender:143: error: no matching function for call to 'IRsend::sendRaw(unsigned int [67], int, int)'
         irsend.sendRaw(rawData, 67, 38);
                                       ^
c:\test\IRSender.ino:143:39: note: candidate is:
In file included from c:\test\IRSender.ino:17:0:
C:\Users\joseph\Documents\Arduino\libraries\IRremoteESP8266\src/IRsend.h:37:8: note: void IRsend::sendRaw(uint16_t*, uint16_t, uint16_t)
   void sendRaw(uint16_t buf[], uint16_t len, uint16_t hz);
        ^
C:\Users\joseph\Documents\Arduino\libraries\IRremoteESP8266\src/IRsend.h:37:8: note:   no known conversion for argument 1 from 'unsigned int [67]' to 'uint16_t* {aka short unsigned int*}'
IRSender:149: error: no matching function for call to 'IRsend::sendRaw(unsigned int [20], int, int)'
         irsend.sendRaw(rawData, 20, 38);
                                       ^
```

JZ-SmartThings, 2017-05-29:
Following up... using the newer 2.0.0 RC0 has the same problem with sendNEC... BTW, I figured out to change int to uint16_t and all is well with 2.0.0 other than sendNEC.

**Code used to send:**
`irsend.sendNEC(0x1FE40BF, 38);`

**Response received from newest DumpV2 sketch:**
```
Encoding  : JVC
Code      : 7 (16 bits)
Timing[79]: 
     +9050, -4500     + 600, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 650, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, -1700     + 650, -1700
     + 650, -1700     + 600, -1700     + 600, -1700     + 650, -1700
     + 650, -1650     + 600, -1700     + 600, - 550     + 600, - 550
     + 650, -1650     + 600, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, - 550     + 600, -1700
     + 650, - 550     + 600, -1700     + 650, -1700     + 600, -1700
     + 600, -1700     + 600, -1700     + 650, -1700     + 600
uint16_t  rawData[79] = {9050,4500, 600,550, 600,550, 600,550, 600,550, 650,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,1700, 650,1700, 650,1700, 600,1700, 600,1700, 650,1700, 650,1650, 600,1700, 600,550, 600,550, 650,1650, 600,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,1700, 650,550, 600,1700, 650,1700, 600,1700, 600,1700, 600,1700, 650,1700, 600};  // JVC 7
uint32_t  address = 0x0;
uint32_t  command = 0xE0;
uint64_t  data = 0x7;
```

crankyoldgit, 2017-05-29:
I'll check it out in more detail when I'm near a real computer again, but
NEC is a 32bit protocol, not 38.

I.e. irsend.sendNEC(0x1FE40BF, 38); -> irsend.sendNEC(0x1FE40BF, 32);

crankyoldgit, 2017-05-29:
@JZ-SmartThings re: the uintXX_t things, the issue is mostly you are using 'int' instead of 'unsigned int' for your parameter declarations. That said, good feedback. I'll add that to the release notes as lots of other people will probably have that problem too.

Now that I can look at this on a computer rather than a phone, decodeNEC() won't match on the '38' bit NEC message, as it is expecting '32' bits.
I'll take a look at decodeJVC to see if I can get it to stop detecting the wrong size too.

JZ-SmartThings, 2017-05-29:
Thanks again @crankyoldgit I'd also recommend for the release notes to mention the new include declaration rather than the older one matching the library's name.

After I try the 32 bit designation for NEC I'll post back.

JZ-SmartThings, 2017-05-30:
Amazing, worked like a charm. I really appreciate all of the help!

crankyoldgit, 2017-05-30:
FYI, sendNEC() defaults to the correct bit size. As do most of the protocols in v2.0.
So you could make a call of `sendNEC(0x1FE40BF);` and it should work appropriately.

JZ-SmartThings, 2017-05-30:
Perfect, thanks again @crankyoldgit!

crankyoldgit, 2017-06-08:
@kiralikbeyin We haven't heard from you in over a month on this issue. Is this still a problem? Have you tried the v2.0 version?

JZ-SmartThings, 2017-06-08:
You've more than taken care of the problem in 2.0 and we really appreciate it!

One thing missing from his OP is how his sendNEC looked. If he had wrong bit-ness like I did then he'd have the same problems that I did, which started with 1.2 version of the library.

crankyoldgit, 2017-07-02:
Marking this closed. It should be fixed with the latest version, and closing due to no/low activity.

closed 2017-07-02 by crankyoldgit