crankyoldgit/IRremoteESP8266 #229: Inverted Signal to LEDs

JZ-SmartThings, 2017-05-29:
Please consider enhancing the code to provide the inverted signal to be sent to the IR gpio pin. Using a 5 volt feed for 3 LEDs is the motivation. This topic was originally discussed in issue #204 

Thank you!

JZ-SmartThings, 2017-05-30:
Friggin awesome! Tested with a 3 LED 5V configuration using the new parameter. Forgot to remove it before writing to my 2 LED configuration, nothing burned :) then wrote with the correct argument and all of my commands worked again including sendNEC.

What is the ETA of the newer 2.0.0 that you're preparing? Just wondering when to release my code as it's highly dependent on this library.

This works like a charm, please consider it closed. I'll also comment on the other bug that you were correct about NEC being 32 bit. Changed a bunch of mine and they work again.

Thanks again!!!

JZ-SmartThings, 2017-05-30:
BTW, I tested the branch labeled 229 not the newer 231. I'm guessing you'll merge later. Please comment on the release timeline if possible. Sorry to pester about that...

crankyoldgit, 2017-05-30:
re: [Q] When will 2.0 be official? [A] Soon. ;-)

There are three PRs pending. They need to be reviewed and merged before release, and some documentation updated. That realistically will be at least a week, plus any other bugs found etc. Thus plucking some numbers out of the air, I'd expect another release candidate in a few days, and final one in probably a couple of weeks. Sorry, my crystal ball isn't perfect. ;-)

JZ-SmartThings, 2017-05-30:
Thanks for the kind response, I know folks hate getting pinged on release time-lines :)

closed 2017-05-30 by JZ-SmartThings