crankyoldgit/IRremoteESP8266 #238: 'IRsend' does not name a type

phamhaibkit, 2017-06-08:
IRremoteESP8266:3: error: 'IRsend' does not name a type
Please help me solve the issue.
Thanks!

crankyoldgit, 2017-06-08:
More information please.

Please have a read of the [Troubleshooting guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide), especially the section on [Reporting an issue.](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue)

phamhaibkit, 2017-06-08:
I installed IRremoteESP8266, but when i compile IRsendDemo.ino example with NodeMCU 1.0 board then receive an error message:
IRremoteESP8266:3: error: 'IRsend' does not name a type.

I have not made any edits yet!

crankyoldgit, 2017-06-08:
What version of the library are you using?

Step 2: of the "Reporting an issue" link I sent.

> Tell us which version/revision of the library you are using, and when you last updated it etc. Are you using an up-to-date master clone? This helps us identify what might have been a known issue and/or if it has already been fixed.

e.g. Take a look at your *local* copy of this file: https://github.com/markszabo/IRremoteESP8266/blob/master/library.json

crankyoldgit, 2017-06-08:
FYI, the current version(s) (v1.2.0 and v2.0.0) of the library all seem to compile that program fine.
e.g. https://travis-ci.org/markszabo/IRremoteESP8266/jobs/239858539#L3078

phamhaibkit, 2017-06-09:
Oh! Sincerely thank you for the helpful help!
I solved the problem with v2.0.0-RC1.

crankyoldgit, 2017-06-09:
Excellent.

closed 2017-06-09 by crankyoldgit