crankyoldgit/IRremoteESP8266 #241: IR_receive

Sumgit91, 2017-06-13:
It's just a question! I am using Ken Shirriff's IR library for Arduino  UNO and your library For ESP 8266 ESP-12. The ESP is acting as a receiver and Arduino as a transmitter, however, the transmitter part is working fine but at the receiver side, the data is not getting received. I am using TSOP SM0038 receiver. Any suggestions or advice, why the problem persists.

crankyoldgit, 2017-06-13:
Have you tried using the example code on the ESP8266? e.g. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino
It should dump stuff to the serial console if you are getting valid codes.

And which version of the library are you using?

A suggested starter receiving circuit is https://github.com/markszabo/IRremoteESP8266/wiki#ir-receiving

Make sure you connect the output to (or change) the GPIO in the code to what ever you've connected it too.

Check the datasheet for your SM0038 module too. I had a quick look but most of them were in Chinese. Seems like it should work.

crankyoldgit, 2017-06-14:
@Sumgit91 This may be related to #243 
If so the newly minted v2.0.1 release should address it.

crankyoldgit, 2017-06-15:
Please try [v2.0.2](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.2). It fixes some more decoding issues introduced in v2.0.x

crankyoldgit, 2017-07-02:
Friendly ping. Have you had a chance to see if the latest version (2.0.3 as of this comment) fixes your problem?

crankyoldgit, 2017-07-14:
Marking this closed due to inactivity. Please re-open if you get back to us.

closed 2017-07-14 by crankyoldgit