crankyoldgit/IRremoteESP8266 #243: Issue decoding

jorgecis, 2017-06-14:
Version 2.00
Using the example code IRrecvDumpV2.ino

   I got this 

Encoding  : UNKNOWN
Code      : B47AF5B7 (34 bits)
Timing[67]:
     +9000, -4500     + 650, - 550     + 650, -1650     + 600, - 550
     + 650, - 550     + 600, -1650     + 650, - 550     + 600, -1650
     + 650, -1650     + 650, -1650     + 600, - 550     + 650, -1650
     + 650, -1650     + 650, - 550     + 600, -1650     + 650, -1650
     + 650, - 550     + 650, - 550     + 650, -1650     + 650, - 550
     + 650, - 550     + 650, - 550     + 600, - 550     + 650, - 550
     + 650, - 550     + 650, -1650     + 600, - 550     + 650, -1650
     + 650, -1650     + 650, -1650     + 650, -1650     + 650, -1650
     + 650, -1650     + 600
uint16_t  rawData[67] = {9000,4500, 650,550, 650,1650, 600,550, 650,550, 600,1650, 650,550, 600,1650, 650,1650, 65

0,1650, 600,550, 650,1650, 650,1650, 650,550, 600,1650, 650,1650, 650,550, 650,550, 650,1650, 650,550, 650,550, 65
0,550, 600,550, 650,550, 650,550, 650,1650, 600,550, 650,1650, 650,1650, 650,1650, 650,1650, 650,1650, 650,1650, 6
00};  // UNKNOWN B47AF5B7

Using the old code was detected as NEC

crankyoldgit, 2017-06-14:
@jorgecis Thanks for the report. I'll look into it.

First thing to note, it's reporting 34bits, where NEC is 32 bits, plus I know a rawData[] length of 67 is correct for 32 bits, thus something is broken.

crankyoldgit, 2017-06-14:
Hey @jorgecis 
 Can I get you to try this branch please (https://github.com/markszabo/IRremoteESP8266/tree/issue-243) while we wait for the code review etc and for the next version to be pushed?

I think it should address the issue you're seeing.
Oh, and thanks for adding the extra info in your bug report. It made it much more reproducible and helped track down the bug. :-)

crankyoldgit, 2017-06-14:
@jorgecis The fix has been submitted, and a new release created. Can you please confirm it addresses your issue? https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.1

jorgecis, 2017-06-15:
I flash my esp8266 using the last lib 2.01 and I got the same result.

crankyoldgit, 2017-06-15:
Thanks for the feedback. I'll try to set up something similar to your config to test it here.
Can you please send me details of what is producing the IR code etc so I can reproduce it?

jorgecis, 2017-06-15:
To made it simple I'm using the code in the example IRrecvDumpV2.ino as it without any change, my hardware is a adafruit esp8266, the only thing connected is a Infrared receiver in the PIN 14. 

I using a real remote control (ONKYO) but also I try others remotes and also another esp8222 to send codes, I found that the RC6 was detected.

Part of the debug log is

Attempting NEC decode
Matching MARK 9000 vs 9000. Matching: 136 <= 180 <= 228
Matching SPACE 4500 vs 4500. Matching: 66 <= 90 <= 111
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17

   The old library works fine.

crankyoldgit, 2017-06-15:
Thanks for that. I'll try to have a patch for you to test in a few mins. It will give me some more debug information.
In the meantime, can you please try commenting out lines 164-166 in ir_NEC.cpp, and try it again?
I think that should work, but I want to figure out the actual problem. I think the gap value isn't correct or working as I expect it to.

jorgecis, 2017-06-15:
Yes, If I comment those lines the code was detected

Encoding  : NEC
Code      : 4BB640BF (32 bits)
Timing[67]: 
     +9000, -4500     + 650, - 550     + 600, -1650     + 600, - 550
     + 600, - 550     + 600, -1650     + 650, - 550     + 600, -1650
     + 650, -1650     + 650, -1650     + 650, - 550     + 600, -1650
     + 650, -1650     + 650, - 550     + 600, -1650     + 600, -1650
     + 650, - 550     + 600, - 550     + 600, -1650     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, -1650     + 650, - 550     + 600, -1650
     + 600, -1650     + 650, -1650     + 650, -1650     + 650, -1650
     + 600, -1650     + 650
uint16_t  rawData[67] = {9000,4500, 650,550, 600,1650, 600,550, 600,550, 600,1650, 650,550, 600,1650, 650,1650, 650,1650, 650,550, 600,1650, 650,1650, 650,550, 600,1650, 600,1650, 650,550, 600,550, 600,1650, 600,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,1650, 650,550, 600,1650, 600,1650, 650,1650, 650,1650, 650,1650, 600,1650, 650};  // NEC 4BB640BF
uint32_t  address = 0x6DD2;
uint32_t  command = 0x2;
uint64_t  data = 0x4BB640BF;

crankyoldgit, 2017-06-15:
Can you try downloading this branch (https://github.com/markszabo/IRremoteESP8266/tree/issue-243) and try it out for me please?
It should give me the debug output (on serial) that I need to calculate the correct value for the NEC-like remote you have, and possibly show me what is going wrong etc.

I'm looking for the debug lines like:

> Matching ATLEAST 65500 vs 21940. Matching 1310 >= 225 min(329, 225)

crankyoldgit, 2017-06-15:
Please include relevant debug lines around it etc.

jorgecis, 2017-06-15:
I think that  maybe the problem is the rawlen is base 0

if (offset <= results->rawlen - 1 &&

I'll try the code in your branch

crankyoldgit, 2017-06-15:
I think it is correct. re: `offset <= results->rawlen - 1`
Plus, it would skip the gap test _earlier_ and then wouldn't be causing the problem.
I think the problem is there is a repeat code being sent soon after the NEC command. A number of protocols (e.g. Sherwood) do this. It may be throwing off the length as it is being included in the captured data. I believe the issue is the wrong inter-command gap value that it is expecting.

Waiting on your results. And mega thanks for the prompt testing for me.

jorgecis, 2017-06-15:
No problem, but I change the code, the original was offset <= results->rawlen, I think that the -1 is missing.

crankyoldgit, 2017-06-15:
Ah. yes. You are correct, and I was mis-quoting the wrong code.
So to re-state, I do believe the existing code is correct with respect to rawlen.
```
  if (offset <= results->rawlen &&
      !matchAtLeast(results->rawbuf[offset], NEC_MIN_GAP))
    return false;
```

But I'm never 100% certain. :-)

I think the issue lies in NEC_MIN_GAP is probably the wrong value in real-world situations. Hence the extra debug code to work this out.

jorgecis, 2017-06-15:
the function matchAtLeast is never called

Attempting Aiwa RC T501 decode
Attempting Sanyo LC7461 decode
Attempting NEC decode
Matching MARK 9000 vs 9000. Matching: 136 <= 180 <= 228
Matching SPACE 4500 vs 4500. Matching: 66 <= 90 <= 111
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1700 vs 1690. Matching: 23 <= 34 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 550 vs 1690. Matching: 23 <= 11 <= 40
Matching SPACE 550 vs 560. Matching: 6 <= 11 <= 12
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
[... 234 lines cut ...]
Matching SPACE 550 vs 1725. Matching: 24 <= 11 <= 41
Matching SPACE 550 vs 525. Matching: 6 <= 11 <= 11
Matching MARK 600 vs 525. Matching: 9 <= 12 <= 16
Matching ATLEAST 550 vs 11400. Matching 11 >= 171 min(171, 225)
Attempting SAMSUNG decode
Matching MARK 9000 vs 4500. Matching: 69 <= 180 <= 116
Attempting Whynter decode
Attempting DISH decode
Matching: 6 <= 180 <= 11
Attempting Sharp decode
Matching MARK 9000 vs 260. Matching: 4 <= 180 <= 10
Attempting Coolix decode
Encoding  : UNKNOWN
Code      : B47AF5B7 (34 bits)
Timing[67]: 
     +9000, -4500     + 650, - 550     + 600, -1700     + 600, - 550
     + 600, - 550     + 600, -1650     + 650, - 550     + 600, -1700
     + 600, -1650     + 650, -1650     + 600, - 550     + 600, -1700
     + 600, -1650     + 650, - 550     + 600, -1700     + 600, -1650
     + 650, - 550     + 600, - 550     + 600, -1700     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, -1650     + 600, - 550     + 600, -1700
     + 600, -1650     + 650, -1650     + 600, -1650     + 650, -1650
     + 650, -1650     + 600
uint16_t  rawData[67] = {9000,4500, 650,550, 600,1700, 600,550, 600,550, 600,1650, 650,550, 600,1700, 600,1650, 650,1650, 600,550, 600,1700, 600,1650, 650,550, 600,1700, 600,1650, 650,550, 600,550, 600,1700, 600,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,1650, 600,550, 600,1700, 600,1650, 650,1650, 600,1650, 650,1650, 650,1650, 600};  // UNKNOWN B47AF5B7

crankyoldgit, 2017-06-15:
Yes, it is being called.
e.g.
> Matching ATLEAST 0 vs 21940. Matching 0 >= 225 min(329, 225)

And .. that's way unexpected. hmmm.

jorgecis, 2017-06-15:
OK just in case if you need it, the value of offset is 68 the same that results->rawlen

crankyoldgit, 2017-06-15:
Thanks. I'm just uploaded a new commit to _that_ branch. I think I understand what is happening.
I've left the debug info turned on.

Can you let me know if that works? Obviously you'll need to re-download etc etc.

Again, thanks for the quick turnaround. This really helps.

jorgecis, 2017-06-15:
here is the result 
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 650 vs 560. Matching: 9 <= 13 <= 17
Matching SPACE 1650 vs 1690. Matching: 23 <= 33 <= 40
Matching MARK 600 vs 560. Matching: 9 <= 12 <= 17
Matching ATLEAST 600 vs 21940. Matching: 12 >= 225 [min(329, 225)]
Attempting Sony decode

jorgecis, 2017-06-15:
Sorry wrong code, let try again

crankyoldgit, 2017-06-15:
Phew. I saw the email, and thought "ohes noes, wtfbbq" ;-)

jorgecis, 2017-06-15:
sorry for that, 

this is the real result

Matching ATLEAST 0 vs 21940. Matching: 0 >= 225 [min(329, 225)]
Encoding  : NEC
Code      : 4BB640BF (32 bits)
Timing[67]: 
     +9000, -4500     + 650, - 550     + 650, -1650     + 600, - 550
     + 650, - 550     + 650, -1650     + 650, - 550     + 600, -1650
     + 650, -1650     + 650, -1650     + 650, - 550     + 600, -1650
     + 650, -1650     + 650, - 550     + 650, -1650     + 600, -1650
     + 650, - 550     + 650, - 550     + 650, -1650     + 600, - 550
     + 650, - 550     + 650, - 550     + 650, - 550     + 600, - 550
     + 650, - 550     + 650, -1650     + 650, - 550     + 600, -1650
     + 650, -1650     + 650, -1650     + 650, -1650     + 600, -1650
     + 650, -1650     + 650
uint16_t  rawData[67] = {9000,4500, 650,550, 650,1650, 600,550, 650,550, 650,1650, 650,550, 600,1650, 650,1650, 650,1650, 650,550, 600,1650, 650,1650, 650,550, 650,1650, 600,1650, 650,550, 650,550, 650,1650, 600,550, 650,550, 650,550, 650,550, 600,550, 650,550, 650,1650, 650,550, 600,1650, 650,1650, 650,1650, 650,1650, 600,1650, 650,1650, 650};  // NEC 4BB640BF
uint32_t  address = 0x6DD2;
uint32_t  command = 0x2;
uint64_t  data = 0x4BB640BF;

crankyoldgit, 2017-06-15:
Huzzah.
It looks like it's working correctly now. I'll clean it up and submit the PR.

jorgecis, 2017-06-15:
I have other issues with other protocols and I think is the same function, for example
This is a JVC code

Attempting JVC decode
Matching MARK 7450 vs 8400. Matching: 127 <= 149 <= 213
Matching SPACE 3700 vs 4200. Matching: 61 <= 74 <= 103
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 450 vs 1725. Matching: 24 <= 9 <= 41
Matching SPACE 450 vs 525. Matching: 6 <= 9 <= 11
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 450 vs 1725. Matching: 24 <= 9 <= 41
Matching SPACE 450 vs 525. Matching: 6 <= 9 <= 11
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 450 vs 1725. Matching: 24 <= 9 <= 41
Matching SPACE 450 vs 525. Matching: 6 <= 9 <= 11
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 450 vs 1725. Matching: 24 <= 9 <= 41
Matching SPACE 450 vs 525. Matching: 6 <= 9 <= 11
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 450 vs 1725. Matching: 24 <= 9 <= 41
Matching SPACE 450 vs 525. Matching: 6 <= 9 <= 11
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 450 vs 1725. Matching: 24 <= 9 <= 41
Matching SPACE 450 vs 525. Matching: 6 <= 9 <= 11
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching SPACE 1400 vs 1725. Matching: 24 <= 28 <= 41
Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Matching ATLEAST 1700 vs 11400. Matching: 34 >= 171 [min(171, 225)]
Attempting SAMSUNG decode
Attempting Whynter decode
Attempting DISH decode
Matching: 6 <= 149 <= 11
Attempting Sharp decode
Matching MARK 7450 vs 260. Matching: 4 <= 149 <= 10
Attempting Coolix decode
Encoding  : UNKNOWN
Code      : FBEE616 (18 bits)
Timing[35]: 
     +7450, -3700     + 500, -1400     + 500, -1400     + 500, -1400
     + 500, -1400     + 500, - 450     + 500, -1400     + 500, -1400
     + 500, -1400     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, -1400     + 500, - 450     + 500, -1400
     + 500, -1400     + 500
uint16_t  rawData[35] = {7450,3700, 500,1400, 500,1400, 500,1400, 500,1400, 500,450, 500,1400, 500,1400, 500,1400, 500,450, 500,450, 500,450, 500,450, 500,1400, 500,450, 500,1400, 500,1400, 500};  // UNKNOWN FBEE616

crankyoldgit, 2017-06-15:
JVC at least I have a remote for.

jorgecis, 2017-06-15:
Ok, just to inform you, If I comment the gap check in the ir_JVC the code is detected

Matching MARK 500 vs 525. Matching: 9 <= 10 <= 16
Encoding  : JVC
Code      : F70B (16 bits)
Timing[35]: 
     +7450, -3700     + 500, -1400     + 500, -1400     + 500, -1400
     + 500, -1400     + 500, - 450     + 500, -1400     + 500, -1400
     + 500, -1400     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, -1400     + 500, - 450     + 500, -1400
     + 500, -1400     + 500
uint16_t  rawData[35] = {7450,3700, 500,1400, 500,1400, 500,1400, 500,1400, 500,450, 500,1400, 500,1400, 500,1400, 500,450, 500,450, 500,450, 500,450, 500,1400, 500,450, 500,1400, 500,1400, 500};  // JVC F70B
uint32_t  address = 0xEF;
uint32_t  command = 0xD0;
uint64_t  data = 0xF70B;

crankyoldgit, 2017-06-15:
ta.

crankyoldgit, 2017-06-15:
re: your JVC dump.
> Matching ATLEAST 1700 vs 11400. Matching: 34 >= 171 [min(171, 225)]

1700 is an unexpectedly short gap. I think there may be more code in the buffer.

I've half a mind to pull out the gap checking code, but some protocols need it to distinguish from each other.

crankyoldgit, 2017-06-15:
I think I may have found the root cause.
I think the interrupt code isn't recording the final gap/space correctly. Previously, nothing relied on it.
The odd value you've got for NEC may be an artefact of garbage left in the buffer from previous captures.

jorgecis, 2017-06-15:
Ok, I you need another test let me know

crankyoldgit, 2017-06-15:
Sure. Can you please change line 135 in ir_JVC.cpp to:
'<' instead of '<='

And let me know if it works. I think you were right with how I'm calculating the end of the buffer.
i.e. your ' - 1' addition earlier.

jorgecis, 2017-06-15:
Ok I did the change,

I have 2 controls, with the first one the gap is there 

Attempting JVC decode
Matching MARK 7450 vs 8400. Matching: 127 <= 149 <= 213
Matching SPACE 3650 vs 4200. Matching: 61 <= 73 <= 103
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 450 vs 1725. Matching: 24 <= 9 <= 41
Matching SPACE 450 vs 525. Matching: 6 <= 9 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching ATLEAST 9650 vs 11400. Matching: 193 >= 171 [min(171, 225)]
Encoding  : JVC
Code      : C085 (16 bits)
Timing[69]: 
     +7450, -3650     + 550, -1350     + 550, -1350     + 550, - 400
     + 550, - 450     + 550, - 400     + 550, - 400     + 550, - 400
     + 550, - 400     + 550, -1350     + 550, - 400     + 550, - 400
     + 550, - 400     + 550, - 400     + 550, -1350     + 550, - 400
     + 550, -1350     + 550, -9650     + 550, -1350     + 550, -1350
     + 550, - 400     + 550, - 400     + 550, - 400     + 550, - 400
     + 550, - 400     + 550, - 400     + 550, -1350     + 550, - 400
     + 550, - 400     + 550, - 450     + 550, - 400     + 550, -1350
     + 550, - 400     + 550, -1350     + 550
uint16_t  rawData[69] = {7450,3650, 550,1350, 550,1350, 550,400, 550,450, 550,400, 550,400, 550,400, 550,400, 550,1350, 550,400, 550,400, 550,400, 550,400, 550,1350, 550,400, 550,1350, 550,9650, 550,1350, 550,1350, 550,400, 550,400, 550,400, 550,400, 550,400, 550,400, 550,1350, 550,400, 550,400, 550,450, 550,400, 550,1350, 550,400, 550,1350, 550};  // JVC C085
uint32_t  address = 0x3;
uint32_t  command = 0xA1;
uint64_t  data = 0xC085;

With the second one nogap but works 
[... 32 lines cut ...]
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 1350 vs 1725. Matching: 24 <= 27 <= 41
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Matching SPACE 400 vs 1725. Matching: 24 <= 8 <= 41
Matching SPACE 400 vs 525. Matching: 6 <= 8 <= 11
Matching MARK 550 vs 525. Matching: 9 <= 11 <= 16
Encoding  : JVC 
Code      : C03E (16 bits)
Timing[33]: 
     + 550, -1350     + 550, -1350     + 550, - 400     + 550, - 400
     + 550, - 400     + 550, - 400     + 550, - 450     + 550, - 400
     + 550, - 400     + 550, - 400     + 550, -1350     + 550, -1350
     + 550, -1350     + 550, -1350     + 550, -1350     + 550, - 400
     + 550
uint16_t  rawData[33] = {550,1350, 550,1350, 550,400, 550,400, 550,400, 550,400, 550,450, 550,400, 550,400, 550,400, 550,1350, 550,1350, 550,1350, 550,1350, 550,1350, 550,400, 550};  // JVC (Repeat) C03E
uint32_t  address = 0x3;
uint32_t  command = 0x7C;
uint64_t  data = 0xC03E;

crankyoldgit, 2017-06-15:
Interesting. The first one (the one with [69] in the timing/rawData) has two codes in it. i.e. It has a repeat code as well.

crankyoldgit, 2017-06-15:
Thus the gap thing/code is doing it's job. It's detecting a valid break in the message and working out the code is really a JVC code. Great.

crankyoldgit, 2017-06-15:
Can you please re-download and try that branch again. Just made some more fixes (to all protocols)
i.e. test your NEC/Onkyo and JVC remotes again.
Debug code is turned off. You shouldn't need it on, but I assume you know how to turn it on if things are not working as expected.

jorgecis, 2017-06-15:
With the last change, look like everything is working now, like for example JVC with repeat

Encoding  : JVC
Code      : C018 (16 bits)
Timing[35]: 
     +7400, -3650     + 550, -1350     + 550, -1350     + 550, - 400
     + 550, - 400     + 550, - 400     + 550, - 400     + 550, - 400
     + 550, - 400     + 550, - 400     + 550, - 400     + 550, - 450
     + 550, -1350     + 550, -1350     + 550, - 400     + 550, - 400
     + 550, - 400     + 550
uint16_t  rawData[35] = {7400,3650, 550,1350, 550,1350, 550,400, 550,400, 550,400, 550,400, 550,400, 550,400, 550,400, 550,400, 550,450, 550,1350, 550,1350, 550,400, 550,400, 550,400, 550};  // JVC C018
uint32_t  address = 0x3;
uint32_t  command = 0x18;
uint64_t  data = 0xC018;

Encoding  : JVC (Repeat)
Code      : C018 (16 bits)
Timing[33]: 
     + 550, -1350     + 550, -1350     + 550, - 450     + 550, - 400
     + 550, - 400     + 550, - 400     + 550, - 400     + 550, - 400
     + 550, - 400     + 550, - 450     + 550, - 400     + 550, -1350
     + 550, -1350     + 550, - 400     + 550, - 400     + 550, - 400
     + 550
uint16_t  rawData[33] = {550,1350, 550,1350, 550,450, 550,400, 550,400, 550,400, 550,400, 550,400, 550,400, 550,450, 550,400, 550,1350, 550,1350, 550,400, 550,400, 550,400, 550};  // JVC (Repeat) C018
uint32_t  address = 0x3;
uint32_t  command = 0x18;
uint64_t  data = 0xC018;

##########
But for some reason this key  is not decode as JVC but other keys in the same remote works

Encoding  : UNKNOWN
Code      : D69318ED (34 bits)
Timing[67]: 
     +7950, -3950     + 600, - 450     + 600, - 450     + 600, - 450
     + 600, - 450     + 600, - 450     + 550, - 450     + 600, - 450
     + 600, -1400     + 600, -1400     + 600, -1400     + 600, -1400
     + 550, -1400     + 600, - 450     + 600, - 450     + 600, - 450
     + 550, -1400     + 600, -1400     + 600, - 450     + 600, -1400
     + 600, - 450     + 600, - 450     + 600, - 450     + 600, - 450
     + 600, - 450     + 600, - 450     + 550, -1400     + 600, - 450
     + 600, -1400     + 600, -1400     + 550, -1400     + 600, -1400
     + 600, -1400     + 600
uint16_t  rawData[67] = {7950,3950, 600,450, 600,450, 600,450, 600,450, 600,450, 550,450, 600,450, 600,1400, 600,1400, 600,1400, 600,1400, 550,1400, 600,450, 600,450, 600,450, 550,1400, 600,1400, 600,450, 600,1400, 600,450, 600,450, 600,450, 600,450, 600,450, 600,450, 550,1400, 600,450, 600,1400, 600,1400, 550,1400, 600,1400, 600,1400, 600};  // UNKNOWN D69318ED

crankyoldgit, 2017-06-15:
Cool. Not sure about the last one. Looking at it, it doesn't match the JVC spec at all.
Unless there is a 32-bit JVC message variant. I'll do some research, but normally JVC is a 16-bit data protocol from my understanding.

crankyoldgit, 2017-06-15:
FYI, the code/fix for this has made it into next published release https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.2

Marking this issue close, please reopen if you don't believe it is fixed.

closed 2017-06-15 by crankyoldgit

crankyoldgit, 2017-06-17:
@jorgecis FYI, I did some research. I couldn't find any solid reference to a 32bit JVC code. All the official JVC documents I could find all indicated that it was all 16-bit. So that odd button you found. No idea. :-/