crankyoldgit/IRremoteESP8266 #245: Issue decoding panasonic codes

robste11159, 2017-06-14:
### Version/revison of the library used
IRremoteESP8266-2.0.0-RC1
works with 1.2.0

### Expected behavior
Panasonic codes not decoded at all, coming up as raw/unknown unless I comment out the line that checks the PANASONIC_MIN_GAP ... see below

### Actual behavior
Commented out the following line in ir_Panasonic.cpp:

 // if (!matchAtLeast(results->rawbuf[offset], PANASONIC_MIN_GAP))
  //  return false;

And the code is correctly decoded. The PANASONIC_MIN_GAP is being calculated as the value: 41440
No idea what it's used for?

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

Encoding  : PANASONIC
Code      : 4004100BCBD (48 bits)
Timing[99]: 
     +3550, -1750     + 500, - 450     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 450, - 450     + 500, - 450     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, - 450     + 450, - 450     + 500, - 450
     + 450, - 450     + 500, - 450     + 500, - 450     + 450, - 450
     + 450, -1300     + 500, - 450     + 450, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 450, -1300     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, -1300     + 500
uint16_t  rawData[99] = {3550,1750, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 450,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 450,450, 500,450, 450,450, 500,450, 500,450, 450,450, 450,1300, 500,450, 450,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 450,1300, 500,1300, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,1300, 500};  // PANASONIC 4004100BCBD
uint32_t  address = 0x4004;
uint32_t  command = 0x100BCBD;
uint64_t  data = 0x4004100BCBD;

### Steps to reproduce the behavior

Use the 

#### Example code used

IRrecvDumpV2 is all I used

#### Circuit diagram and hardware used (if applicable)

3.3 feeding the IR Receiver into GPIO 5

### I (have/have not) followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide)

### Other useful information
Remote control model: N2QAYB000352.  My TV is a 2008 panasonic plasma.

crankyoldgit, 2017-06-14:
@robste11159 This looks like a duplicate of #243 and has been fixed (See #244)
You can download the [latest release (v2.0.1) ](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.1)

Can you download that version and please verify for me that it fixes your issue?

crankyoldgit, 2017-06-14:
btw. Thanks for the detail report too!

robste11159, 2017-06-14:
Downloaded 2.0.1 and it doesn't fix the issue.
I increased RAWBUF to 500 and still getting overflow:

IR code too long. Edit IRremoteInt.h and increase RAWBUF
Timing[99]: 
     +3550, -1750     + 500, - 450     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, -1300     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, -1300     + 500
uint16_t  rawData[99] = {3550,1750, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,1300, 500};  // UNKNOWN F61E2A57

I'll do some digging and work out where it's leaving the decodePanasonic() routine

robste11159, 2017-06-14:
Had again to comment out this:

 // if (offset <= results->rawlen &&
 //     !matchAtLeast(results->rawbuf[offset], PANASONIC_MIN_GAP))
 //   return false;

After that:

Encoding  : PANASONIC
Code      : 4004100BCBD (48 bits)
Timing[99]: 
     +3550, -1700     + 500, - 450     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 450, - 450     + 500, - 450     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, -1300     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, -1300     + 500
uint16_t  rawData[99] = {3550,1700, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 450,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,1300, 500};  // PANASONIC 4004100BCBD
uint32_t  address = 0x4004;
uint32_t  command = 0x100BCBD;
uint64_t  data = 0x4004100BCBD;

crankyoldgit, 2017-06-14:
@robste11159 I'm having trouble reproducing the problem.
I created a test case for it based on the rawData you provided, and it _should_ decode fine, and appears to do so. (see https://github.com/markszabo/IRremoteESP8266/tree/issue-245 & https://github.com/markszabo/IRremoteESP8266/compare/issue-245)

The _overflow_ is worrying. It shouldn't be doing that unless there are lots of codes being sent.

crankyoldgit, 2017-06-14:
FYI, don't trust the 0x4004100BCBD value. I just noticed a bug in it. It's a bug in the dumping example code, not the library. The value should be 0x40040100BCBD ... i.e. an extra '0' after the 0x4004.

crankyoldgit, 2017-06-14:
@robste11159 Given your feedback/solution and my playing with the code, I'm guessing the calculation of PANASONIC_MIN_GAP is wrong/bad/poor. That is, it is to large for what you are seeing/getting.
I don't have any panasonic devices to test with, so I need your help.

I've added some specific debug lines and turn on debugging. Can I get you to try out https://github.com/markszabo/IRremoteESP8266/tree/issue-245 and send me the output?
That should help me track down what the correct gap size is.

I assume you're getting code from a 'real' Panasonic remote. (i.e. the one listed in the initial comment)

crankyoldgit, 2017-06-14:
The key output I'm looking for will look something like this:
> ...
Matching: 6 <= 10 <= 11
Matching: 19 <= 26 <= 33
Matching: 6 <= 10 <= 11
**rawlen = 100**
**offset = 100**
**mingap = 41440**
**gap (raw) = 20**
**gap (usecs) = 1000**
...

robste11159, 2017-06-14:
Attempting Sanyo LC7461 decode
Attempting NEC decode
Attempting Sony decode
Attempting Mitsubishi decode
Attempting RC5 decode
Attempting RC6 decode
Attempting RC-MM decode
Attempting Denon decode
Attempting Panasonic decode
Attempting LG (28-bit) decode
Attempting LG (32-bit) decode
Attempting JVC decode
Attempting SAMSUNG decode
Attempting Whynter decode
Attempting DISH decode
Attempting Sharp decode
Attempting Coolix decode
Attempting Aiwa RC T501 decode
Matching MARK 3550 vs 9000. Matching: 136 <= 71 <= 228
Attempting Sanyo LC7461 decode
Matching MARK 3550 vs 9000. Matching: 136 <= 71 <= 228
Attempting NEC decode
Matching MARK 3550 vs 9000. Matching: 136 <= 71 <= 228
Attempting Sony decode
Matching MARK 3550 vs 2400. Matching: 37 <= 71 <= 63
Attempting Mitsubishi decode
Matching MARK 3550 vs 303. Matching: 5 <= 71 <= 11
Attempting RC5 decode
Matching: 41 <= 71 <= 70
Matching: 28 <= 71 <= 47
Matching: 14 <= 71 <= 25
Attempting RC6 decode
Matching MARK 3550 vs 2666. Matching: 41 <= 71 <= 70
Attempting RC-MM decode
Matching MARK 3550 vs 416. Matching: 7 <= 71 <= 13
Attempting Denon decode
Matching MARK 3550 vs 3456. Matching: 53 <= 71 <= 89
Matching MARK 1750 vs 1728. Matching: 27 <= 35 <= 46
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 26 <= 33
Matching: 6 <= 10 <= 11
Matching: 19 <= 8 <= 33
Matching: 6 <= 8 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
Matching: 6 <= 10 <= 11
Matching: 19 <= 9 <= 33
Matching: 6 <= 9 <= 11
[... 261 lines cut ...]
Attempting Whynter decode
Matching MARK 3550 vs 750. Matching: 12 <= 71 <= 22
Attempting DISH decode
Matching: 6 <= 71 <= 11
Attempting Sharp decode
Matching MARK 3550 vs 260. Matching: 4 <= 71 <= 10
Attempting Coolix decode
Matching MARK 3550 vs 4480. Matching: 68 <= 71 <= 115
Matching SPACE 1750 vs 4480. Matching: 65 <= 35 <= 110
IR code too long. Edit IRremoteInt.h and increase RAWBUF
Timing[99]: 
     +3550, -1750     + 500, - 450     + 500, -1300     + 500, - 400
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, -1300     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 400     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, -1300     + 500
uint16_t  rawData[99] = {3550,1750, 500,450, 500,1300, 500,400, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,400, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,1300, 500};  // UNKNOWN F61E2A57

robste11159, 2017-06-14:
[image]

crankyoldgit, 2017-06-14:
@robste11159 Thanks for that. That's really helpful.
So it seems the correct value for the gap at the end of a panasonic command is 5000.
Which seems really odd that its a neat number.

I've updated that branch with a change that should match it. Can you please try it out? It should match the panasonic remote now.
I'm still working on the other issues.

crankyoldgit, 2017-06-14:
I should also have updated the example code to not fail, but warn, if the buffer overflowed.
Also, it now should print the hex digits correctly.

robste11159, 2017-06-14:
Top work! Problem solved!

Encoding  : PANASONIC
Code      : 40040100BCBD (48 bits)
Timing[99]: 
     +3550, -1700     + 550, - 400     + 500, -1300     + 500, - 400
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 400
     + 550, - 400     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 400     + 500, -1300     + 500, - 400
     + 500, - 400     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 400     + 500, - 450     + 500, - 450
     + 500, -1300     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, - 450     + 500, -1300     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, -1300     + 500, - 450
     + 500, -1300     + 500
uint16_t  rawData[99] = {3550,1700, 550,400, 500,1300, 500,400, 500,450, 500,450, 500,450, 500,400, 550,400, 500,450, 500,450, 500,450, 500,450, 500,400, 500,1300, 500,400, 500,400, 500,450, 500,450, 500,450, 500,450, 500,400, 500,450, 500,450, 500,1300, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,450, 500,1300, 500,450, 500,1300, 500,1300, 500,1300, 500,1300, 500,450, 500,1300, 500};  // PANASONIC 40040100BCBD
uint32_t  address = 0x4004;
uint32_t  command = 0x100BCBD;
uint64_t  data = 0x40040100BCBD;

robste11159, 2017-06-15:
Can I ask an unrelated question?

I just need some guidance rather than a solution, as part of my project I'll need to send discrete codes which obviously I can't capture from my remote (ie discrete power on, power off, hdmi 1, hdmi 2 etc)

I found these here: ftp://ftp.panasonic.com/pub/panasonic/drivers/monitors/Discrete-remote-control-codesProntoCCFformat.pdf

What's the easiest way to convert a pronto code into the equivalent of this:
uint32_t address = 0x4004;
uint32_t command = 0x100BCBD;
uint64_t data = 0x40040100BCBD;

Any existing libraries?

Thanks! Rob

crankyoldgit, 2017-06-15:
We don't currently support _Pronto_ codes, but it's something I want to add. Please create a feature request/separate issue for it, and that way I'll remember.

What I personally use for creating codes I can't capture myself is the Global Cache codes. (Get them from https://irdb.globalcache.com/) and use the `sendGC()` method.

If you are comfortable with C/C++ programming, you can have a look under the test/ directory and see how I use `sendGC()` in most of the protocol's tests. You can use that as a hacky-way to convert Global Cache codes into native IRremote codes for the various protocols.

crankyoldgit, 2017-06-15:
FYI, the original bug/issue should be fixed in the current [v2.0.2 release](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.2)

Marking this issue closed. Please feel free to re-open if it isn't addressed in that release.

closed 2017-06-15 by crankyoldgit

crankyoldgit, 2017-06-19:
@robste11159 Can I get you to please try v2.0.3 with debug on with your panasonic remote?
I'm looking for the "at least" value. i.e. the post-message gap. We changed it to 5000 for your problem, but I think I've found a subsequent issue that means that value was probably junk.
I expect it will show '0' as the value now, with that new version.