crankyoldgit/IRremoteESP8266 #246: Duplicate NEC signals (receive -> send)

mortenmoulder, 2017-06-14:
I am trying to duplicate signals. I basically want to receive a signal and send the same one. When I run the example which does:

    if (irrecv.decode(&results)) {
        Serial.println((uint32_t) (results.value & 0xFFFFFFFF), HEX);
    }

I get `D7E84B1B` when I hit the button on the remote. However, this is not a HEX value I can send using `irsend.sendNEC()`. I [found a website](https://eletroarduino.wordpress.com/tag/ir-receiver-module/) that shows this:

>‘real’ decode: FF02FD, hash decode: D7E84B1B

When I put in the FF02FD HEX value like `irsend.sendNEC(0x00FF02FD, 32)` and I try that, it actually works. However, how do I get the FF02FD HEX value using this library? What I basically want to do is just send the exact same IR signal out as I receive. Why does that have to be so hard? haha

I noticed the site uses IRremote, so doing something like:

    Serial.println(results.value, HEX);

that should work, but I get a "call of overloaded 'println(uint64_t&, int)' is ambiguous" error when I try (probably also because IRremote isn't supported with the ESP8266).

I know this isn't an issue with this library, but this is probably the only place where I can get a qualified answer. Help appreciated! Thanks.

crankyoldgit, 2017-06-15:
Hey @mortenmoulder 
I fixed a bug in decoding NEC commands yesterday (See #243).
I created a new release (v2.0.1) which should address the NEC decoding issue. i.e. Wrong value.

To solve your printing issue of uint64's, you should try using the routine that's due for release soon **(Branch https://github.com/markszabo/IRremoteESP8266/tree/issue-245)**.

If you check out the updated dump examples in that branch, you'll see a call of the following should work for you:
> #include <IRutils.h>
>
> serialPrintUint64(results.value, 16);
> Serial.println("");

You can try HEX instead of 16. I think that will work.

The root of the printing issue is that the Arduino framework doesn't support 64 bit values. Which is a major bummer. Hence having to write our own routine.

crankyoldgit, 2017-06-15:
Please try the latest release ([v2.0.2](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.2))

mortenmoulder, 2017-06-15:
Hi again, @crankyoldgit 

It looks like it works yeah! Thanks a lot.

So maybe I got this wrong or something, but can you help me understand something here?

So I upload the sketch to my ESP8266 and I look at the Serial Monitor. I press the same button a few times and it basically goes like this:

    FF02FD
    FF02FD
    FF02FD
    FF02FD
    FF02FD
    FF02FD
    D7E84B1B
    D7E84B1B
    D7E84B1B
    D7E84B1B
    DB11B4E0
    F5CD1280
    D7E84B1B

Basically the first few ones are FF02FD, which is the button I press, then it starts to turn into D7E84B1B and other randomness. After the first few clicks/reads, I basically never see the FF02FD again. I have my circuit wired up exactly like this: https://cdn.instructables.com/FM7/TZ9F/ITTD0OFJ/FM7TZ9FITTD0OFJ.MEDIUM.jpg

I have confirmed this probably 5-6 times now, that FF02FD turns up the first few times, then it's all random after that.

crankyoldgit, 2017-06-15:
Got a link to your code as well?

mortenmoulder, 2017-06-15:
Sure thing. I also noticed this only happens to my cheap Chinese LED strip remote. It's not the remote's fault, I don't think, because it's so consitent. If I pick up my LG remote and I hit the button many times, I would say 5-10% is wrong. Under 5% if I wait maybe a second between each press.

    #include <IRutils.h>
    #include <IRrecv.h>
    #include <IRsend.h>

    IRrecv irrecv(D1);
    IRsend irsend(D2);

    decode_results results;

    void setup() {
        irrecv.enableIRIn();
        irsend.begin();
        Serial.begin(115200);
    }

    void loop() {
        if (irrecv.decode(&results)) {
            Serial.println(String((uint32_t) results.value, HEX));
            irsend.sendNEC(results.value);
            irrecv.resume();  // Receive the next value
        }
        delay(100);
    }

crankyoldgit, 2017-06-15:
I'll need to check, but I think you are using the serial port gpios for
your LEDs. Check that, and use different ones.

mortenmoulder, 2017-06-15:
The serial GPIO are the TX/RX, right? Those are at the very bottom just above the GND and 3v output.

crankyoldgit, 2017-06-15:
Also, you might be interfering with yourself if the receiver can see the
transmitter.

mortenmoulder, 2017-06-15:
The receiver and the transmitter are under and above the table. I did try removing the sending LED/transmitter but it's still the same.

mortenmoulder, 2017-06-15:
So it basically comes down to, that none of my cheap Chinese LED strip remotes work consistently, but all my other remotes for TVs and such work fine. Amazingly fine, actually.

What could the cause of this be?

mortenmoulder, 2017-06-15:
Example of a few clicks (that actually went surprisingly well in the beginning):

https://pastebin.com/raw/wJvvyFCz

crankyoldgit, 2017-06-15:
You may have answered your own question. :-)

Try using the dumpv2 example code with the cheap remotes and see what you
get.

mortenmoulder, 2017-06-15:
Not a lot of a difference, if I have to be honest:

https://pastebin.com/raw/byPYLMXy

Many of the timings are different and are of different lengths. I see some people have the exact same issue as me: https://forum.arduino.cc/index.php?topic=149498.0 - Someone on page 3 explains my very issue. 

Why it works fine RIGHT after I upload the new code is so effing weird.

crankyoldgit, 2017-06-15:
Try increasing the TOLERANCE value in the code.
Some of those commands are very close to being valid. I.e. one value in
say, 60, is off/out of spec.

The remotes design might mean it doesn't have enough electrical power to
send frequent signals, if it is one of those coin battery types. Guessing
only.

mortenmoulder, 2017-06-15:
I tried, man. I did all the tolerances from 0.1 all the way up to 0.95 and it was still the exact same 😞

I bet you 100% it's not an issue with the remote. If that was the case, I would be getting different codes all the time. My money is on the receiver. Can the wires be too long? They're probably 15 centimeters at the moment. I have no resistors or anything in between. Simply hooked directly up to the GPIO.

crankyoldgit, 2017-06-15:
A poor receiver module is certainly a possibility.

Just checking. The TOLERANCE value is an integer between 0 and 100 from
memory

mortenmoulder, 2017-06-15:
The tolerance you're talking about is the one in `IRrecv.cpp` right? 

    int16_t IRrecv::compare(uint16_t oldval, uint16_t newval) {
      if (newval < oldval * 0.8)
        return 0;
      else if (oldval < newval * 0.8)
        return 2;
      else
      return 1;
    }

crankyoldgit, 2017-06-15:
If it's working for 'good' remotes (like TV's etc) and not for the cheap
ones, then it sounds like it isn't the receiver

mortenmoulder, 2017-06-15:
Then what? 😞

crankyoldgit, 2017-06-15:
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h line
31

mortenmoulder, 2017-06-15:
Oh okay. So I did some testing between 10 and 100. The lower tolerance, the more of `d7e84b1b` I get. Getting the `ff02fd` (which is what I need) seems completely random, still. Over 50 clicks at 10, 25, 50, 80, and 100 and I barely get any `ff02fd`.

crankyoldgit, 2017-06-15:
So. To test my battery/power theory. Try waiting a few minutes after it
goes too garbage, then do a couple of single presses. If that works, then I
think the remote has a current draw problem.

mortenmoulder, 2017-06-15:
I've done that for the past few hours and it's the same problem. It's the exact same issue for the 4-5 other remotes I have (3 are different but works for the same LED strip receiver).

crankyoldgit, 2017-06-15:
I'm assuming the remote is working off a single lithium coin/button cell,
and it charges a capacitor slowly to give it power to send signals

mortenmoulder, 2017-06-15:
That is correct. Charging up a capacitor shouldn't take minutes or seconds, but for the sake of argument, I do agree.

However, putting down the remote for minutes and picking it up, still gives me different values.

crankyoldgit, 2017-06-15:
Okay. I'll see if I can debug it when I get home

mortenmoulder, 2017-06-15:
Thanks a lot. If you have once of these remotes, that would be very cool.

crankyoldgit, 2017-06-16:
Got a link or a picture of the remotes?

crankyoldgit, 2017-06-16:
I can't find the remote I am thinking of, which similarly controls a cheap-o LED strip. That remote runs off a coin cell. I did find another one, based on Mitsubishi codes, that also is a cheap-o coin cell remote.

Like you experienced, it worked for a few codes, and them gave garbage or nothing at all. I suspect the battery is very old and flat. It's been sitting in a draw for years.

I did test a bunch of physical remotes that use AAA's or AA's, NEC protocols and others. They were all rock solid. None of them or the ESP & non-name cheap-o IR receiver module I got off ebay/aliexpress  missed a beat.
Everything was as expected. There was a cheap knock off remote that produced invalid NEC codes. As in, they look like NEC codes, but the internal checks failed. i.e. wrong values, not bad signal/pulses etc.

I pumped hundreds of codes at the dumpv2 code, and it all worked as I expected. All the codes were repeatable and didn't waver.

I'm guessing it is the cheap remotes, and the coin cell. They can't do a decent current draw. They are not designed to do it, but they are cheap, and okay for very infrequent use.

You might want to check the battery in the remote(s) you are using is good too. I've had a number of remotes/coin cells from china that were basically had-it on arrival. They are cheap for a reason after all. ;-)

esa-n, 2017-06-16:
Decoding of NEC code fails. Wrong code will be received after several times.

crankyoldgit, 2017-06-17:
@esa-n Can you be more specific? As in provide more details/data, and how I can reproduce your problem?
Have you tried these steps? https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#receiving-an-ir-signal

You may also wish to read https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue to help you get a better/faster response.

You may also want to create your own issue/bug, as it may not be the same one as this one.

As I said in the previous post it appears to be handling reception of NEC codes correctly, even after hundreds of codes.

More than happy to help, but I need more to work with.

crankyoldgit, 2017-06-17:
@mortenmoulder For the record, I found a cheap-o coin-cell remote that does the NEC protocol. With a fresh battery in it, I got 100% code recognition and they were all the same, even after tens to hundreds of button presses.
I'm using the example decoding circuit on the Wiki page, a cheap-o nodemcu & a TSOP38238 module I ordered from Aliexpress.

That was all done using v2.0.2 of the library and the IRrecvDumpV2 example code.

All of that makes me think it's a hardware (non-library) related issue at your end. Sorry.

mortenmoulder, 2017-06-17:
Hi again, @crankyoldgit 

So you suggest trying to change the coin cell battery? I guess that could explain why the remotes work the same, because I just changed the batteries.

Weird. I'll give it a go once I get home Sunday. Thanks a lot!

crankyoldgit, 2017-06-17:
Okay. I've some how managed to get my ESP device *once* into the same condition you've reported. Not sure how though. It only happened after I was playing with one of the cheap-o remotes with a low battery. It was then b0rk3n for even normal remotes after that. The raw codes it was displaying after it went crazy do in fact decode correctly when manually and subsequently by the library.

It seems there is something that is causing the library to get tripped up.
i.e. same code, after the ESP is reset has worked for 20 minutes solid with 'good' remotes and cheap remotes alike after thousands of codes being sent.

esa-n, 2017-06-17:
Hello. Does not it work at this signal timing of Toshiba TV?
Raw (68): 9100 -4400 700 -500 700 -500 700 -500 700 -500 700 -500 700 -500 700 -1600 700 -500 700 -1600 700 -1600 700 -1600 ....

crankyoldgit, 2017-06-17:
@esa-n I can't tell unless you supply the entire rawData structure.

crankyoldgit, 2017-06-17:
@mortenmoulder I was finally able to reliably cause the issue to happen, thus I was able to track down the bug. I believe I have a fix that works.
Feel free to try the code in branch: https://github.com/markszabo/IRremoteESP8266/tree/issue-246

mortenmoulder, 2017-06-17:
@crankyoldgit Great, I'll give it a try tomorrow when I get home!

crankyoldgit, 2017-06-18:
[v2.0.3](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.3) has been released including a patch to fix this issue.

crankyoldgit, 2017-06-19:
@mortenmoulder Were you able to give it a go?

mortenmoulder, 2017-06-21:
@crankyoldgit Hell yeah, dude! Works like a damn charm right now. Sorry for the late response, I was preparing exam this whole week.

Thanks a lot! What turned out to be the problem?

crankyoldgit, 2017-06-22:
@mortenmoulder Excellent. Glad to hear it. Marking this issue closed.

The problem/solution is best described here:
https://github.com/markszabo/IRremoteESP8266/commit/74797a045a375b72703d9a271a584cc643e0d32b
or
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L160

closed 2017-06-22 by crankyoldgit

teemue, 2018-05-31:
Sorry for bringing up this issue but I'm trying to achieve the exactly same thing: To replicate NEC signal. I get the value but how am I supposed to send it forward? irsend.sendNEC(results.value, results.bits) doesnt work.

crankyoldgit, 2018-05-31:
Have you checked `results.decode_type` is NEC? i.e. 3

```
// Only resend it if the message was an NEC one.
if (results.decode_type == NEC) {
  irsend.sendNEC(results.value, results.bits);
}
```

P.S. Please don't add to a long-ago closed issue. Create a new one, and reference this one if you need to.