crankyoldgit/IRremoteESP8266 #250: 'IRrecv' does not name a type

roybm, 2017-06-15:
### Version/revison of the library used
_Typically located in the `library.json` file in the root directory of the library.
e.g. v2.0.0, 'master' as at 1st of June, 2017. etc._

### Expected behavior
try to compile nodemcu ir blaster...

### Actual behavior
I got many errors like:
IRController:29: error: 'IRrecv' does not name a type...

IRController:30: error: 'IRsend' does not name a type...
.
.
.
 I believe the problem is in the library installation although I did it several times manually and also try through the library manager. 

I will be happy for  some help here.

thanks

Eden-Sun, 2017-06-15:
Re-install and upgrade v2.0.2 might solve your problem.
Or try manually download the folder into your `libraries` folder.

crankyoldgit, 2017-06-15:
@roybm I think you (or the code/project you are using) needs to upgrade how it talks to the library.

e.g. https://github.com/markszabo/IRremoteESP8266/wiki/Upgrading-to-v2.0

closed 2017-06-15 by roybm