crankyoldgit/IRremoteESP8266 #255: Error using example. It doesn't compile

kryptoss, 2017-06-24:
### Version/revison of the library used
version 2.0.2

### Expected behavior
I tried to use IRrecvDumpV2 example after install library using arduino manager

### Actual behavior
It doesn't compile

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
Arduino IDE 1.8.3 error:
In file included from /home/userArduino/libraries/IRremoteESP8266/src/IRrecv.h:10:0,
                 from /home/user/Arduino/libraries/IRremoteESP8266/src/ir_JVC.cpp:5:
/home/user/Arduino/libraries/IRremoteESP8266/src/ir_JVC.cpp: In member function 'void IRsend::sendJVC(uint64_t, uint16_t, uint16_t)':
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
 #define max(a,b) ((a)>(b)?(a):(b))
                  ^
/home/user/Arduino/libraries/IRremoteESP8266/src/ir_JVC.cpp:60:16: note: in expansion of macro 'max'
     space(std::max(JVC_MIN_GAP, JVC_RPT_LENGTH - usecs.elapsed()));

It seems Arduino.h for esp8266 defines the macro:
#define max(a,b) ((a)>(b)?(a):(b))
So, precompiler tries to replace "max" string in 
space(std::max(JVC_MIN_GAP, JVC_RPT_LENGTH - usecs.elapsed())); 
using the macro, leading to incorrect syntax. So, It doesn't compile.

Regards

crankyoldgit, 2017-06-25:
Hey @kryptoss 
I just tried this on a clean/fresh install of Arduino 1.8.3/ESP8266 core for Arduino 2.3.0/IRremoteESP8266 v2.0.3 under Windows and I can't reproduce your reported issue. It compiled/verified fine for me. I also have it installed under Linux and it works there too. Plus we test/check every change to the tree with [Travis](https://travis-ci.org/markszabo/IRremoteESP8266). As you can see [here](https://travis-ci.org/markszabo/IRremoteESP8266/jobs/245809733#L2230) & [here](https://travis-ci.org/markszabo/IRremoteESP8266/jobs/245809733#L2594), it compiled fine on the last update.

This leads me to think you may have something else in your environment which is causing this.

I note you are using v2.0.2 of our library. v2.0.3 is the latest. It has a decoder bug fix, but nothing that would change IRrecvDumpV2.ino's compilation status.

For the record. It (Arduino IDE) shouldn't be using it's max() macro, as we are using the std::max() method from `#include <algorithm>` (see [here](https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_JVC.cpp#L4))

Not sure what to suggest other than, go into board manager, and update, and then into library manager and update. Then try compiling a fresh/clean copy from the examples menu.

crankyoldgit, 2017-06-25:
[image]

roberthardesty, 2017-07-01:
I am having the same issue

crankyoldgit, 2017-07-01:
Can you tell me more about your environment so I can replicate the problem?
E.g. why platform, and what versions of the software and libraries you are
using. Especially those I listed in the previous reply.

That might help me track down your problem.
In the meantime, check you are using the latest versions of everything.

zuidwijk, 2017-07-02:
Same issues here... Got latest arduino 1.8.3 for Mac, placed all the included libraries in the main library. Default code without any adjustments gives me too the errors...
```
Arduino: 1.8.3 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

WARNING: Category 'LED' in library CHT16K33 LED Matrix Library is not valid. Setting to 'Uncategorized'
WARNING: Category 'Input' in library MechInputs is not valid. Setting to 'Uncategorized'
xdrv_ir_send:46: error: 'IRsend' does not name a type
 IRsend *irsend = NULL;
 ^
/Users/marcel/Downloads/Sonoff-Tasmota-master-2/sonoff/xdrv_ir_send.ino: In function 'void ir_send_init()':
xdrv_ir_send:50: error: 'irsend' was not declared in this scope
   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND
   ^
xdrv_ir_send:50: error: expected type-specifier before 'IRsend'
   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND
                ^
xdrv_ir_send:50: error: expected ';' before 'IRsend'
/Users/marcel/Downloads/Sonoff-Tasmota-master-2/sonoff/xdrv_ir_send.ino: In function 'boolean ir_send_command(char*, uint16_t, char*, uint16_t, int16_t, char*, uint16_t)':
xdrv_ir_send:99: error: 'irsend' was not declared in this scope
           if      (!strcmp_P(protocol,PSTR("NEC")))     irsend->sendNEC(data, bits);
                                                         ^
xdrv_ir_send:100: error: 'irsend' was not declared in this scope
           else if (!strcmp_P(protocol,PSTR("SONY")))    irsend->sendSony(data, bits);
                                                         ^
xdrv_ir_send:101: error: 'irsend' was not declared in this scope
           else if (!strcmp_P(protocol,PSTR("RC5")))     irsend->sendRC5(data, bits);
                                                         ^
xdrv_ir_send:102: error: 'irsend' was not declared in this scope
           else if (!strcmp_P(protocol,PSTR("RC6")))     irsend->sendRC6(data, bits);
                                                         ^
xdrv_ir_send:103: error: 'irsend' was not declared in this scope
           else if (!strcmp_P(protocol,PSTR("DISH")))    irsend->sendDISH(data, bits);
                                                         ^
xdrv_ir_send:104: error: 'irsend' was not declared in this scope
           else if (!strcmp_P(protocol,PSTR("JVC")))     irsend->sendJVC(data, bits, 1);
                                                         ^
xdrv_ir_send:105: error: 'irsend' was not declared in this scope
           else if (!strcmp_P(protocol,PSTR("SAMSUNG"))) irsend->sendSAMSUNG(data, bits);
                                                         ^
exit status 1
'IRsend' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

crankyoldgit, 2017-07-02:
@zuidwijk That **isn't** the same issue as the folk here are experiencing. Your problem (or the project/library you are using) is you need to update your code for the changes in v2.0+ of this library. You can read about it here: https://github.com/markszabo/IRremoteESP8266/wiki/Upgrading-to-v2.0

crankyoldgit, 2017-07-08:
@kryptoss Have you had any success?

crankyoldgit, 2017-07-14:
Marking this closed due to no response. It clearly is compiling for the majority of people. Willing to re-open if more details are provided.

closed 2017-07-14 by crankyoldgit