crankyoldgit/IRremoteESP8266 #257: Receiving different code for same buttons

amit996, 2017-06-29:
code i am using from example
/*
 * IRremoteESP8266: IRrecvDemo - demonstrates receiving IR codes with IRrecv
 * An IR detector/demodulator must be connected to the input RECV_PIN.
 * Version 0.1 Sept, 2015
 * Based on Ken Shirriff's IrsendDemo
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff, http://arcfn.com
 */

#include <IRrecv.h>

// an IR detector/demodulator is connected to GPIO pin 2
uint16_t RECV_PIN = 2;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup() {
  Serial.begin(9600);
  irrecv.enableIRIn();  // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    // print() & println() can't handle printing long longs. (uint64_t)
    // So we have to print the top and bottom halves separately.
    if (results.value >> 32)
      Serial.print((uint32_t) (results.value >> 32), HEX);
    Serial.println((uint32_t) (results.value & 0xFFFFFFFF), HEX);
    irrecv.resume();  // Receive the next value
  }
  delay(100);
}

crankyoldgit, 2017-06-30:
Thanks for reporting the issue. We'll need some more information from you in order to assist you.
e.g. What remote controller are you using to send with? What version of the library are you using etc etc.
Please have a read of this and see if it helps. (https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide)

I also strongly suggest you try the [IRrecvDumpV2.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) example in the meantime.

amit996, 2017-06-30:
Today i got a update for the library. Now its working fine. Thanks for the help.

closed 2017-06-30 by amit996