crankyoldgit/IRremoteESP8266 #262: Newbie question: Receiving a signal on one and passing it to an other pin

frekel, 2017-07-04:
Hi all,

Just bare with me (quite a IR newbie).....

I have a heater with a IR power on/off remote.  I want it turn off/on with a ESP8266.  Can someone point me into the right direction?

I have used the examples of this lib and tried this:

`/* IRremoteESP8266: IRsendDemo - demonstrates sending IR codes with IRsend.
 *
 * Version 1.0 April, 2017
 * Based on Ken Shirriff's IrsendDemo Version 0.1 July, 2009,
 * Copyright 2009 Ken Shirriff, http://arcfn.com
 *
 * An IR LED circuit *MUST* be connected to ESP8266 pin 4 (D2).
 *
 * TL;DR: The IR LED needs to be driven by a transistor for a good result.
 *
 * Suggested circuit:
 *     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
 *
 * Common mistakes & tips:
 *   * Don't just connect the IR LED directly to the pin, it won't
 *     have enough current to drive the IR LED effectively.
 *   * Make sure you have the IR LED polarity correct.
 *     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
 *   * Typical digital camera/phones can be used to see if the IR LED is flashed.
 *     Replace the IR LED with a normal LED if you don't have a digital camera
 *     when debugging.
 *   * Avoid using the following pins unless you really know what you are doing:
 *     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
 *     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
 *     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
 *   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
 *     for your first time. e.g. ESP-12 etc.
 */

#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

uint16_t powerOff[15] = {950,150, 150,150, 150,150, 250,150, 150,250, 150,350, 150,150, 350};  // UNKNOWN 85223F05  (8 bits)
uint16_t powerOn[15] = {950,150, 250,150, 150,250, 150,150, 150,150, 150,350, 150,150, 350};  // UNKNOWN 31464401  (8 bits)

IRsend irsend(4);  // An IR LED is controlled by GPIO pin 4 (D2)

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
  Serial.println("powerOff");
  irsend.sendRaw(powerOff, 15, 38);
  delay(2000);
  Serial.println("powerOn");
  irsend.sendRaw(powerOn, 15, 38);
  delay(2000);
  Serial.println("powerOff");
  irsend.sendRaw(powerOff, 15, 8);
  delay(2000);
  Serial.println("powerOn");
  irsend.sendRaw(powerOn, 15, 8);
  delay(2000);
  
}`

Thanks in advance,
Frank

crankyoldgit, 2017-07-05:
That looks like an unusual protocol for starters. I don't know many that are that short, but if that is what dumpv2 example says, it's probably right.

Your second set of poweron/poweroff's is probably wrong. The [third argument to sendRaw()](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L252) is the frequency. I don't know of any devices using 8kHz.

38kHz you've chosen before that is a typically good first choice., try 40, 36, 37, 56 etc. Have a look through the ir_Blah.cpp files for the enableIROut() lines for common frequencies.

How sure are you that your circuit is transmitting a good signal? e.g. Try following: https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#sending-an-ir-signal

crankyoldgit, 2017-09-10:
Just chasing up. How are you going with this? If we don't get a response back in 1-2 weeks, I'm going to close this due to inactivity.

crankyoldgit, 2017-09-21:
Closing due to inactivity. Feel free to re-open.

closed 2017-09-21 by crankyoldgit