crankyoldgit/IRremoteESP8266 #263: Getting unknown result on every decode receive?

vinevinz, 2017-07-05:
Using version 2.1, nodemcu v3

Hi, just curious if this result is normal or something.. I am kinda new to IoT and nodemcu/arduino stuff
I get every same unknown result on every remote I tried...

Trying to decode my panasonic ac remote here..

here is example of my result

Encoding  : UNKNOWN
Code      : 54920561 (18 bits)
Timing[35]: 
     + 450, - 500     + 450, - 500     + 450, - 450     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 450     + 450, -1350     + 450, -1350     + 450, -1350
     + 450, -1350     + 450, - 450     + 450, - 500     + 450, - 500
     + 450, -1350     + 450
uint16_t  rawData[35] = {450,500, 450,500, 450,450, 450,500, 450,500, 450,500, 450,500, 450,500, 450,450, 450,1350, 450,1350, 450,1350, 450,1350, 450,450, 450,500, 450,500, 450,1350, 450};  // UNKNOWN 54920561

or is it problem with my receiver?...

crankyoldgit, 2017-07-05:
Interesting. That's not a protocol that I instantly recognise. It doesn't appear to have a typical header (start sequence) either.

Air Conditioner remotes are typically complex beasts with respect to their code generation. They often send their command as multiple sequences (mini-separate-ish commands) which often confuses this library. In short, decoding a A/C remote is very tricky and typically not supported.
The rawData however should be fairly correct, if you want to try replay/resend it.

> I get every same unknown result on every remote I tried...

I'm more concerned about that statement. Typically for TV remotes etc, the library and the dumpV2 program should work and give a) consistent results, and b) decode results into the respective protocols.

I just checked the library (2.1) & example code (dumpv2) with a Samsung, Sherwood(NEC), & JVC remotes, and it decoded them as expected. What brand/device/etc remotes are you using that are giving inconsistent/unexpected results?

dham1012, 2017-07-05:
Having similar issues, only tested with Panasonic TV remote, I've had the issues since v2.03. v2.02 works and categorises as Panasonic. 2.02 throws  'IR code too long. Edit IRrecv.h and increase RAWBUF' though

vinevinz, 2017-07-06:
Forgive me for my laziness to read the troubleshooting page, and oh, I tried with other AC remote too. I should've stated that hehe.

Actually, I manage to get it work to decode my Panasonic AC remote which size of the array is sooo big [439] considering the samsung TV remote I tried only have size of [9]. You just need to increase RAWBUF size up to whatever you think suitable up to 65535. I increase mine to 1000, working perfectly fine. However, I think using value close to the raw data array size is better, save some memory.

I don't know why you stated that max RAWBUF is only 255 while you define it as uint16 which could store value up to 65535.

crankyoldgit, 2017-07-06:
It was only a maximum of 255 out it would crash before v2. Can you point
out for me where the comment is saying the old me value is? It needs to be
updated. FYI the available memory on the ESP 8266 is quite small. A bigger
buffer is okay for a user to change but as a library we get used in other
projects so we need to be minimal by default.

crankyoldgit, 2017-07-06:
FYI I think your Samsung remote reading is wrong. They are 32 bit normally,
so a real size if about 67±

vinevinz, 2017-07-06:
Oh well probably I forgot what the value is or something wrong with my samsung remote, only tried it once or two, it said samsung lol.

here : https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#problems-decoding
1. Air conditioner protocols tend to be quite large, regularly larger than the size specified in RAWBUF by default. Try upping it to the current maximum of 255.

Ah, I see, so v2 is new... unfortunately I was new here too :(
Yeah, I understand that, library need to fit all users need. But I'm new on C++ too so no comment on that 👍

crankyoldgit, 2017-07-06:
Thanks. I'll update that when I get home.

crankyoldgit, 2017-07-06:
I've updated the wiki page to remove the limit, and suggested 512 when decoding A/C remotes.
And wow, 439 .. I think that's a record. That is indeed huge.

Was there anything else? Or have we covered everything and can close this issue?

vinevinz, 2017-07-06:
Woah looks like my AC is breaking new record :p
I think that is all, please close it. Thanks for the help!!

Here you go, maybe you want to take a look :D 

POWER ON: 
uint16_t  rawData[439] = {3550,1700, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,1300, 500,1300, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,10000, 3550,1700, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,1300, 500,1300, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,1300, 500,1300, 500,1300, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,1300, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,1300, 500,400, 500,400, 500,400, 500,1300, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,1300, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,1300, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,400, 500,1300, 500,400, 500,400, 500,1300, 500};  // PANASONIC 400407200000

closed 2017-07-06 by vinevinz