crankyoldgit/IRremoteESP8266 #269: How to IR sendRAW using IRremoteESP8266

mocheffendi, 2017-07-08:
Dear Markszabo,

I use IRremoteESP8266 library, I have success using irRecvDumpV2, to receive Samsung AC code, but i don't know to to send IR with IR SendRAW 

Thanks

crankyoldgit, 2017-07-08:
@mocheffendi Have a look at [this](https://github.com/markszabo/IRremoteESP8266/commit/f44e755afee44aa130446671f9c2027207debfb8) for an [example](https://raw.githubusercontent.com/markszabo/IRremoteESP8266/f44e755afee44aa130446671f9c2027207debfb8/examples/IRsendDemo/IRsendDemo.ino).

That should probably do what you want. Just change the values to meet your requirements. The frequency is most likely 38kHz.

mocheffendi, 2017-07-09:
for LG TV sendRaw() is working but for Samsung Air Conditioner it is not working.

crankyoldgit, 2017-07-10:
I suggest you try following the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). You probably need to supply some more information if you need further help.

mocheffendi, 2017-07-10:
@crankyoldgit, 

I changed rawbuff to 512 and it worked flawlessly

cheer

crankyoldgit, 2017-07-11:
Glad to hear it.

closed 2017-07-11 by crankyoldgit