crankyoldgit/IRremoteESP8266 #272: sendGC for Samsung robot doesn't work with 2.1.0

1technophile, 2017-07-09:
### Version/revison of the library used
2.1.0
1.2.0

### Expected behavior
sendGC with an array:
1.2.0 OK (same hardware setup)
2.1.0 KO (same hardware setup)

sendNEC with a decimal :
551489775
1.2.0 OK (same hardware setup)
2.1.0 OK (same hardware setup)

### Actual behavior
tested with IRGCSendDemo

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Not applicable

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
```
uint16_t Samsung_power_toggle[71] = {
    38000,1,1,171,171,21,64,21,21,21,21,21,21,21,21,21,21,21,21,21,64,21,64,21,21,21,21,21,21,21,21,21,21,21,21,21,64,21,21,21,21,21,21,21,64,21,21,21,21,21,21,21,21,21,64,21,64,21,64,21,21,21,64,21,64,21,64,21,64,21,1114};

IRsend irsend(14);  // An IR LED is controlled by GPIO pin 4 (D2)

void setup() {
  irsend.begin();
  Serial.begin(115200);
}

void loop() {
  Serial.println("Toggling power");
  irsend.sendGC(Samsung_power_toggle, 71);
  delay(10000);
}
```

#### Circuit diagram and hardware used (if applicable)
Not applicable (working with sendNEC)

### I (have/have not) followed the steps in the [Troubleshooting Guide]
I have

crankyoldgit, 2017-07-10:
@1technophile I think this is a duplicate of #266 which I discovered and [fixed](https://github.com/markszabo/IRremoteESP8266/commit/839ad501269e22d536b46d511c175e8608d8339c) 2 days ago.

I'm waiting on a couple of other pull requests to be reviewed/merged and then I'll push a new version. ETA a couple of days. In the meantime, this is fixed in 'master' branch now.

crankyoldgit, 2017-07-10:
@1technophile I should have added, please let me know if the _master_ does or doesn't fix your problem. Cheers.

1technophile, 2017-07-10:
@crankyoldgit I confirm it's corrected into the master branch.
Thanks and sorry for the inconvenience

closed 2017-07-10 by 1technophile

crankyoldgit, 2017-07-11:
@1technophile No worries. Thanks for the report and for testing/confirming it. Always best to be sure.

crankyoldgit, 2017-07-11:
@1technophile FYI. I just pushed [v2.1.1](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.1.1) which contains the discussed fix.

1technophile, 2017-07-11:
Thanks, tested and integrated to the OpenMQTTGateway.