crankyoldgit/IRremoteESP8266 #277: IRremoteESP8266 not detected for board NodeMCU 1.0

Bimblis, 2017-07-14:
### Version/revison of the library used
Last version (I think its 2.1)

### Expected behavior
The library should be detected with board NodeMCU 1.0

### Actual behavior
Library not detected

### Steps to reproduce the behavior
Select NodeMCU 1.0 board and install the library

#### Example code used
#include <IRremoteESP8266.h> //The color indicates it is not detected by this board

crankyoldgit, 2017-07-14:
Can you explain this in more detail please, or in a different way. I'm not sure what you mean.

Bimblis, 2017-07-15:
[image]

Bimblis, 2017-07-15:
Seems like the Library IRremoteESP8266 is not being detected.

Bimblis, 2017-07-15:
Ok, nevermind. Just #include <IRsend.h> was missing

crankyoldgit, 2017-07-15:
Yep. Exactly. https://github.com/markszabo/IRremoteESP8266/wiki/Upgrading-to-v2.0

closed 2017-07-24 by crankyoldgit