crankyoldgit/IRremoteESP8266 #281: Trying to RAW Dump a Kabel Deutschland Receiver

atc1441, 2017-07-23:
Hi i am tring to receive the raw ir codes from a german "Kabel Deutschland" Receiver
i use library version 2.0.0 

with IRrecvDumpV2.ino it only outputs me raw data with 7 messages or so.
i tried to higher the rawbuf but that didnt help.

i think the problem is that the ir code has a feew 89728 micros delays.

here is one working raw dump:

`uint16_t  lauter[100] = {448, 276, 150, 285, 164, 613, 163, 447, 162, 613, 164, 445, 164, 776, 167, 278, 163, 280, 163, 280, 162, 611, 168, 444, 163, 612, 164, 277, 168, 447, 157, 282, 165, 276, 165, 89729, 451, 272, 161, 279, 164, 612, 164, 445, 165, 612, 164, 445, 164, 777, 165, 278, 165, 278, 164, 277, 167, 610, 166, 443, 166, 609, 167, 278, 166, 444, 165, 274, 169, 278, 159, 89728, 453, 266, 168, 276, 165, 612, 164, 445, 164, 612, 164, 444};
`

**if i use this code:** 
`#define maxLen 85
unsigned int lastmill;
unsigned int interval=3000;
boolean senden = false;

volatile  unsigned int irBuffer[maxLen]; //stores timings - volatile because changed by ISR
volatile unsigned int x = 0; //Pointer thru irBuffer - volatile because changed by ISR

void setup() {
  Serial.begin(115200); 
  attachInterrupt(2, rxIR_Interrupt_Handler, CHANGE);
}

void loop() {

  if ((x>85)) { 
    Serial.println();
    Serial.print(F("IR Code: (")); //dump raw header format - for library
    Serial.print((x - 1));
    Serial.print(F(") "));
    detachInterrupt(0);//stop interrupts & capture until finshed here
    for (int i = 1; i < x; i++) { //now dump the times
      //if (!(i & 0x1)) Serial.print(F("-"));
      Serial.print(irBuffer[i] - irBuffer[i - 1]);
      Serial.print(F(", "));
    }
    x = 0;
    Serial.println();
    Serial.println();
    attachInterrupt(0, rxIR_Interrupt_Handler, CHANGE);//re-enable ISR for receiving IR signal
    lastmill=millis();
    senden=false;
  }

}

void rxIR_Interrupt_Handler() {
  if (x > maxLen) return; //ignore if irBuffer is already full
  if (micros() - irBuffer[x-1]>100000) x=0;
  irBuffer[x++] = micros(); //just continually record the time-stamp of signal transitions

}`

i get the right pattern and i can send it later with SendRaw(bla,85,39);

but i want to get it over the library. 

Maybe someone has an idea :)

atc1441, 2017-07-23:
Sorry for double post. 

the hex code seems to be nokia32

on this website it can be decoded http://irdb.tk/decode/

crankyoldgit, 2017-07-23:
First, I recommend upgrading to v2.1.1 of the library. A whole load of bugs we found and fixed w.r.t. receiving since 2.0.0.

Next, You are pretty much correct, the issue is that long gap of about 89ms. The library tends to think that a gap of over 15ms is the end of a command/message.

The updated example code in IRrecvDumpV2 might help you. It saves the decode to a second buffer so it can start hunting for another code as soon as possible, rather than after it has printed out the result.
It _might_ be able to pick up the second half of the code.

If it doesn't, you can also try to adjust the timeout value TIMEOUT_MS (in milli-seconds) at https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L33 to a larger value than your gap. That's what triggers the 'stop listening' and ends a command/message sequence.

FYI, don't use the current _master_ branch at the moment, I'm waiting on a Pull Request (#278) to be reviewed which fixes something I broke when updating/improving capture.

crankyoldgit, 2017-07-23:
The code captured looks like an RC-MM variant.

I'll try to re-create it to see why the library didn't decode it

atc1441, 2017-07-23:
Thank you for the fast answer
i looked and i was allready using v2.1.1 tought it was the older one.

i allready tested to expand the gab of 15ms to 150 but it didn't changed anything.

i will try a feew other things. 

thank you

atc1441, 2017-07-23:
soafter a feew tests if i increase the TIEMOUT_MS to 100ms it works to get the irCode if i hold the remote faar away about 1meter. 
all other test where made with the remote in a range of 20cm.

But if i try to send it the receiver don't recognises it.
i think that is because the rawData from IRrecvDumpV2 is rounding the numbers.

this is what i got now: 
`uint16_t  rawData[107] = {500,300, 200,300, 200,650, 200,450, 200,650, 200,450, 200,750, 200,300, 200,300, 200,300, 200,650, 200,450, 200,650, 200,300, 200,450, 200,300, 200,300, 200,89750, 500,300, 200,300, 200,650, 200,500, 200,650, 200,450, 200,800, 200,300, 200,300, 200,300, 200,650, 200,450, 200,650, 200,300, 200,450, 200,300, 200,300, 200,89750, 500,300, 200,300, 200,650, 200,450, 200,650, 200,450, 200,750, 200,300, 200,250, 250,650, 200,650, 200,450, 200,650, 200,300, 200,450, 200,300, 200,300, 200};  `

and this is the Working code for it:
`uint16_t  rawData[107] = {448, 276, 150, 285, 164, 613, 163, 447, 162, 613, 164, 445, 164, 776, 167, 278, 163, 280, 163, 280, 162, 611, 168, 444, 163, 612, 164, 277, 168, 447, 157, 282, 165, 276, 165, 89729, 451, 272, 161, 279, 164, 612, 164, 445, 165, 612, 164, 445, 164, 777, 165, 278, 165, 278, 164, 277, 167, 610, 166, 443, 166, 609, 167, 278, 166, 444, 165, 274, 169, 278, 159, 89728, 453, 266, 168, 276, 165, 612, 164, 445, 164, 612, 164, 444};
`

crankyoldgit, 2017-07-23:
Interesting. I just created a unit test based on your data, it seems it should decode as RC-MM. The timing is very tight with that protocol. i.e. A slight bit of gitter causes will confuse the library.

The code in your original message is a 32-bit RC-MM message with a value of 0x26702610.
i.e. `sendRCMM(0x26702610, 32);` should do the trick.

crankyoldgit, 2017-07-23:
Yes, the rounding is an issue.
Try reducing USECPERTICK to 10 or something smaller than the default of 50.

crankyoldgit, 2017-07-23:
Also, 89729 is to big a value for a uint16_t. The maximum value is 65535 for a uint16_t.

atc1441, 2017-07-23:
Ok sendRCMM(0x26702610, 32); did it, it is turning the volume up as expected.
nice :) 
it will eat alot less space not to have the raw data, how did you decode it? i will then do it for every button.

could i use uint64_t instead? arduino is not throwing an error on this but is sending the ir as needed.

crankyoldgit, 2017-07-23:
I decoded it by doing this: https://github.com/markszabo/IRremoteESP8266/commit/0cc0d70585c47c8215410764c8505d9d7d3e1c89

crankyoldgit, 2017-07-23:
sendRaw() doesn't understand uint64_t's in it's array. Sorry. Something we could fix of course. i.e. support uint32_t's.
Typically a gap larger than 10,000 usecs means it's another command. Especially one as large as 89,000 usecs.

crankyoldgit, 2017-07-23:
If you want, you can also decode by hand.
Basically you need to convert the even values in the raw data, using the following method:
A value of around 277 means '00' in binary.
444 means '01' in binary.
611 means '10' in binary.
777 means '11' in binary.
Start from the 4th entry to skip the header. Most significant bits are sent first.
That should give you a 32 bit (binary) value at the end. That's what you need to send to `sendRCMM()`

atc1441, 2017-07-23:
Ah okay i tried to decode it by hand and did understand it :) 
`RawData= 285, 164, 613, 163, 447, 162, 613, 164, 445, 164, 776, 167, 278, 163, 280, 163, 280, 162, 611, 168, 444, 163, 612, 164, 277, 168, 447, 157, 282, 165, 276, 165, 89729
Converted=00        10        01        10        01        11        00        00        00        10        01        10        00        01        00              

Binary= 00100110011100000010011000010000 In Hex ->  26702610`

But i did not unterstand how https://github.com/markszabo/IRremoteESP8266/commit/0cc0d70585c47c8215410764c8505d9d7d3e1c89 this works. is it a function? how can i use it, i hope that is not a to noob question :D

i really like your will to help. thank you so much

crankyoldgit, 2017-07-23:
It's not a noob question.
If you've got a linux machine you can run/compile the unit tests.
e.g.
```
cd test
make
./ir_RCMM_test
```

If you change the contents of that pull request/diff I created with the raw capture you've been collecting. It should try to decode for you, but will _fail_ due to a different code/button having a different value. It will tell you what value it got, and what it was expecting. Thus you can work out what it is.

Again, if you don't have a access to a linux box, that won't help you.

crankyoldgit, 2017-07-23:
So, I tried to reverse the data from that site you mentioned: http://irdb.tk/capture/ (a.k.a. http://irdb.tk/decode/)

Their result is:
`Protocol Nokia32, device 38, subdevice 112, OBC 16, misc = X=38`

Which matches up with what we saw.
i.e.
38 = 0x26
112 = 0x70
16 = 0x10
38 = 0x26

So, you can work out from that how to potentially get our value we need for `sendRCMM(0x26702610, 32)` too.

crankyoldgit, 2017-07-23:
FYI, I just saw your earlier edited comment:
> Function: 13 Volume+ 0000 0073 0000 0012 000F 000A 0006 000A 0006 0016 0006 0010 0006 0016 0006 0010 0006 001C 0006 000A 0006 000A 0006 000A 0006 0016 0006 0010 0006 0016 0006 000A 0006 0010 0006 000A 0006 000A 0006 0C9D

That's a looks like a pronto code. You can use `sendPronto()` to do that. There is even example code to show you how, if you get stuck.

atc1441, 2017-07-23:
Great so now there are 4 solutions :D 
i have a linux mashine running but that is to much digging for me now.

i used this function to convert the irRecv code into RCMM
`#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

uint16_t RECV_PIN = 2;

IRrecv irrecv(RECV_PIN);

decode_results results;  // Somewhere to store the results
irparams_t save;         // A place to copy the interrupt state while decoding.

void setup() {
  // Status message will be sent to the PC at 115200 baud
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
  irrecv.enableIRIn();  // Start the receiver
}

void decodercmm(decode_results *results) {
 if (results->rawlen > 60){
  Serial.print("irsend.sendRCMM(0x");  
  String decoded="";   
for (int i = 4; i < 36; i++) {
    if (results->rawbuf[i] * USECPERTICK < 300) decoded += ("00");
    else if (results->rawbuf[i] * USECPERTICK < 500)decoded += ("01");
    else if (results->rawbuf[i] * USECPERTICK < 650)decoded += ("10");
    else if (results->rawbuf[i] * USECPERTICK < 810)decoded += ("11");
    i++;
  }
  Serial.print(decoded);
  Serial.println(");"); 
  Serial.println(""); 
}}

// The repeating section of the code
//
void loop() {
  // Check if the IR code has been received.
  if (irrecv.decode(&results, &save)) {
    decodercmm(&results);  
  }
}`

But i still need to convert the String to HEX 

Since there are so much other options now i dont think i need this anymore

crankyoldgit, 2017-07-24:
If you have a look at decodeRCMM() you'll see similar code, and how to do the math. ;-)

It looks like you've now got this sorted. I'm going to close this issue now. Please feel free to re-open if you think it isn't.

atc1441, 2017-07-24:
Yes this helped a lot. 
I still don't found decodeRCMM() where is it ? 
Is made now a complete decodeRCMM and it is working as it should.

Thank you a lot.

crankyoldgit, 2017-07-24:
https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_RCMM.cpp#L102

atc1441, 2017-07-24:
Ah ok i got it :)

atc1441, 2017-07-24:
So here are now my final solution:
in https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_RCMM.cpp#L40
change the `RCMM_TOLERANCE 10U to 25U`
and in https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L32
change `USECPERTICK   50U to 5U`

that is working like every other protocol for me.

as an adition you can add `case RCMM:       Serial.print("RCMM");        break;` 
after line https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L61
in the IRecvDumpV2

I think i understand the library much more after that digging :)

have a nice day that will be my last comment in this topic :D

crankyoldgit, 2017-07-24:
Thanks. I'll look at incorporating that into the library.

crankyoldgit, 2017-09-10:
The master branch has a much finer resolution of the timings now.
Marking this closed.

closed 2017-09-10 by crankyoldgit