crankyoldgit/IRremoteESP8266 #285: LG AC control

xpokor02, 2017-07-25:
### Version/revison of the library used v2.1

### Expected behavior
Should control AC unit

### Actual behavior
I used similar code from ardruino example (works fine with nano), unfortunately doesn't work with ‎WEMOS D1 mini

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
I am going to test it.

### Steps to reproduce the behavior
I am trying to turn on AC, It doesn't work.
 I am using suggested schematic and tested that led is blinking with mobile camera

#### Example code used
it is based on IRremote example
https://github.com/z3t0/Arduino-IRremote/blob/master/examples/LGACSendDemo/LGACSendDemo.ino

#### Circuit diagram and hardware used (if applicable)
suggested one in wiki

### I (have/have not) followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide)
I did, but not all of them yet, connection is fine.

### Other useful information
_More information is always welcome. Be verbose._
It would be great if someone could adjusted LG AC example from original library to b working

xpokor02, 2017-07-25:
Hi,
dump from remote control:
Encoding  : LG
Code      : 88C0051 (28 bits)
Timing[59]: 
     +8400, -4150     + 550, -1550     + 550, - 450     + 600, - 450
     + 600, - 450     + 600, -1500     + 600, - 450     + 550, - 500
     + 550, - 500     + 550, -1500     + 600, -1500     + 600, - 450
     + 600, - 450     + 600, - 450     + 550, - 500     + 550, - 500
     + 550, - 500     + 550, - 450     + 600, - 450     + 600, - 450
     + 600, - 450     + 600, - 450     + 600, -1500     + 550, - 500
     + 550, -1550     + 550, - 450     + 600, - 450     + 600, - 450
     + 600, -1500     + 550
unsigned int  rawData[59] = {8400,4150, 550,1550, 550,450, 600,450, 600,450, 600,1500, 600,450, 550,500, 550,500, 550,1500, 600,1500, 600,450, 600,450, 600,450, 550,500, 550,500, 550,500, 550,450, 600,450, 600,450, 600,450, 600,450, 600,1500, 550,500, 550,1550, 550,450, 600,450, 600,450, 600,1500, 550};  // LG 88C0051
unsigned int  data = 0x88C0051;

dump from WEMOS D1 mini:
Encoding  : LG
Code      : 88C0051 (28 bits)
Timing[59]: 
     +7850, -3800     + 650, -1600     + 600, - 500     + 650, - 450
     + 650, - 500     + 600, -1600     + 650, - 450     + 650, - 500
     + 600, - 500     + 600, -1600     + 650, -1600     + 600, - 500
     + 600, - 500     + 650, - 450     + 650, - 500     + 600, - 500
     + 650, - 450     + 650, - 450     + 650, - 500     + 600, - 500
     + 650, - 450     + 650, - 500     + 600, -1600     + 650, - 450
     + 650, -1600     + 600, - 500     + 600, - 500     + 650, - 450
     + 650, -1600     + 600
unsigned int  rawData[59] = {7850,3800, 650,1600, 600,500, 650,450, 650,500, 600,1600, 650,450, 650,500, 600,500, 600,1600, 650,1600, 600,500, 600,500, 650,450, 650,500, 600,500, 650,450, 650,450, 650,500, 600,500, 650,450, 650,500, 600,1600, 650,450, 650,1600, 600,500, 600,500, 650,450, 650,1600, 600};  // LG 88C0051
unsigned int  data = 0x88C0051;

Hope it helps solve it.
Thanks

crankyoldgit, 2017-07-26:
The codes are very similar/the same. Only slight timing differences. You could try adjusting/tweaking the values at the top of ir_LG.cpp if I recall correctly. That should allow you to tune it closer to your actual remote. Let us know what values you end up with.

xpokor02, 2017-07-26:
Sure I will try it, but I don't know logic behind it.
Can I change LG_TICK to something small like 50 and adjust all multipliers to higher number?
Thanks

crankyoldgit, 2017-07-26:
You can do anything you like. It's only code. ;-)
Changing LG_TICK up or down should make subtle adjustments. It's probably all it will take to make it work for you.

xpokor02, 2017-07-26:
ok so I changed values to following ones and it works:

#define LG_TICK                     50U
#define LG_HDR_MARK_TICKS            160U
#define LG_HDR_MARK                 (LG_HDR_MARK_TICKS * LG_TICK)
#define LG_HDR_SPACE_TICKS            80U
#define LG_HDR_SPACE                (LG_HDR_SPACE_TICKS * LG_TICK)
#define LG_BIT_MARK_TICKS             12U
#define LG_BIT_MARK                 (LG_BIT_MARK_TICKS * LG_TICK)
#define LG_ONE_SPACE_TICKS            32U
#define LG_ONE_SPACE                (LG_ONE_SPACE_TICKS * LG_TICK)
#define LG_ZERO_SPACE_TICKS           11U
#define LG_ZERO_SPACE               (LG_ZERO_SPACE_TICKS * LG_TICK)
#define LG_RPT_SPACE_TICKS            45U
#define LG_RPT_SPACE                (LG_RPT_SPACE_TICKS * LG_TICK)
#define LG_MIN_GAP_TICKS             795U
#define LG_MIN_GAP                  (LG_MIN_GAP_TICKS * LG_TICK)
#define LG_MIN_MESSAGE_LENGTH_TICKS 2161U
#define LG_MIN_MESSAGE_LENGTH       (LG_MIN_MESSAGE_LENGTH_TICKS * LG_TICK)
#define LG32_HDR_MARK_TICKS           90U
#define LG32_HDR_MARK               (LG32_HDR_MARK_TICKS * LG_TICK)
#define LG32_HDR_SPACE_TICKS          89U
#define LG32_HDR_SPACE              (LG32_HDR_SPACE_TICKS * LG_TICK)
#define LG32_RPT_HDR_MARK_TICKS      179U
#define LG32_RPT_HDR_MARK           (LG32_RPT_HDR_MARK_TICKS * LG_TICK)

Thanks

xpokor02, 2017-07-26:
Here is code I used to control my AC it may be useful for others.

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_LG.h>

IRsend irsend(14);  // An IR LED is controlled by GPIO pin 14 (D5)

const int AC_TYPE  = 1;
// 0 : TOWER
// 1 : WALL
//

int AC_HEAT = 1;
// 0 : cooling
// 1 : heating

int AC_POWER_ON    = 0;
// 0 : off
// 1 : on

int AC_AIR_ACLEAN  = 0;
// 0 : off
// 1 : on --> power on

int AC_TEMPERATURE = 24;
// temperature : 18 ~ 30

int AC_FLOW        = 0;
// 0 : low
// 1 : mid
// 2 : high
// if AC_TYPE =1, 3 : change
//

const int AC_FLOW_TOWER[3] = {
  0, 4, 6};
const int AC_FLOW_WALL[4]  = {
  0, 2, 4, 5};

unsigned long AC_CODE_TO_SEND;

void ac_send_code(unsigned long code)
{
  Serial.print("code to send : ");
  Serial.print(code, BIN);
  Serial.print(" : ");
  Serial.println(code, HEX);

  irsend.sendLG(code, 28);
}

void ac_activate(int temperature, int air_flow, int heat)
{

  AC_HEAT = heat;
  int AC_MSBITS1 = 8;
  int AC_MSBITS2 = 8;
  int AC_MSBITS3 = 0;
  int AC_MSBITS4 ;
  if ( AC_HEAT == 1 ) {
    // heating
    AC_MSBITS4 = 4;
  } 
  else {
    // cooling
    AC_MSBITS4 = 0;
  }
  int AC_MSBITS5 = temperature - 15;
  int AC_MSBITS6 ;
[... 216 lines cut ...]
          delay(100);
        }
        ac_activate(AC_TEMPERATURE, AC_FLOW, AC_HEAT);
      }
      break;
    default:
      if ( 18 <= a && a <= 30 ) {
        if ( 0 <= b && b <= 2 ) {
          ac_activate(a, b, AC_HEAT);
        }
      }
    }
  
  delay(100);
  Serial.println("TEMPERATURE");
  Serial.println(AC_TEMPERATURE);
  Serial.println("AC_FLOW");
  Serial.println(AC_FLOW);
  Serial.println("AC_HEAT");
  Serial.println(AC_HEAT);
  Serial.println("AC_POWER_ON");
  Serial.println(AC_POWER_ON);
    
  
}

closed 2017-07-31 by xpokor02

baegopooh, 2018-05-31:
I'm trying to do same thing-using this library on esp8266 to control lg ac- and this timing does not work(version 2.4). or, it works on transmit side(sendLG funtion), but does not work on receiver side.

I looked along, and found more strict timing based on oscciloscope, it says " The transmission starts with a pulse of 9 ms followed by 4.5ms space. Since the IR receiver is actively high, a 9ms pulse from a transmitter got translated into a 9 ms ‘dip’ in the receiver. A logical ‘1’ is started by a 562.5us pulse followed by 1.6875ms space (3 times 562.5us). A logical ‘0’ is represented by a 562.5us pulse followed by a 562.5us space."
So i Changed code based on above timing. and it works much better on transmitting signal and receiving signal.

#define LG_TICK                       56U
#define LG_HDR_MARK_TICKS            161U
#define LG_HDR_MARK                  (LG_HDR_MARK_TICKS * LG_TICK)
//9016us. close to 9ms
#define LG_HDR_SPACE_TICKS            80U
#define LG_HDR_SPACE                 (LG_HDR_SPACE_TICKS * LG_TICK)
//4480us. close to 4.5ms
#define LG_BIT_MARK_TICKS             10U
#define LG_BIT_MARK                  (LG_BIT_MARK_TICKS * LG_TICK)
//560us. close to 562.5us
#define LG_ONE_SPACE_TICKS            30U
//1680us. close to 1687.5us.
#define LG_ONE_SPACE                 (LG_ONE_SPACE_TICKS * LG_TICK)
#define LG_ZERO_SPACE_TICKS           10U
#define LG_ZERO_SPACE                (LG_ZERO_SPACE_TICKS * LG_TICK)