crankyoldgit/IRremoteESP8266 #286: Samsung Remote Gets RC6 Decoded but with 0bits

atc1441, 2017-07-25:
Hi here i am again.

 i tried to dump the codes from my samsung remote but they get recognised wrong. 

i know that the button "Power OnOff" should be 0x1000C ,20.
With the older library 1... i god the codes and after the update to 2.1.1 they don't get recognised anymore

i reinstalled the library to make shure nothing is changed.

`Encoding  : RC6
Code      : 0 (0 bits)
Timing[39]: 
     +135350, -42600     +24250, -42650     +24150, -20350     +24250, -20400
     +68650, -64700     +24250, -20400     +24250, -20300     +25550, -19100
     +25500, -18900     +24300, -20350     +24200, -20400     +24100, -20350
     +24350, -20350     +24150, -20400     +24200, -20300     +24200, -20400
     +46450, -20400     +24100, -42650     +25550, -19050     +24050
uint16_t  rawData[39] = {135350,42600, 24250,42650, 24150,20350, 24250,20400, 68650,64700, 24250,20400, 24250,20300, 25550,19100, 25500,18900, 24300,20350, 24200,20400, 24100,20350, 24350,20350, 24150,20400, 24200,20300, 24200,20400, 46450,20400, 24100,42650, 25550,19050, 24050};  // RC6 0
uint64_t  data = 0x0;
` 
The code comes from the IRrecvDumpV2

atc1441, 2017-07-25:
Okay that one is srtrange. after a rebot of my computer everything works... this can be closed...

the timing looks way of it the code i posted

closed 2017-07-25 by atc1441

crankyoldgit, 2017-07-26:
yeah. There is no way those values are normal.