crankyoldgit/IRremoteESP8266 #290: support for XMP protocol

teohell, 2017-07-27:
I have a remote (italian MySkyHD) which is working with XMP protocol, transmitting 71 bits (checked with AnalysIR)
This is an example raw (for number 0) dumped with AnalysIR

{240,1016,244,1152,236,748,244,2788,240,1288,244,1288,244,1148,244,880,240,12956,240,1016,244,2652,236,748,244,740,236,748,240,744,244,740,240,744,244,79738,244,1012,244,1148,244,744,244,2788,240,1288,244,1288,244,1148,240,880,240,12960,232,1024,236,1428,244,1836,240,744,244,736,240,884,236,744,244,740,248}

currently I'm sending the code by splitting in two and letting Arduino do the 79738 delay in microseconds.

By the way, the code is not dumped correctly because the library only dumps a part (18 bits), and they also are not corresponding (this may be due to the sensor at 38Khz but the signal seems to be at 36Khz)
For number 0 I get the following dump:

[350,950, 350,1100, 350,650, 350,2700, 350,1200, 350,1200, 350,1100, 350,800, 350,12850, 350,950, 350,2600, 350,650, 350,650, 350,650, 350,650, 350,650, 350,650, 350]

would it be possible for the library to send XMP protocol, and also dumping it correctly?

Matteo

crankyoldgit, 2017-07-28:
Potentially. Do you have a link or documentation for the XMP protocol at all?

Looking at the data you've provided.
It appears that the ~79000 value is the gap between a repeat of the 'main' signal.
and the 'main' signal is in fact two messages, itself separated by the ~12900 value/gap.

It looks like a space-encoded protocol. The bit mark's appear fixed at around 240ms.
There however seem to be several rough values for the 'space's. i.e. ~750, ~1000, ~1200, ~1500, ~1800, ~2700

crankyoldgit, 2017-07-28:
Some limited reading I've done on XMP indicates it has 16 possible values for the space value.
In theory, sending a XMP code should be do-able, decoding one, on the other hand, is going to be a nightmare with respect to accurate timing/acquisition. The margin of error is probably too small for us to do a decode.

crankyoldgit, 2017-07-28:
FYI, the 79738 delay is going to confuse our library. It interprets a gap larger than 15ms as a separate code/message.
Thus it will break the capture into two parts.
So, that's what you're seeing with the shorter (18 bit) message.

e.g.
> [350,950, 350,1100, 350,650, 350,2700, 350,1200, 350,1200, 350,1100, 350,800, 350,12850, 350,950, 350,2600, 350,650, 350,650, 350,650, 350,650, 350,650, 350,650, 350]

corresponds with (the first half of):
> {240,1016,244,1152,236,748,244,2788,240,1288,244,1288,244,1148,244,880,240,12956,240,1016,244,2652,236,748,244,740,236,748,240,744,244,740,240,744,244, ...

You could try messing with TIMEOUT_MS. e.g. setting it to 100, or around 13.

Setting it to 100 may allow you to capture it all in one go/message. Dropping it to 13 might give it more time to try decoding and then start the next capture.
(That's all assuming you are using the latest released version of the lib. i.e. 2.1.1)

crankyoldgit, 2017-07-28:
Note: The current resolution of the library (see USECPERTICK) is rather low (50ms), you may need to drop it to 10ms to get close enough timings needed for XMP based on my readings. However, doing that will screw up decoding of normal (other protocols), so you are on your own if changing it breaks other things. ;-)

teohell, 2017-07-28:
Also I think that sending the code as json with current library will fail because the 71 bytes exceed the expected data type buffer

crankyoldgit, 2017-07-28:
I think you are mistaken. It's not a 71 bit protocol. There are 71 _items_ in the AnalyizeIR array. That's not the same as 71 bits of information.
Plus we have ways to deal with larger protocols/bits. e.g. See some of the aircon remotes.

teohell, 2017-07-28:
but if I try to send the raw data straight away, it will fail, because the buffer of sendraw is uint16_t:

void IRsend::sendRaw(uint16_t buf[], uint16_t len, uint16_t hz)

crankyoldgit, 2017-07-28:
I'm not sure what you mean by you can't send it straight away.
This is how you use sendRaw. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino

teohell, 2017-07-28:
because uint16_t is maximum 65,535 and 79,000+ is out of data type

crankyoldgit, 2017-07-28:
Do a _0_ length mark. e.g.

Make it {..., 65535, 0, (79000 - 65535), ...}

crankyoldgit, 2017-07-28:
or split it into two arrays and do a delay(79) between the two sends.

teohell, 2017-07-28:
Using a lenght zero mark is a great idea, I'm currently splitting but using a double tick is a better and cleaner way! Thanks

crankyoldgit, 2017-07-29:
I'll add it to our dumping program to automatically do that, after you confirm it works of course. ;-)

teohell, 2017-09-04:
sorry for late reply but i was on holiday :o)

anyway today I tried by splitting the long delay in two halves and adding a zero lenght mark between the two, and - TADA!
it works :-)

crankyoldgit, 2017-09-05:
Thanks for the confirmation that the '0' encoding trick works. I've got pending code that will automatically rawdump using the zero-encoding for when those situations arise in future. That is, should produce an output that is always compatible with sendRaw etc.

Marking this issue closed. If you don't think it is, please feel free to reopen.

closed 2017-09-05 by crankyoldgit