crankyoldgit/IRremoteESP8266 #292: Error in Compiling the Example code's. Not compiling

santhosh8438, 2017-07-29:
### Version/revison of the library used
v2.1

### Expected behavior
1. I tried to compile IRGCSendDemo  
2. Copied the IRGCSendDemo.ino to src folder then i changed the libraries to internal by making them as a string like this
 #include "IRremoteESP8266.h"
#include "IRrecv.h"
#include "IRutils.h"
3. Then i restarted Arduino IDE and gave compile but it's shown the ERROR message and it's not compiled

### Actual behavior
I just wanted to compile the program , I gave compile but it's not compiled

#### Output of raw data from IRGCSendDemo.ino (if applicable)
Arduino: 1.8.3 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 921600, 4M (3M SPIFFS)"

In file included from sketch\IRrecv.h:10:0,

                 from sketch\ir_JVC.cpp:5:

sketch\ir_JVC.cpp: In member function 'void IRsend::sendJVC(uint64_t, uint16_t, uint16_t)':

C:\Users\Sandy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token

 #define max(a,b) ((a)>(b)?(a):(b))

                  ^

sketch\ir_JVC.cpp:71:16: note: in expansion of macro 'max'

     space(std::max(JVC_MIN_GAP, JVC_RPT_LENGTH - usecs.elapsed()));

                ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

#### Example code used
/*
 * IRremoteESP8266: IRrecvDemo - demonstrates receiving IR codes with IRrecv
 * An IR detector/demodulator must be connected to the input RECV_PIN.
 * Copyright 2009 Ken Shirriff, http://arcfn.com
 * Example circuit diagram:
 *  https://github.com/markszabo/IRremoteESP8266/wiki#ir-receiving
 * Changes:
 *   Version 0.2 June, 2017
 *     Changed GPIO pin to the same as other examples.
 *     Used our own method for printing a uint64_t.
 *     Changed the baud rate to 115200.
 *   Version 0.1 Sept, 2015
 *     Based on Ken Shirriff's IrsendDemo Version 0.1 July, 2009
 */

#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include "IRremoteESP8266.h"
#include "IRrecv.h"
#include "IRutils.h"

// An IR detector/demodulator is connected to GPIO pin 14(D5 on a NodeMCU
// board).
uint16_t RECV_PIN = 14;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();  // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    // print() & println() can't handle printing long longs. (uint64_t)
    serialPrintUint64(results.value, HEX);
    Serial.println("");
    irrecv.resume();  // Receive the next value
  }
  delay(100);
}

#### Circuit diagram and hardware used (if applicable)
Nothing. I just compiled that's it.

### Other useful information
I have tried in other laptop's still having the same issues.

Thank you!!

crankyoldgit, 2017-07-29:
I noticed this in your report:

> C:\Users\Sandy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token

i.e. **2.0.0**

I think you maybe using an ancient version (Nov '15) of the esp8266 arduino core (https://github.com/esp8266/Arduino/releases/tag/2.0.0)

The latest (current) release is 2.3.0 https://github.com/esp8266/Arduino/releases/tag/2.3.0

Try following [these instructions](https://github.com/esp8266/Arduino#installing-with-boards-manager) to install it correctly.

crankyoldgit, 2017-07-29:
I'm not sure why you've needed to modify or move the example code to the _src_ directory.
If you've installed everything correctly, you should be able to select the _IRGCSendDemo_ from the examples menu and the Arduino IDE will create a new window/project/etc for it and it should just work.

I've tried this previously on a fresh install of the Arduino IDE on a Windows 10 PC, then installed the esp8266 core, via the board manager, and then installed the IRremoteESP8266 library via the library manager. It works.

We also have a continuous build process in Travis, and it would scream problems if there were compile errors.

The compiler error message, to me, is indicative of something odd or not-up-to-date in your arduino environment.

Have you checked via Board Manager and Library Manager, that **everything** is up-to-date?

I am curious as to what might be causing your problem. This is the second report of this issue. I'd love to find out what is causing it.

crankyoldgit, 2017-07-29:
Ref #255

crankyoldgit, 2017-07-30:
For what it is worth. I confirmed it compiles under Windows 10 (64 bit), Arduino IDE 1.8.3, esp8266 core 2.3.0, & IRremoteESP8266 2.1.1

santhosh8438, 2017-07-31:
Hi, I have tried the above steps and it Works!!

Thank you very much for the help!! 
Have a nice day.

closed 2017-07-31 by santhosh8438