crankyoldgit/IRremoteESP8266 #293: Denon IR-Remote not detected

SciLor, 2017-08-03:
### Version/revison of the library used
2.1.0

### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRRecv
2. Receive all commands
### Actual behavior
For ex Denon IR-Commands are not detected.
You may receive broken commands by putting a delay into the loop

e.g.
#### Output of raw data from IRrecvDumpV2.ino (if applicable)
Nothing

### Steps to reproduce the behavior
Use a Denon X3000 remote.

#### Example code used
IRrecvDumpV2.ino

#### Circuit diagram and hardware used (if applicable)
Standard one

crankyoldgit, 2017-08-03:
What output do you get?

crankyoldgit, 2017-08-03:
And what output do you get with your delay? And where did you put it?

SciLor, 2017-08-03:
Without delay I get nothing.
With delay I get protocol -1 and random numbers.

crankyoldgit, 2017-08-03:
How big a delay and where are you putting it?
What is the raw data is displaying?

crankyoldgit, 2017-08-03:
I.e. IRrecvDumpV2.ino is fairly verbose when it receives a signal even if it doesn't understand it.

SciLor, 2017-08-03:
I am using PIN D5 of my NodeMCU
Serial is slowed down to 9600.

Without delay there is no output. With a delay of delay(100)

I pressed the power button several times, different output each time:
```
Encoding  : UNKNOWN
Code      : EE4CAD61 (48 bits)
Timing[96]: 
     + 400, - 550     + 400, - 550     +1250, - 550     + 400, - 550
     +1250, - 500     + 400, - 550     +1250, - 500     + 400, - 550
     + 400, - 550     +1250, - 550     + 400, - 550     + 400, - 550
     +1250, - 500     +1250, - 550     + 400, - 550     + 400, - 550
     + 400, - 550     + 400, - 500     + 400, - 500     + 400, - 550
     + 400, - 500     + 400, - 550     +1250, - 550     + 400, - 550
     +1250, - 550     + 400, - 550     + 400, - 550     + 400, - 550
     +1250, - 550     + 400, - 550     +1250, - 550     + 400, - 550
     + 400, - 550     + 400, - 550     + 400, - 550     + 400, - 500
     + 400, - 500     + 400, - 550     + 400, - 550     + 400, - 550
     +1250, - 500     + 400, - 550     + 400, - 550     + 400, - 550
     +1250, - 550     + 400, - 550     + 400, - 500     + 400, - 500

uint16_t  rawData[96] = {400,550, 400,550, 1250,550, 400,550, 1250,500, 400,550, 1250,500, 400,550, 400,550, 1250,550, 400,550, 400,550, 1250,500, 1250,550, 400,550, 400,550, 400,550, 400,500, 400,500, 400,550, 400,500, 400,550, 1250,550, 400,550, 1250,550, 400,550, 400,550, 400,550, 1250,550, 400,550, 1250,550, 400,550, 400,550, 400,550, 400,550, 400,500, 400,500, 400,550, 400,550, 400,550, 1250,500, 400,550, 400,550, 400,550, 1250,550, 400,550, 400,500, 400,500 };  // UNKNOWN EE4CAD61

Encoding  : UNKNOWN
Code      : 581BC552 (20 bits)
Timing[40]: 
     +1250, - 500     + 400, - 500     +1250, - 550     + 400, - 550
     + 400, - 550     + 400, - 550     + 400, - 500     + 400, - 550
     + 400, - 550     + 400, - 550     + 400, - 550     + 400, - 500
     +1250, - 550     + 400, - 500     + 400, - 550     + 400, - 550
     +1250, - 550     + 400, - 550     + 400, - 550     + 400, - 500

uint16_t  rawData[40] = {1250,500, 400,500, 1250,550, 400,550, 400,550, 400,550, 400,500, 400,550, 400,550, 400,550, 400,550, 400,500, 1250,550, 400,500, 400,550, 400,550, 1250,550, 400,550, 400,550, 400,500 };  // UNKNOWN 581BC552

Encoding  : UNKNOWN
Code      : 5F446CEE (49 bits)
Timing[97]: 
     + 550, - 400     + 500, - 400     + 450, -1250     + 550, - 400
     + 500, -1250     + 550, - 400     + 500, -1250     + 550, - 400
     + 500, - 400     + 500, -1250     + 550, - 400     + 450, - 450
     + 500, -1250     + 500, -1250     + 500, - 450     + 500, - 400
     + 500, - 400     + 500, - 400     + 500, - 400     + 450, - 450
     + 450, - 450     + 450, - 450     + 500, -1250     + 500, - 450
     + 500, -1250     + 500, - 450     + 500, - 400     + 500, - 400
     + 500, -1250     + 500, - 450     + 500, -1250     + 500, - 450
     + 500, - 400     + 500, - 400     + 500, - 400     + 500, - 400
     + 500, - 400     + 500, - 400     + 450, - 450     + 500, - 450
     + 500, -1250     + 500, - 450     + 500, - 400     + 450, - 450
     + 500, -1250     + 550, - 400     + 500, - 400     + 500, - 400
     + 500
uint16_t  rawData[97] = {550,400, 500,400, 450,1250, 550,400, 500,1250, 550,400, 500,1250, 550,400, 500,400, 500,1250, 550,400, 450,450, 500,1250, 500,1250, 500,450, 500,400, 500,400, 500,400, 500,400, 450,450, 450,450, 450,450, 500,1250, 500,450, 500,1250, 500,450, 500,400, 500,400, 500,1250, 500,450, 500,1250, 500,450, 500,400, 500,400, 500,400, 500,400, 500,400, 500,400, 450,450, 500,450, 500,1250, 500,450, 500,400, 450,450, 500,1250, 550,400, 500,400, 500,400, 500};  // UNKNOWN 5F446CEE

Encoding  : UNKNOWN
Code      : D7DDF908 (49 bits)
Timing[98]: 
     +1650, - 550     + 400, - 550     + 400, - 550     +1250, - 550
     + 400, - 550     +1250, - 550     + 400, - 550     +1250, - 550
     + 400, - 500     + 400, - 550     +1250, - 550     + 400, - 550
     + 400, - 550     +1250, - 550     +1250, - 550     + 400, - 550
     + 400, - 550     + 400, - 500     + 400, - 500     + 400, - 550
     + 400, - 550     + 400, - 550     + 400, - 550     +1250, - 550
     + 400, - 550     +1250, - 550     + 400, - 500     + 400, - 500
     + 400, - 550     +1250, - 500     + 400, - 550     +1250, - 500
     + 400, - 550     + 400, - 550     + 400, - 550     + 400, - 550
     + 400, - 500     + 400, - 500     + 400, - 500     + 400, - 550
     + 400, - 500     +1250, - 500     + 400, - 550     + 400, - 550
     + 400, - 550     +1250, - 550     + 400, - 550     + 400, - 550
     + 400, - 500
uint16_t  rawData[98] = {1650,550, 400,550, 400,550, 1250,550, 400,550, 1250,550, 400,550, 1250,550, 400,500, 400,550, 1250,550, 400,550, 400,550, 1250,550, 1250,550, 400,550, 400,550, 400,500, 400,500, 400,550, 400,550, 400,550, 400,550, 1250,550, 400,550, 1250,550, 400,500, 400,500, 400,550, 1250,500, 400,550, 1250,500, 400,550, 400,550, 400,550, 400,550, 400,500, 400,500, 400,500, 400,550, 400,500, 1250,500, 400,550, 400,550, 400,550, 1250,550, 400,550, 400,550, 400,500 };  // UNKNOWN D7DDF908

```

crankyoldgit, 2017-08-04:
Thanks for the output. That helps to see where the problem may be. I'll take a look at that shortly.

However, can you please tell me:

a) _Where are you putting the delay?_ That doesn't make any sense to me why that would make it work/not work etc. hence I want to work out where you are adding it.

b) _Why are you running at 9600 baud?_ I *strongly* recommend running at a faster baud rate. The time it takes to print out the decode/capture is lost time it has to capture/decode incoming messages. The faster it can print, means more time it has to decode & capture.

crankyoldgit, 2017-08-04:
I've had a look at the dumps. They look very odd. So odd I suspect something is very wrong with your capture setup.

I dug up the pronto codes for the Denon X3000 an it correctly decodes it as a Denon protocol/message.
(https://usa.denon.com/us/downloads/ProductDownloads?FileName=Denon%20Master%20IR%20Hex.xls)

Have you been able to decode signals (consistently) using/from another remote source?

SciLor, 2017-08-04:
I am using a TSOP4838 on my NodeMCU connected with 3.3V, GND and D5.
I prevoiusly used it with an Arduino and it worked fine.
I placed the delay at the bottom of the loop() function.

Other IR-Remotes work fine for me.
Example
`
Encoding  : RC6
Code      : C80521059 (36 bits)
Timing[59]: 
     +2700, - 800     + 550, - 450     + 500, - 400     + 550, - 800
     + 500, - 850     +1450, - 800     + 500, - 450     + 500, - 450
     + 500, - 400     + 500, - 450     + 550, - 400     + 550, - 400
     + 500, - 450     + 950, - 850     + 950, - 850     + 550, - 400
     + 950, - 850     + 550, - 450     + 500, - 450     + 500, - 400
     + 950, - 850     + 550, - 450     + 500, - 450     + 500, - 400
     + 550, - 450     + 900, - 850     + 950, - 400     + 550, - 800
     + 500, - 450     + 950
uint16_t  rawData[59] = {2700,800, 550,450, 500,400, 550,800, 500,850, 1450,800, 500,450, 500,450, 500,400, 500,450, 550,400, 550,400, 500,450, 950,850, 950,850, 550,400, 950,850, 550,450, 500,450, 500,400, 950,850, 550,450, 500,450, 500,400, 550,450, 900,850, 950,400, 550,800, 500,450, 950};  // RC6 C80521059
uint32_t  address = 0xC805210;
uint32_t  command = 0x59;
uint64_t  data = 0xC80521059;
`

crankyoldgit, 2017-08-04:
It really shouldn't need a delay there at all. I re-tested it on a NodeMCU board earlier today, without a delay. It works as expected.

The earlier captures you reported all look very unusual. They almost look like a panasonic code, which denon also use, but they are very malformed. e.g. typically, after you exclude the header and footer, either the _mark_ values stay roughly the same (i.e. the odd, or positive values depending on what you look at) for a SPACE encoded message, or the opposite, the _space_ values (even, negative values etc) will stay roughly the same for a MARK encoded message. In your original examples, there was a mix of space AND mark encoding in the same message. That's very atypical.

Again, I strongly recommend using the example code (IRrecvDumpV2.ino) in its unaltered form, and use a baud-rate of 115200.

If that still doesn't provide a more consistent capture, you mentioned you have an arduino. Try capturing the X3000's messages using the IRremote (for arduino) library and give us the raw output here. We may then have a shot at trying to get the library to decode what ever signal your remote is sending.

That all said, given you seem to be able to capture other protocols okay with the setup on the ESP8266, it does seem to indicate that your circuit/IR hardware is okay.

SciLor, 2017-09-11:
Sorry, I havn't seen your message. I will test it asap

crankyoldgit, 2017-10-10:
Gentle reminder.

crankyoldgit, 2017-10-14:
Marking closed due to inactivity. Please re-open if you have an update.

closed 2017-10-14 by crankyoldgit