crankyoldgit/IRremoteESP8266 #296: Daikin ECONO mode

skubrik, 2017-08-04:
Hi,
this is a request...
i'm using this awesome library to control daikin ac.
Is there the possibility to add the "ECONO" mode to the daikin library?
The ECONO mode is a function that sets a limit for power consumption...and it would be very helpful.
Thanks!

crankyoldgit, 2017-08-05:
You'll probably need @sillyfrog to help with this one.
As I don't have one of these units, nor do I have access to one, you will have to find & capture the raw data to work out what bits are needed to be flipped to turn that mode on.

That, or find someone else who has implemented it on ardunio or similar and we can then borrow from their code.

I'm happy to try to help if you can gather the needed resources, but without that, I can't help.

skubrik, 2017-08-05:
Using IRrecvDumpV2 from examples i logged these values:

Econo ON:

```
Encoding  : UNKNOWN
Code      : B3E2CB06 (6 bits)
Timing[11]: 
     + 550, - 400     + 500, - 400     + 500, - 400     + 500, - 450
     + 450, - 450     + 500
uint16_t  rawData[11] = {550,400, 500,400, 500,400, 500,450, 450,450, 500};  // UNKNOWN B3E2CB06

Encoding  : UNKNOWN
Code      : BCA77DE0 (15 bits)
Timing[30]: 
     + 400, - 500     + 450, - 500     + 450, - 450     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 450     +1350, - 450
     +1350, - 450     +1350, - 450     + 450, - 500     +1300, - 450
     + 450, - 500     +1300, - 450     +1350, - 500
uint16_t  rawData[30] = {400,500, 450,500, 450,450, 450,500, 450,500, 450,500, 450,450, 1350,450, 1350,450, 1350,450, 450,500, 1300,450, 450,500, 1300,450, 1350,500 };  // UNKNOWN BCA77DE0

Encoding  : UNKNOWN
Code      : 4A1B17EB (15 bits)
Timing[30]: 
     +1300, - 500     + 450, - 450     +1350, - 500     +1300, - 500
     +1300, - 500     + 400, - 450     + 450, - 500     + 450, - 500
     +1300, - 500     +1300, - 450     +1300, - 500     +1300, - 500
     +1300, - 450     + 450, - 450     +1350, - 450
uint16_t  rawData[30] = {1300,500, 450,450, 1350,500, 1300,500, 1300,500, 400,450, 450,500, 450,500, 1300,500, 1300,450, 1300,500, 1300,500, 1300,450, 450,450, 1350,450 };  // UNKNOWN 4A1B17EB

```

ECONO OFF:

```
Encoding  : UNKNOWN
Code      : B3E2CB06 (6 bits)
Timing[11]: 
     + 450, - 450     + 450, - 500     + 450, - 500     + 450, - 500
     + 400, - 500     + 450
uint16_t  rawData[11] = {450,450, 450,500, 450,500, 450,500, 400,500, 450};  // UNKNOWN B3E2CB06

Encoding  : UNKNOWN
Code      : BCA77DE0 (15 bits)
Timing[30]: 
     + 450, - 450     + 450, - 450     + 500, - 450     + 450, - 450
     + 500, - 450     + 500, - 450     + 500, - 400     +1400, - 400
     +1400, - 400     +1400, - 400     + 500, - 450     +1350, - 450
     + 500, - 450     +1350, - 450     +1350, - 450
uint16_t  rawData[30] = {450,450, 450,450, 500,450, 450,450, 500,450, 500,450, 500,400, 1400,400, 1400,400, 1400,400, 500,450, 1350,450, 500,450, 1350,450, 1350,450 };  // UNKNOWN BCA77DE0

Encoding  : UNKNOWN
Code      : 4A1B17EB (15 bits)
Timing[30]: 
     +1350, - 450     + 500, - 450     +1350, - 450     +1350, - 450
     +1350, - 450     + 450, - 450     + 500, - 450     + 500, - 450
     +1350, - 450     +1350, - 450     +1350, - 450     +1350, - 450
     +1350, - 450     + 500, - 400     +1400, - 400
uint16_t  rawData[30] = {1350,450, 500,450, 1350,450, 1350,450, 1350,450, 450,450, 500,450, 500,450, 1350,450, 1350,450, 1350,450, 1350,450, 1350,450, 500,400, 1400,400 };  // UNKNOWN 4A1B17EB
```

Could this be helpful?

skubrik, 2017-08-05:
I forgot the AC hardware:
remote: arc452a3
internal unit: FTXS25G2V1B

crankyoldgit, 2017-08-05:
I don't think that data is really useful, but there is some good news. The next version (current version in the 'master' tree) should be able to capture A/C units better. If you can, you can try to download the master version, and modify the new recv dump example [here](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L30)

crankyoldgit, 2017-08-05:
FYI, we are looking for either one message with about 250+ items in raw data, or two messages with about 128 items in raw data.

skubrik, 2017-08-05:
Like these?

Econo ON:

```
Encoding  : UNKNOWN
Code      : 8629B8F5 (292 bits)
Timing[583]: 
   +   430, -   414,    +   460, -   440,    +   432, -   412,    +   430, -   416, 
   +   458, -   414,    +   456, - 25426,    +  3514, -  1748,    +   436, -  1310, 
   +   434, -   440,    +   434, -   410,    +   462, -   410,    +   432, -  1338, 
   +   408, -   440,    +   434, -   414,    +   458, -   412,    +   460, -   384, 
   +   460, -  1310,    +   436, -   412,    +   460, -  1310,    +   434, -  1308, 
   +   436, -   412,    +   460, -  1310,    +   436, -  1310,    +   434, -  1308, 
   +   436, -  1310,    +   436, -  1310,    +   436, -   388,    +   456, -   412, 
   +   460, -  1310,    +   434, -   414,    +   458, -   414,    +   428, -   416, 
   +   458, -   438,    +   432, -   438,    +   434, -   386,    +   460, -   412, 
   +   458, -   414,    +   460, -   410,    +   432, -   414,    +   458, -  1310, 
   +   436, -   414,    +   456, -  1310,    +   436, -   412,    +   460, -   386, 
   +   458, -   414,    +   458, -  1310,    +   434, -  1310,    +   434, -   412, 
   +   462, -   412,    +   432, -   412,    +   460, -   412,    +   460, -   438, 
   +   404, -   440,    +   434, -   438,    +   432, -   414,    +   458, -   412, 
   +   432, -   414,    +   460, -   438,    +   434, -   414,    +   430, -   416, 
   +   456, -   440,    +   434, -   410,    +   462, -   386,    +   458, -  1310, 
   +   434, -  1308,    +   436, -  1310,    +   436, -   438,    +   434, -  1308, 
   +   436, -   438,    +   434, -  1310,    +   434, -  1310,    +   434, - 35000, 
   +  3516, -  1746,    +   436, -  1310,    +   434, -   412,    +   432, -   412, 
   +   460, -   412,    +   460, -  1310,    +   436, -   412,    +   430, -   440, 
   +   434, -   412,    +   460, -   410,    +   460, -  1312,    +   434, -   412, 
   +   434, -  1310,    +   436, -  1308,    +   436, -   438,    +   434, -  1308, 
   +   436, -  1310,    +   434, -  1310,    +   436, -  1308,    +   436, -  1308, 
   +   436, -   438,    +   432, -   412,    +   432, -  1338,    +   406, -   416, 
   +   458, -   414,    +   458, -   414,    +   428, -   416,    +   458, -   412, 
   +   460, -   410,    +   462, -   386,    +   460, -   438,    +   434, -   412, 
   +   460, -   438,    +   404, -   440,    +   434, -  1310,    +   436, -   438, 
   +   434, -   412,    +   460, -   386,    +   458, -   438,    +   434, -  1310, 
   +   436, -   438,    +   432, -   438,    +   406, -   440,    +   434, -  1308, 
   +   436, -  1310,    +   434, -   414,    +   458, -   410,    +   462, -  1308, 
   +   436, -  1310,    +   436, -   386,    +   458, -  1310,    +   436, -   412, 
   +   460, -  1310,    +   436, -  1310,    +   434, -  1310,    +   436, -   414, 
   +   460, -   384,    +   460, -   440,    +   432, -  1308,    +   436, -   438, 
   +   434, -  1308,    +   436, -  1310,    +   436, -   438,    +   406, -  1338, 
   +   406, -   440,    +   434, - 34996,    +  3516, -  1746,    +   436, -  1310, 
   +   434, -   412,    +   462, -   414,    +   484, -   358,    +   462, -  1310, 
   +   436, -   410,    +   460, -   410,    +   460, -   414,    +   430, -   414, 
   +   458, -  1308,    +   436, -   438,    +   434, -  1308,    +   436, -  1310, 
   +   436, -   412,    +   432, -  1338,    +   406, -  1338,    +   408, -  1338, 
   +   408, -  1338,    +   406, -  1338,    +   406, -   414,    +   460, -   440, 
   +   432, -  1310,    +   434, -   438,    +   432, -   414,    +   432, -   438, 
   +   434, -   414,    +   460, -   438,    +   406, -   412,    +   462, -   412, 
   +   460, -   438,    +   434, -   412,    +   434, -   412,    +   460, -   438, 
   +   434, -   416,    +   426, -   440,    +   434, -   438,    +   434, -   438, 
   +   430, -   416,    +   434, -   438,    +   434, -   410,    +   460, -  1310, 
   +   436, -   410,    +   434, -   410,    +   462, -  1308,    +   436, -  1308, 
   +   462, -  1284,    +   436, -   436,    +   434, -   410,    +   432, -   440, 
   +   434, -   412,    +   462, -   436,    +   458, -   388,    +   434, -  1308, 
   +   436, -  1310,    +   436, -   412,    +   460, -   410,    +   462, -   438, 
   +   406, -   438,    +   434, -   440,    +   456, -   416,    +   404, -   438, 
   +   436, -   438,    +   434, -   438,    +   458, -   388,    +   434, -   440, 
   +   432, -   436,    +   460, -   414,    +   406, -   440,    +   434, -   414, 
   +   484, -  1284,    +   436, -   438,    +   434, -  1308,    +   436, -   412, 
   +   432, -   438,    +   434, -   412,    +   462, -   438,    +   404, -   414, 
   +   460, -   438,    +   458, -   414,    +   434, -   410,    +   436, -   410, 
   +   462, -   436,    +   434, -   410,    +   434, -   438,    +   434, -   438, 
   +   434, -   438,    +   404, -   440,    +   434, -   438,    +   434, -   436, 
   +   436, -  1308,    +   436, -  1310,    +   460, -   360,    +   484, -   390, 
   +   458, -   438,    +   434, -   438,    +   406, -   440,    +   434, -   438, 
   +   434, -   436,    +   436, -   410,    +   458, -   414,    +   460, -   412, 
   +   436, -  1308,    +   436, -  1308,    +   436, -   410,    +   434, -   438, 
   +   434, -   438,    +   432, -   438,    +   406, -   438,    +   434, -   438, 
[... 69 lines cut ...]
   +   432, -   438,    +   432, -   440,    +   432, -   412,    +   432, -   440, 
[... cut ...] ,  432, 388,  458, 440,  432, 440,  408, 464,  404, 442,  432, 440,  432, 440,  404, 442,  430, 442,  430, 440,  432, 1312,  434, 414,  432, 440,  432, 440,  432, 440,  402, 442,  432, 1310,  432, 1314,  432, 442,  430, 442,  432, 412,  430, 442,  430, 442,  432, 440,  402, 442,  430, 442,  432, 442,  430, 414,  432, 440,  430, 442,  432, 412,  430, 440,  430, 442,  430, 442,  402, 442,  430, 1312,  434, 438,  430, 442,  432, 412,  408, 464,  432, 1312,  432, 440,  432};  // UNKNOWN 8FA82C91
```

sillyfrog, 2017-08-05:
I found capturing the Daikin IR codes on the ESP very poor... ie: just didn't work. The reason (I believe) is the size of the RAM needed, as the Daikin codes have a very long delay towards the start, I think it just requires to many samples to be of value (but I'm still pretty new to all of this, so maybe completely wrong).

The work I did is from another library that I restructured a bit, and made it work with the ESP (see comments at the top of the file). To get it going I did mess with some of the bits to see if I could figure it out, but just enough for my system (it does not have an Eco mode unfortunately). I also wasn't able to get the timer going at the time (which is something I do want to do - one day).

I would suggest trying the setQuiet (and maybe setPowerful) bits (assuming you don't have them on yours), as it may well be the same bits - no harm at least trying :).

PS: Apologies for leaving in the decodeDaikin method there - it never worked and I meant to strip it out before the PR.

PPS: Since typing the above I see a lot of chatter, maybe you can get it going! Good luck! 👍

crankyoldgit, 2017-08-05:
@skubrik Yep. Much much better.

What will be also useful is knowing what all the other settings are on those messages.

skubrik, 2017-08-05:
Eh...it's not quite easy
Daikin remote sends every settings when it speaks with unit.
I've found some info with google...but i don't know if they can be helpful...

http://rdlab.cdmt.vn/project-2013/daikin-ir-protocol
https://github.com/blafois/Daikin-IR-Reverse

i think it uses same bit of powerful and quite..that i've also have in this remote.

crankyoldgit, 2017-08-05:
I know. A/C remotes are a bug-bear to workout. This isn't going to be easy.

sillyfrog, 2017-08-05:
Based on https://github.com/adafruit/Raw-IR-decoder-for-Arduino/commit/8b13ad962fd48175784d194aa3693d10881f33d8 - there is a chance that setting bit 3 on byte 24 (Aux2) would do what you need. See setPowerful/getPowerful for a reference (this is working with byte 21, but should be easy to modify from there).

skubrik, 2017-08-05:
I'm not a great coder... lol
i'm reading:

setAux(DAIKIN_POWERFUL);
and these constants:
#define DAIKIN_POWERFUL       0b00000010
#define DAIKIN_SILENT         0b00100000

I think to have to change the constant value...but with what?

crankyoldgit, 2017-08-05:
Based soley on what @sillyfrog just said, this *might* work (not tested or compiled etc)

```
void IRDaikinESP::setEcon(bool econ) {
  if (econ)
    daikin[24] |=  0x4;
  else
    daikin[24] &= 0xFB;
  checksum();
}
```

skubrik, 2017-08-05:
Nothing...the unit has strange behavior..fan stops and start flipping like when poweroff...

skubrik, 2017-08-05:
Also i'm discovering that powerful doesn't work...but quiet works.

sillyfrog, 2017-08-05:
I have put a couple of updates here based on the links you provided above https://github.com/sillyfrog/IRremoteESP8266 . It does look like Powerful is working, if you can test Econo mode, I would be keen to know if that does work, we can then push to the main library.

skubrik, 2017-08-05:
@sillyfrog Powerful seems working now..but econo still no.

crankyoldgit, 2017-09-10:
Any luck or progress?

crankyoldgit, 2017-09-21:
ping

sillyfrog, 2017-09-21:
Hi, have been traveling. I have ordered some cables so I can try and reverse engineer what maybe happening, but it’s a few weeks before they arrive and I’ll need to find some time once they do. 
Cheers.

PSKAdam, 2017-09-25:
Hi, @skubrik, could you share your code here? I am wondering why your DAIKIN IR signal is so long compared to mine as I am also duplicating the IR code from a DAIKIN air-con remote control. I intend to use it to control my DAIKIN air-con with feather huzzah esp8266 board. Here is the raw data I received when I decoded the IR code of ON/OFF button:

Timing[135]: 
   +  9808, -  9776,    +  9810, -  9776,    +  4614, -  2482,    +   388, -   338, 
   +   388, -   922,    +   388, -   926,    +   388, -   348,    +   388, -   926, 
   +   384, -   348,    +   386, -   344,    +   386, -   356,    +   388, -   338, 
   +   388, -   922,    +   388, -   348,    +   388, -   344,    +   388, -   348, 
   +   384, -   352,    +   384, -   348,    +   384, -   938,    +   384, -   348, 
   +   384, -   926,    +   384, -   352,    +   384, -   348,    +   382, -   926, 
   +   384, -   352,    +   384, -   348,    +   384, -   356,    +   388, -   338, 
   +   388, -   346,    +   388, -   348,    +   382, -   348,    +   388, -   348, 
   +   384, -   352,    +   384, -   348,    +   384, -   356,    +   388, -   338, 
   +   388, -   348,    +   388, -   348,    +   384, -   348,    +   384, -   926, 
   +   388, -   348,    +   384, -   348,    +   384, -   360,    +   384, -   344, 
   +   382, -   348,    +   388, -   350,    +   384, -   348,    +   384, -   926, 
   +   384, -   352,    +   384, -   346,    +   384, -   356,    +   388, -   344, 
   +   384, -   348,    +   388, -   926,    +   388, -   348,    +   388, -   348, 
   +   384, -   930,    +   388, -   346,    +   388, -   356,    +   384, -   344, 
   +   384, -   352,    +   384, -   930,    +   384, -   926,    +   386, -   348, 
   +   384, -   352,    +   384, -   348,    +   384, -   934,    +   384

But somehow I just couldn't get any response from my air-con as it wasn't turned on at all. Does anyone has any idea what goes wrong? What should I do? Please advise...

crankyoldgit, 2017-09-25:
[sidenote]
If you update the `TIMEOUT` value to be larger than the standard 15ms, it often results in captures of multiple 'IR' messages. Setting that value to be larger than 15, is often needed to handle some A/C remotes as they send so much data at once, often splitting it over multiple IR messages.
the dumpv2 program really isn't designed to handle back-to-back messages like some A/Cs send. Hence some times you need to tweak that option and do some manual decoding etc.

sillyfrog, 2017-10-22:
@skubrik I have been trying to get this going with no luck - the issue for me is capturing the signal outside the ESP - I have been trying the sound card trick with zero success, probably because all my computers have a combined headphone/microphone port, and it does not recognise the impedance of the "mic" (IR receiver), I have tried matching a real mic and a bunch of things without success :(

I have also not had much luck recording signals from the Daikin reliably (using an ESP), the timing was always off and I could not get a good sync going. To record the signals (because the RAM was often too small to get everything (I think) when working with a long delay), I have been using this tool that I wrote: https://github.com/sillyfrog/RFreplayESP
It says it's for RF (which is what I did it for originally), but it also (sort of) works for IR signals - it works by writing to flash rather than RAM to accommodate very long signals (comparatively).

Sorry I can't be more help at help at the moment - but I'll update things should I ever get a chance to get this going further.

crankyoldgit, 2017-10-22:
@sillyfrog Have you tried with the updated dumpV2 and library per chance? Some recent changes I added (in the last couple of months) should make raw capturing A/C remotes a bit easier. At least I would have thought they would, as that was the intention.
If you have, then yeah, I think we've kind of exhausted what we can realistically do for now without more data etc.

sillyfrog, 2017-10-22:
@crankyoldgit I'll have to check, but I don't think I have - I'll give this a go when I get a chance, thanks!

sillyfrog, 2017-10-30:
@crankyoldgit I'm just trying the V2 dump library now, and it's working great! I'll hopefully get the decode function working in the next week (just figuring out what it needs to do). Thanks for your help!

crankyoldgit, 2017-10-30:
No worries. Glad it's working for you.

sillyfrog, 2017-11-05:
I have done a major update/ rewrite here: https://github.com/sillyfrog/IRremoteESP8266

This includes decode support and now sends the full output as per the remote control.

Please give this a go and tell me what you think and then we can merge into the main repo.

crankyoldgit, 2017-11-05:
@sillyfrog Wow. That's a lot of effort and a huge improvement in Daikin capability. Congrats and thanks. I hope it works for @skubrik 
I've been working on supporting decoding of Kelvinator A/C codes and if you take a look at the [branch](https://github.com/markszabo/IRremoteESP8266/compare/kelvinator_recv), you can see how I'm thinking of extending the `decode_results` class to accommodate the larger A/C state data. Thoughts/feedback welcome.

sillyfrog, 2017-11-05:
@crankyoldgit Thanks for taking a look, I have also already noticed a number of things I can improve as well looking at the final diff.

There is a common header (of which I have broken out one group), but I could probably do more, the issue is it's useful for debugging, but it'll take up precious RAM most of the time. (Not sure of the solution just yet... thinking something like `PROGMEM`, but it may not be required as the compiler will not include this code if it is not used in the sketch).

It'll be a few days before I can take a look at this again, but I will for sure and I'll take a good look at that branch you mentioned.

If you're interested, I also have a very simple Python script I have been using to look at the differences in the binary output (it reads directly from the serial port so you can press the button on the remote and see what changed) so I can workout what each bit does. [comparebits.py.zip](https://github.com/markszabo/IRremoteESP8266/files/1443932/comparebits.py.zip) (as a ZIP because that's how GitHub wanted it.)

Cheers.

crankyoldgit, 2017-11-05:
I too, am a little concerned about the footprint of the library. I'm thinking of having the A/C decode routines off by default (via #defines) as I don't think most people will want it normally. i.e. It's not like someone will want to use their A/C remote to control their ESP8266/home automation setup .. I think.

Feel free to add your comparebits.py to the tools/ directory. Others may find it useful.

For the record, I'm really impressed you worked out the timer stuff for Daikin. Mega-kudos. I gave up with the Kelvinator protocol. Reverse engineering it looked too hard for me, plus I could do _timer_ modes via the microcontroller/home automation systems, rather than doing it in the original hardware etc. I know _exactly_ how much effort there is in reversing these A/C protocols. Again, kudos for what you've done. I _wish_ manufacturers publicly documented these damn protocols. ;-)

crankyoldgit, 2017-11-10:
@skubrik Have you had a chance yet to test out @sillyfrog's test branch?

crankyoldgit, 2017-11-22:
FYI @skubrik , we believe we have a fix for this with the almost full re-write of Daikin code.
It will be available when v2.3 gets officially released. You can try it out before release by using this branch: https://github.com/markszabo/IRremoteESP8266/tree/v2.3.0-dev

It may be a week or more before the official release comes out.
As it is on track for release, I'm going to mark this issue closed for now. Feel free to re-open if you find an issue with v2.3.0-dev branch etc.

closed 2017-11-22 by crankyoldgit