crankyoldgit/IRremoteESP8266 #297: unknown type name 'IRsend'

MehranMazhar, 2017-08-11:
#IRremoteESP8266 Library  Version:   v2.1.0 
Hi .
I have a c file and i want to use IRsend library its in 

#include <IRremoteESP8266.h>
#include <IRsend.h>
IRsend irsend(10);  // An IR LED is controlled by GPIO pin 4 (D2)
[image]

crankyoldgit, 2017-08-14:
Frankly, I'm confused. That's not expected as you seem to have coded it correctly from what limited info I can see in the screenshot. Can you share/host your code somewhere so I can download it and try it myself?
The only suggestions I can make before that is to make sure you're using the latest release of the library and esp8266 board library. e.g. Go into the arduino library manager and check it's using 2.1.1 of IRremoteESP8266 and in the Board Manager, that it's using 2.3.0 of the ESP8266 config.

crankyoldgit, 2017-08-14:
Oh wait, I think it _may be_ because it's a C file, not a C++ file.

You should probably be modifying https://github.com/Azure/azure-iot-arduino/blob/master/examples/simplesample_http/simplesample_http.ino not https://github.com/Azure/azure-iot-arduino/blob/master/examples/simplesample_http/simplesample_http.c

I don't think *C* has the concept of classes. *.c files mean C, \*.cpp/\*.ino/etc mean C++/normal arduino.

crankyoldgit, 2017-08-25:
Did you have any luck?

santhosh8438, 2017-08-31:
Go to Sketch -> include library -> manage libraries in your arduino IDE and search for IRremoteESP8266 and install it properly...
Then restart your IDE ... Then try the codes which is in the examples ........!!
You may be have a luck !!
 Have a great day

MehranMazhar, 2017-09-02:
@crankyoldgit  thanks for the tips . its worked for me .

closed 2017-09-02 by MehranMazhar