crankyoldgit/IRremoteESP8266 #298: Exception 29: StoreProhibited

erpalma, 2017-08-14:
### Version/revison of the library used
IRremoteESP8266, Arduino ESP8266, PubsubClient all master @ 14th Aug

### Actual behavior
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 7 results
0x40209b4d: IRrecv::copyIrParams(irparams_t volatile*, irparams_t*) at /home/.../Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp line 168 (discriminator 2)

while in main loop @ line: if (irrecv.decode(&results, &save))

I'll continue to investigate.

erpalma, 2017-08-14:
My fault, I didn't initialize save.rawbuf!

crankyoldgit, 2017-08-14:
Mental note to self: I should probably add a method to the class that creates a save buffer for a user to avoid this issue.

closed 2017-08-25 by crankyoldgit