crankyoldgit/IRremoteESP8266 #302: irRcvPin only works on pin 0

BattlefieldHou, 2017-09-09:
using library v2.1.1
Adafruit Huzzah 8266 and Linknode D1 (both exhibit same problem)

I'm initializing my IR pin to 14, but can not receive any data on that pin.  If I change it to pin 0, I can receive data, but only if I open the Serial monitor inside the Arduino IDE.  I've tried other pins, but only can receive data from pin 0 with serial monitor open.  If the serial monitor is closed, I can tell the code is running, but it ignores any data from the IR receiver.  The IR receiver is a 56KHZ TSOP.

This happens when using the example code: IRRecvDump.ino and my own code.

Since I can't find anyone else reporting this problem, I am guessing it has to be something I'm doing.  The TSOP is wired correctly since I get the data if I have it on pin 0 with the serial window open.

The HUZZAH 8266 is configured with PIN 0 / PIN 14 going to pin 3 of the TSOP, +3.3V to PIN 2, GND to PIN 1.  I'm not initializing or using the WiFi.  Taking the IRRecvDump.ino without changes other than changing the RECV_PIN to either 0 or 14.

I'm trying to understand why it doesn't receive on pin 14, and why pin 0 requires the serial console to be open before it will start receiving data.   I've been working this on/off for a few weeks.  Tried to flush the serial, different pins, the order that I init serial port, don't init serial port or use it at all, so on and so on. 

#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

// an IR detector/demodulator is connected to GPIO pin 2
uint16_t RECV_PIN = 0;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();  // Start the receiver
}

crankyoldgit, 2017-09-10:
I just confirmed the library does work as expected (on a nodemcu board) using different gpio pins. e.g. D5/GPIO-14 *and* D6/GPIO-12.

Sorry, I don't have a Huzzah board to test against.

FYI, you should use IRRecvDumpV2.ino. It's better in almost all ways.
Also, D3/GPIO-0 is probably not a good choice, as it controls the Flash mode of the ESP8266 chip and needs to be held to the correct state while booting, but hey, you seem to have it working okay.

I'm going to assume you are using the Arduino IDE/program. You might want to check the board manager is up to date.

Try D3 instead of 0, just in case there is something odd with the board.:
```
uint16_t RECV_PIN = D3;
```

If you do find out what was wrong, please update this so someone else might find the solution.

BattlefieldHou, 2017-09-10:
Thanks for confirming.  I know it has to be something on my side since I couldn't find anyone else talking about it.  I was able to get it to work on D4 late last night on the Huzzah 8266 feather board.  I know I had tried that pin before.  I was excited to get it to work even though I feel like I didn't change anything code or hardware wise.  But that's what we all say...  

This afternoon, I decided to try and recreate the issue and move it back to pin 14.  Somehow that works now.  I turned all my debug code back on which prints out to serial, and it still works.  At the moment, I'm not sure what to think.  I have been fighting this issue for a couple of weeks and kept putting it off since I had so much other code to write.  Now I'm worried that it will come back when I put things into play.  

Well, can't chase something that isn't there at the moment.  I appreciate you confirming that things worked.  I had already decided that it was something on my side.  No way a pin wouldn't work and no one had mentioned it on any earlier bug issues.  If it shows up again, I'll update this post in case others see something similar.  I didn't update libraries (already running the latest) and I was already using the latest board manager files so those didn't update either.  I was able to get it to work with the example code IRRecvDump.ino last night.  I was then able to get it to work in my code after moving to pin D4/4.  Now I'm back on Pin 14 and all is working.

closed 2017-09-10 by BattlefieldHou

crankyoldgit, 2017-09-10:
FWIW, I had a similar issue when trying to reproduce your reported issue. My problem however was it wasn't compiling and uploading the code I had changed. It was uploading old code. Just a thought/possiblity.

BattlefieldHou, 2017-09-11:
Interesting theory. Maybe I changed something that caused a complete recompile. Perhaps I had older bin files using different board manager boards. Tough to test theory. I'll update if it recurs. Thanks for the insight.