crankyoldgit/IRremoteESP8266 #305: Issue - sendRAW using IRremoteESP8266

kapiloberoi, 2017-09-11:
Hi,

I face a unique issue while using the IRGCSendDemo (Version 0.2 June, 2017). While sending multiple RAW codes to my DTH receiver - only the last code is sent and all in between are skipped. While looking with camera I can see that the LED is working absolutely fine and transmits. But why only does only the last code is accepted by the receiver? The raw codes works fine and if I modify the .ino to send only one raw code - the receiver receives it fine indicating that the code are perfectly fine. What am I missing here ? Tried changing the nodemcu board but still same issue :(

`#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

// Airtel DTH Remote Code - Put channel number 451

//key Four
uint16_t  KeyFour[36] = {417, 278, 167, 278, 167, 611, 167, 444, 167, 611, 167, 278, 167, 278, 167, 278, 167, 278, 167, 278, 167, 611, 167, 444, 167, 611, 167, 278, 167, 278, 167, 444, 167, 278, 167, 65535};

//key Five
uint16_t  KeyFive[36] = {417, 278, 167, 278, 167, 611, 167, 444, 167, 611, 167, 278, 167, 278, 167, 278, 167, 278, 167, 278, 167, 611, 167, 444, 167, 611, 167, 278, 167, 278, 167, 444, 167, 444, 167, 65535};   

//key One
uint16_t  KeyOne[36] = {417, 278, 167, 278, 167, 611, 167, 444, 167, 611, 167, 278, 167, 278, 167, 278, 167, 278, 167, 278, 167, 611, 167, 444, 167, 611, 167, 278, 167, 278, 167, 278, 167, 444, 167, 65535};

IRsend irsend(4);  // An IR LED is controlled by GPIO pin 4 (D2)

void setup() {
  irsend.begin();
  Serial.begin(115200);
}

void loop() {
         // this should send 451 and result in putting channel 451
         irsend.sendRaw(KeyFour, 36, 38);
         irsend.sendRaw(KeyFive, 36, 38);
         irsend.sendRaw(KeyOne, 36, 38);
  delay(10000);
}`

crankyoldgit, 2017-09-11:
You probably need a bigger delay between sending each code.

kapiloberoi, 2017-09-11:
Thanks for the quick reponse. Yes, started from 15ms and went up to 3000ms  in frustration but still same behaviour.

crankyoldgit, 2017-09-11:
Just to confirm (and now I'm on a real computer/keyboard) I was talking about something like:

```
void loop() {
  // this should send 451 and result in putting channel 451
  irsend.sendRaw(KeyFour, 36, 38);
  delay(1000);  // a second
  irsend.sendRaw(KeyFive, 36, 38);
  delay(1000);  // a second
  irsend.sendRaw(KeyOne, 36, 38);
  delay(10000);  // 10 seconds
}
```

Have you tried sending each individual code on it's own to confirm the codes are correct?
Have you checked that the codes when recorded in the above sequence from the remote match the individual key presses? (This is a long shot)

FYI, the code above should be able to be visually distinguishable if you use a digital camera etc. If not, try modifying the delays to be sure you are sending what you think you are.

Oh, and to state the obvious, try updating to the latest releases of this library and the ESP8266 board manager libs.

kapiloberoi, 2017-09-12:
@crankyoldgit 

Thank you. The issue is resolved with the same code though I had already tried with delay from 15 to 3000. 

I had the DTH receiver on my work table and the distance between IR and DTH was not more than 4-5 inches. For some reason I had to move to a different desk today from which the DTH receiver was around 3-4 meters and as I plugged in ESP - the code started firing itself. 

Working for two days and now I realize that having IR LED too too close to the receiver doesn't always help. Hope this helps someone !!

closed 2017-09-12 by kapiloberoi

crankyoldgit, 2017-09-12:
Ouch! Yeah. That has bitten me also. So I've added it to the Troubleshooting guide as something to lookout for.

https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide/_compare/cda0ec3e00abea79053643e566c7397de5815003...7b612ee32daeeb43aada538b365384154e455e93

imshreshthaj, 2018-03-02:
@kapiloberoi 
Hi,
 Can you please help me to decode the IR signal from Airtel DTH or share the program. 

I was able to decode my tv and speaker ir signal but not able to decode airtel dth signal.
I also tried to use your decoded channel value of 451 and it just worked as charm.

_

> // Airtel DTH Remote Code - Put channel number 451
>
> //key Four
> uint16_t KeyFour[36] = {417, 278, 167, 278, 167, 611, 167, 444, 167, 611, 167, 278, 167, 278, 167, 278, 167, 278, 167, 278, 167, 611, [...]

_

kapiloberoi, 2018-03-02:
Hi,

Give me a day or two as i havent touched ESP since last 3 months. Will send
you IR codes over the weekend.

Thanks

imshreshthaj, 2018-03-02:
@kapiloberoi 

sure.
do you've any reference for the program (i mean any web site that you referred)  that you used to decode ?
Appreciate your help!

Thanks,
Shreshtha

imshreshthaj, 2018-03-02:
@kapiloberoi 
never mind figured it out.