crankyoldgit/IRremoteESP8266 #309: no matching function for irRaw error

Laventissi, 2017-10-01:
### Version/revison of the library used
latest version downloaded just today

### Expected behavior
I edited the IRserver example to suit my needs, I am not great at coding but it's not working and I am getting an error.

"Hi, I am not very good at coding so please excuse my silly mistakes if there are any and help me fix them :D

I am using the example code of IRserver to make a simple web page to control my set top box and TV. When I recorded the codes the set top box ones came as 32 bit NEC and the TV ones as unknown 100 bit and 52 bit. I have tried making this code here, excuse the "samsung TV" part that I have commented in the code. 

I do not know what is going wrong, I get the error "no matching function for call to 'IRsend::sendRaw(unsigned int[100], int, int)'". It's in the last part where I have the "else if" lines in void handleIr. I have tried using unsigned int, uint and also uint16_t but I always get the same type of error. Please help me fix it! Attached below is my code. Any tips and tricks appreciated.

And also the set top box codes that I put there also do not work, I have tried checking with my code receiver what code comes out and the code that comes out is totally different from the one I put in the sketch. "

#### Example code used
`/*
 * IRremoteESP8266: IRServer - demonstrates sending IR codes controlled from a webserver
 * Version 0.2 June, 2017
 * Copyright 2015 Mark Szabo
 *
 * An IR LED circuit *MUST* be connected to ESP8266 pin 4 (D2).
 *
 * TL;DR: The IR LED needs to be driven by a transistor for a good result.
 *
 * Suggested circuit:
 *     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
 *
 * Common mistakes & tips:
 *   * Don't just connect the IR LED directly to the pin, it won't
 *     have enough current to drive the IR LED effectively.
 *   * Make sure you have the IR LED polarity correct.
 *     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
 *   * Typical digital camera/phones can be used to see if the IR LED is flashed.
 *     Replace the IR LED with a normal LED if you don't have a digital camera
 *     when debugging.
 *   * Avoid using the following pins unless you really know what you are doing:
 *     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
 *     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
 *     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
 *   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
 *     for your first time. e.g. ESP-12 etc.
 */
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <WiFiClient.h>

const char* ssid = "._. 2.0";
const char* password = "rtkl12345";
MDNSResponder mdns;

ESP8266WebServer server(80);

IRsend irsend(4);  // An IR LED is controlled by GPIO pin 4 (D2)

void handleRoot() {
  server.send(200, "text/html",
              "<html>" 
                "<head><title>IR Control</title></head>" 
                "<h1>Nikai TV</h1>" 
                "<body>" 
                  "<p><a href=\"ir?nikaion\">ON/OFF</a></p>" 
                  "<p><a href=\"ir?nikaimute\">Mute</a></p>" 
                  "<p><a href=\"ir?nikaivup\">Vol +</a></p>" 
                  "<p><a href=\"ir?nikaivdn\">Vol -</a></p>" 
[... 102 lines cut ...]
}

void loop(void) {
  server.handleClient();
}`

Also when I try checking what comes out when I try the set top box code from the web page this is all i get from all of them instead of the actual one I put in the code:
`327327 (32 bits)
Raw (68): -8924 -4526 -536 -592 -536 -588 -536 -588 -534 -588 -536 -564 -558 -588 -534 -588 -536 -588 -534 -590 -536 -588 -536 -588 -536 -588 -534 -588 -534 -588 -536 -588 -536 -586 -536 -566 -556 -590 -534 -588 -536 -588 -536 -588 -536 -586 -536 -1708 -536 -1708 -536 -588 -536 -588 -536 -1708 -536 -588 -536 -588 -534 -1708 -536 -1708 -536 -1686 -558 
327327 (32 bits)
Raw (68): -8976 -4474 -558 -594 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -534 -590 -536 -588 -536 -588 -536 -588 -536 -588 -534 -588 -534 -590 -534 -586 -536 -588 -536 -566 -558 -588 -536 -590 -534 -588 -534 -588 -560 -1662 -558 -1706 -536 -588 -536 -588 -536 -1708 -536 -588 -534 -588 -536 -1708 -536 -1684 -560 -1708 -536 
327327 (32 bits)
Raw (68): -8974 -4510 -536 -592 -536 -588 -536 -588 -536 -588 -536 -590 -534 -588 -534 -588 -508 -618 -536 -588 -536 -586 -538 -586 -536 -586 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -588 -536 -1708 -536 -1708 -536 -588 -536 -588 -536 -1706 -536 -588 -536 -588 -536 -1708 -534 -1710 -536 -1710 -536 
327327 (32 bits)
Raw (68): -8930 -4522 -558 -592 -536 -588 -536 -588 -534 -588 -536 -588 -534 -588 -536 -588 -536 -588 -536 -590 -534 -566 -558 -588 -534 -588 -536 -588 -536 -586 -536 -588 -536 -590 -534 -588 -536 -588 -536 -588 -536 -586 -536 -588 -534 -588 -534 -1708 -536 -1708 -536 -590 -534 -588 -536 -1686 -558 -586 -536 -588 -536 -1706 -536 -1708 -536 -1708 -536 
327327 (32 bits)
Raw (68): -8948 -4536 -536 -592 -536 -588 -536 -590 -534 -588 -536 -590 -534 -588 -536 -564 -560 -588 -534 -588 -536 -588 -536 -588 -536 -588 -534 -590 -534 -588 -536 -588 -536 -588 -534 -588 -536 -566 -558 -588 -536 -588 -536 -588 -536 -588 -536 -1684 -558 -1708 -536 -588 -536 -588 -536 -1708 -536 -590 -534 -588 -536 -1708 -566 -1678 -536 -1712 -536 
327327 (32 bits)
Raw (68): -8978 -4496 -536 -592 -536 -588 -536 -588 -536 -588 -536 -564 -558 -588 -534 -588 -534 -588 -536 -588 -534 -588 -536 -566 -558 -588 -534 -588 -536 -588 -534 -588 -534 -588 -536 -588 -536 -588 -536 -590 -536 -588 -536 -588 -536 -588 -536 -1706 -536 -1710 -536 -566 -558 -588 -536 -1708 -536 -588 -536 -590 -534 -1706 -536 -1710 -534 -1708 -536 
`

#### Circuit diagram and hardware used (if applicable)
used the exact same circuit provided in the github

thanks for any help in advance and im sorry if there are silly mistakes and stupidity, I am not good in english or coding but any help and tips are appreciated

crankyoldgit, 2017-10-02:
For starters, your code is odd/broken with how you are storing numbers.
e.g.
```
    unsigned int nikonoff[100] = {-4060,-3918,-508,-2004,-508,-2002,-510,-2002,-508,-2004,-506,-1050,-508,-1048,-510,-2004,-508,-1048,-508,-2002,-510,-1050,-508,-2004,-510,-1048,-508,-1050,-508,-1048,-508,-1050,-508,-1050,-508,-2004,-508,-2002,-510,-1048,-508,-2004,-508,-1050,-506,-2004,-508,-1048,-510,-2002,-456,-8446,-3956,-3998,-508,-2004,-508,-2002,-508,-2004,-508,-1978,-532,-1050,-508,-1050,-508,-2002,-508,-1050,-508,-2004,-508,-1050,-508,-2002,-510,-1050,-508,-1050,-508,-1048,-508,-1050,-508,-1050,-508,-2002,-510,-2002,-508,-1050,-508,-2002,-510,-1050,-508,-2002,-508};
```

`unsigned` means the number(s) are >= 0. i.e. positive. You are using negative numbers, which can only be stored in a 'signed' int. i.e. no, u or unsigned. (Technical aside: It's actually more complicated than that, as a negative number will be stored as a two's complement.) Anyway, the upshot is, your number won't be the number you think it is. Use positive numbers only.

so change your code for example,  to:
```
      uint16_t nikonoff[100] = {4060,3918,508,2004,508,2002,510,2002,508,2004,506,1050,508,1048,510,2004,508,1048,508,2002,510,1050,508,2004,510,1048,508,1050,508,1048,508,1050,508,1050,508,2004,508,2002,510,1048,508,2004,508,1050,506,2004,508,1048,510,2002,456,8446,3956,3998,508,2004,508,2002,508,2004,508,1978,532,1050,508,1050,508,2002,508,1050,508,2004,508,1050,508,2002,510,1050,508,1050,508,1048,508,1050,508,1050,508,2002,510,2002,508,1050,508,2002,510,1050,508,2002,508};
```

See also the example for sending raw codes in https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino#39

FYI, Typical Samsung TV codes are 32 bits. Your code seems to be storing them as 64 bit. It is unlikely that they are 64bit codes.
e.g.
Your code:
`code=E0E0A659E0E0A659`
should probably be:
`code=E0E0A659`
Note: your version(s) is effectively the code repeated, in a way.

Hope that helps you some.

Laventissi, 2017-10-02:
Thank you so much for the very quick reply! I really appreciate your help, I have also tried int because I know that they are not supposed to be unsigned but that didn't work. I will try removing the negatives and replacing them with same number but positive, also thanks for letting me know about the samsung code, I think I should've bought the Vishay IR receiver instead of the one I got as I've heard this one is not accurate. The brand of my set top box is Startrack and model number I think SR 9090PLUS(thats whats written on the front). I also have an A/C remote that I tried reading the IR codes from but one click sends 2-3 different IR codes at the same time and repeating it doesn't give any results that are consistent so I am not able to find out an average, any idea for that? The AC brand is Eurostar, remote model GZ-1002A-E3. Thanks very much once again and I will try your suggested changes right now! :)

Laventissi, 2017-10-02:
The IR reciever I have is the VS1838B by the way.

crankyoldgit, 2017-10-02:
Note: Your code has/uses: `irsend.sendNEC(code, 32);` You probably want to use `irsend.sendSAMSUNG(code, 32);`

Btw, how are you obtaining (capturing) the codes? Can you give us an example of a capture from the dumpv2 program for the samsung remote code?

Laventissi, 2017-10-03:
I have tried the nikai TV today, no luck.. it doesn't work. :(

I will change the samsung one now, thanks.

I am obtaining the codes using dumpv2, here are the codes for the remotes:

```
Reciever:
ON/OFF(NEC) - 807F00FF
	Raw (68): -8992 -4518 -532 -1726 -530 -590 -530 -592 -530 -590 -532 -590 -530 -590 -530 -590 -530 -596 -530 -590 -558 -1698 -532 -1726 -530 -1726 -530 -1724 -532 -1726 -530 -1726 -530 -1732 -532 -590 -530 -590 -532 -590 -532 -590 -530 -590 -530 -590 -530 -590 -558 -568 -530 -1726 -530 -1726 -530 -1726 -532 -1724 -532 -1724 -532 -1724 -530 -1702 -554 -1730 -530 

Mute(NEC) - 807FC03F
	Raw (68): -9022 -4490 -560 -1696 -560 -560 -560 -560 -560 -560 -560 -560 -562 -560 -560 -560 -560 -566 -560 -560 -562 -1696 -560 -1696 -560 -1696 -562 -1696 -560 -1696 -560 -1696 -560 -1702 -560 -1696 -562 -1694 -562 -560 -560 -560 -562 -560 -560 -560 -560 -560 -562 -564 -560 -560 -562 -558 -562 -1694 -562 -1694 -562 -1694 -562 -1694 -562 -1694 -562 -1700 -562 

Ch + (NEC) - 807FA857
	Raw (68): -8992 -4518 -530 -1726 -530 -590 -530 -590 -532 -590 -530 -590 -530 -590 -530 -590 -530 -596 -532 -590 -530 -1724 -532 -1726 -530 -1726 -530 -1726 -530 -1726 -530 -1724 -532 -1732 -530 -1726 -530 -590 -530 -1728 -530 -590 -530 -1726 -532 -590 -530 -590 -530 -596 -530 -590 -530 -1726 -530 -590 -530 -1724 -532 -566 -552 -1726 -530 -1726 -532 -1728 -530 

Ch - (NEC) - 807F6897
	Raw (68): -8990 -4518 -530 -1726 -530 -590 -532 -590 -530 -590 -530 -590 -530 -590 -530 -590 -530 -598 -530 -590 -530 -1724 -532 -1724 -530 -1726 -530 -1726 -532 -1724 -530 -1726 -530 -1732 -530 -590 -530 -1726 -530 -1724 -530 -590 -532 -1726 -530 -590 -532 -590 -530 -596 -530 -1726 -530 -590 -530 -592 -530 -1726 -530 -590 -530 -1726 -530 -1726 -530 -1730 -530 

Vol + (NEC) - 807FE817
	Raw (68): -8990 -4498 -554 -1724 -532 -590 -530 -590 -530 -590 -530 -590 -532 -590 -530 -590 -532 -596 -532 -588 -532 -1726 -530 -1724 -532 -1724 -532 -1724 -532 -1724 -532 -1724 -530 -1732 -532 -1724 -532 -1724 -530 -1724 -532 -588 -532 -1724 -530 -566 -556 -590 -530 -594 -532 -588 -532 -568 -552 -588 -532 -1724 -532 -588 -532 -1724 -532 -1724 -530 -1730 -530 

Vol - (NEC) - 807F28D7
	Raw (68): -8992 -4516 -532 -1724 -534 -588 -532 -588 -532 -588 -532 -586 -534 -588 -532 -588 -534 -594 -534 -562 -556 -1724 -532 -1722 -534 -1724 -532 -1722 -532 -1722 -534 -1722 -534 -1730 -532 -586 -534 -588 -532 -1724 -534 -586 -534 -1722 -534 -588 -532 -588 -532 -594 -534 -1724 -532 -1722 -532 -588 -534 -1722 -532 -588 -532 -1724 -532 -1724 -534 -1726 -534 

1 (NEC) - 807F807F
	Raw (68): -8994 -4494 -554 -1724 -530 -588 -532 -588 -532 -588 -532 -588 -532 -588 -532 -590 -530 -596 -532 -588 -532 -1722 -532 -1724 -532 -1724 -534 -1722 -532 -1722 -532 -1724 -532 -1732 -532 -1724 -530 -588 -532 -590 -530 -588 -532 -588 -532 -588 -532 -588 -532 -594 -532 -588 -532 -1724 -532 -1724 -532 -1724 -532 -1724 -532 -1724 -536 -1698 -556 -1728 -534 

2 (NEC) - 807F40BF
	Raw (68): -8994 -4516 -536 -1722 -532 -588 -534 -588 -534 -586 -534 -586 -534 -586 -534 -588 -532 -594 -532 -588 -534 -1722 -534 -1722 -532 -1724 -534 -1722 -532 -1724 -534 -1700 -556 -1730 -534 -586 -536 -1720 -534 -586 -534 -586 -534 -586 -534 -586 -534 -586 -534 -592 -532 -1724 -532 -588 -532 -1722 -534 -1722 -534 -1722 -534 -1722 -534 -1722 -532 -1728 -534 

3 (NEC) - 807F609F
	Raw (68): -8992 -4514 -534 -1722 -536 -586 -534 -588 -534 -588 -532 -586 -534 -586 -534 -586 -534 -594 -534 -586 -532 -1724 -534 -1722 -534 -1722 -532 -1722 -534 -1722 -534 -1722 -536 -1728 -534 -586 -534 -1722 -532 -1726 -532 -586 -534 -586 -536 -586 -536 -584 -536 -592 -532 -1722 -536 -586 -534 -586 -534 -1722 -532 -1722 -534 -1720 -534 -1722 -534 -1726 -534 

4 (NEC) - 807F20DF
	Raw (68): -8994 -4516 -534 -1722 -534 -586 -534 -586 -534 -586 -534 -586 -534 -586 -534 -586 -534 -592 -534 -586 -534 -1722 -536 -1722 -534 -1722 -534 -1722 -534 -1722 -534 -1720 -536 -1726 -534 -588 -534 -586 -534 -1722 -532 -588 -534 -586 -534 -586 -534 -588 -532 -592 -534 -1722 -534 -1722 -536 -586 -534 -1722 -534 -1720 -536 -1722 -534 -1722 -532 -1726 -534 

5 (NEC) - 807FA05F
	Raw (68): -8994 -4492 -558 -1698 -556 -588 -532 -588 -534 -586 -532 -588 -534 -586 -534 -586 -534 -592 -536 -586 -532 -1722 -536 -1720 -534 -1722 -534 -1720 -534 -1722 -534 -1722 -534 -1728 -534 -1722 -534 -586 -534 -1722 -534 -586 -534 -564 -556 -586 -534 -586 -534 -590 -534 -586 -534 -1722 -532 -588 -534 -1722 -534 -1720 -534 -1700 -556 -1722 -534 -1726 -534 

6 (NEC) - 807FE01F
	Raw (68): -8996 -4514 -534 -1722 -534 -586 -534 -586 -534 -586 -534 -586 -534 -564 -558 -586 -534 -594 -534 -586 -532 -1724 -534 -1722 -534 -1722 -534 -1722 -534 -1722 -534 -1722 -534 -1728 -534 -1722 -534 -1700 -556 -1722 -534 -586 -534 -586 -534 -586 -534 -586 -534 -592 -534 -586 -534 -588 -532 -586 -534 -1722 -534 -1722 -532 -1722 -534 -1720 -534 -1726 -534 

7 (NEC) - 807F10EF
	Raw (68): -8992 -4494 -558 -1722 -534 -588 -534 -586 -534 -586 -534 -584 -536 -586 -534 -588 -534 -592 -534 -586 -536 -1722 -532 -1722 -536 -1722 -534 -1722 -534 -1722 -534 -1722 -534 -1728 -534 -586 -534 -586 -534 -586 -534 -1722 -534 -586 -534 -586 -534 -586 -534 -590 -534 -1722 -534 -1722 -534 -1722 -534 -586 -534 -1722 -534 -1722 -534 -1722 -534 -1726 -534 

8 (NEC) - 807F50AF
	Raw (68): -8992 -4516 -536 -1720 -536 -586 -532 -586 -534 -586 -534 -586 -534 -586 -536 -584 -536 -592 -534 -586 -534 -1722 -534 -1722 -534 -1722 -534 -1722 -536 -1720 -534 -1722 -534 -1728 -534 -588 -534 -1722 -534 -586 -536 -1720 -534 -586 -534 -586 -534  [... cut ...]  -612 -510 -1734 -510 -612 -510 -1734 -512 -1734 -510 -1736 -510 -1734 -510 -1736 -510 -1712 -532 

MUTE-

E0E0F00FE0E0F00F (32 bits)
Raw (68): -4500 -4538 -534 -1734 -510 -1732 -512 -1734 -510 -612 -510 -610 -512 -610 -512 -610 -510 -610 -510 -1734 -510 -1734 -510 -1734 -510 -612 -510 -612 -510 -588 -534 -612 -510 -612 -510 -1734 -512 -1734 -510 -1734 -510 -1734 -512 -610 -512 -612 -510 -612 -510 -610 -512 -610 -510 -612 -510 -610 -512 -610 -512 -1734 -510 -1734 -512 -1734 -510 -1734 -512 

```

Laventissi, 2017-10-03:
In the nikai TV remote I get unknown encoding of I think 100 bits and 52 bits so I didn't save those numbers and saved just raw because I thought I wont be able to use those

crankyoldgit, 2017-10-03:
Just checking, the nikai code, is that the star track device?

crankyoldgit, 2017-10-03:
I'm asking as I'm developing the code to send/decode it (the 100 long code)

crankyoldgit, 2017-10-03:
@Laventissi If you care to download and try the new [StarTrack branch](https://github.com/markszabo/IRremoteESP8266/tree/StarTrack), it should decode & send the codes you've labled as NIKAI TV.

Let me know how you go etc, so I can include it officially in the library.

Laventissi, 2017-10-03:
The Nikai TV is not the set top box, it is a TV and the branding is Nikai.. it's a pretty old LCD TV I have.

The Star Track device is labelled as "receiver". I will try the the startrack branch you made but I am not sure what to do/how to use it.. Sorry, I am new to github.. thanks :)

Laventissi, 2017-10-03:
I have also tried using sendsamsung as you suggested, not working, I checked the code that comes out with my IR reciever and all buttons show this;

```
D7BAA9C4Unknown encoding: D7BAA9C4 (34 bits)
Raw (68): -4494 -4516 -562 -566 -562 -562 -562 -562 -562 -560 -562 -562 -562 -560 -562 -562 -562 -562 -562 -562 -562 -562 -562 -562 -562 -560 -562 -560 -564 -560 -564 -562 -562 -562 -562 -562 -562 -562 -562 -560 -562 -560 -562 -562 -562 -562 -562 -562 -562 -562 -562 -562 -562 -560 -564 -562 -562 -560 -562 -560 -564 -562 -562 -562 -564 -568 -564 
D7BAA9C4Unknown encoding: D7BAA9C4 (34 bits)
Raw (68): -4496 -4508 -564 -566 -562 -560 -564 -562 -562 -562 -564 -560 -564 -560 -562 -560 -564 -560 -562 -562 -562 -562 -564 -560 -562 -562 -562 -560 -564 -560 -562 -562 -562 -562 -562 -562 -564 -560 -562 -562 -562 -562 -564 -560 -562 -560 -564 -562 -560 -562 -562 -562 -562 -562 -564 -560 -564 -562 -562 -560 -562 -560 -564 -560 -562 -566 -562 
D7BAA9C4Unknown encoding: D7BAA9C4 (34 bits)
Raw (68): -4496 -4516 -564 -566 -562 -562 -564 -560 -564 -560 -578 -544 -564 -560 -562 -562 -562 -560 -562 -562 -562 -538 -584 -562 -562 -562 -564 -562 -560 -562 -562 -560 -562 -560 -562 -538 -584 -562 -562 -560 -562 -562 -562 -562 -562 -562 -562 -560 -562 -560 -562 -560 -562 -562 -562 -562 -564 -560 -562 -562 -562 -560 -562 -560 -562 -568 -562 
D7BAA9C4Unknown encoding: D7BAA9C4 (34 bits)
Raw (68): -4496 -4508 -564 -566 -562 -562 -562 -560 -562 -562 -562 -560 -562 -562 -562 -560 -564 -562 -562 -560 -564 -560 -564 -560 -562 -560 -562 -562 -562 -560 -564 -562 -562 -562 -562 -538 -586 -560 -562 -562 -562 -560 -562 -560 -562 -562 -562 -562 -564 -560 -564 -562 -562 -562 -562 -562 -562 -560 -562 -560 -562 -562 -562 -560 -562 -566 -564 

```

Laventissi, 2017-10-03:
This is the way I have my circuit set up.. 
https://imgur.com/a/dEFAh
I am using a 2n222 transistor and the LED I am using is a high power one, this one to be precise:
https://www.aliexpress.com/item/10-Pieces-3W-LED-Light-Bulb-Lamp-IR-Infrared-940nm-1-2-1-4V-400mA-500mA/32819847514.html?spm=a2g0s.9042311.0.0.dywrdQ

crankyoldgit, 2017-10-04:
I've renamed the branch and code to Nikai.
i.e. https://github.com/markszabo/IRremoteESP8266/tree/Nikai

crankyoldgit, 2017-10-04:
I'm a bit confused as to what program you are using that is giving you the aforementioned "Raw" output.
That's not the output I expect from the dumpv2 program in our library.
If it is, then there is a problem with it.

Can you please give me the complete (unedited) output of the dumpv2 program/sketch for the Samsung 'MUTE' button?
i.e.
It should NOT be in the format:

> E0E0F00FE0E0F00F (32 bits)
> Raw (68): -4500 -4538 -534 -1734 -510 -1732 -512 -1734 -510 -612 -510 -610 -512 -610 -512 -610 -510 -610 -510 -1734 -510 -1734 -510 -1734 -510 -612 -510 -612 -510 -588 [...]

It *should* look something like this if you are using the right program:

> Encoding  : UNKNOWN
> Code      : 4A1B17EB (15 bits)
> Timing[30]: 
>      +1350, - 450     + 500, - 450     +1350, - 450     +1350, - 450
>      +1350, - 450     + 450, - 450     + 500, - 450     + 500, - 450
>      +1350, - 450     +1350, - 450     +1350, - 450     +1350, - 450
>      +1350, - 450     + 500, - 400     +1400, - 400
> uint16_t  rawData[30] = {1350,450, 500,450, 1350,450, 1350,450, 1350,450, 450,450, 500,450, 500,450, 1350,450, 1350,450, 1350,450, 1350,450, 1350,450, 500,400, 1400,400 };  // UNKNOWN 4A1B17EB

Laventissi, 2017-10-04:
Hi, I have just realized I am not using dumpv2 but dump. I apologize for my immense stupidity, this is the code for the MUTE button on the Samsung Remote:
```
Encoding  : SAMSUNG
Code      : E0E0F00F (32 bits)
Timing[67]: 
   +  4526, -  4510,    +   560, -  1708,    +   538, -  1706,    +   538, -  1708, 
   +   538, -   582,    +   538, -   584,    +   538, -   584,    +   536, -   588, 
   +   534, -   586,    +   538, -  1706,    +   512, -  1710,    +   534, -  1734, 
   +   536, -   584,    +   538, -   584,    +   538, -   584,    +   512, -   588, 
   +   534, -   610,    +   536, -  1684,    +   536, -  1732,    +   538, -  1706, 
   +   538, -  1708,    +   538, -   584,    +   538, -   586,    +   510, -   610, 
   +   538, -   584,    +   538, -   584,    +   538, -   584,    +   512, -   610, 
   +   538, -   584,    +   536, -  1708,    +   538, -  1708,    +   536, -  1708, 
   +   538, -  1708,    +   512
uint16_t rawData[67] = {4526, 4510,  560, 1708,  538, 1706,  538, 1708,  538, 582,  538, 584,  538, 584,  536, 588,  534, 586,  538, 1706,  512, 1710,  534, 1734,  536, 584,  538, 584,  538, 584,  512, 588,  534, 610,  536, 1684,  536, 1732,  538, 1706,  538, 1708,  538, 584,  538, 586,  510, 610,  538, 584,  538, 584,  538, 584,  512, 610,  538, 584,  536, 1708,  538, 1708,  536, 1708,  538, 1708,  512};  // SAMSUNG E0E0F00F
uint32_t address = 0x7;
uint32_t command = 0xF;
uint64_t data = 0xE0E0F00F;

```

I will try getting all the codes with dumpv2 now and try using them again and let you know soon whether it works or not. Thanks!

Laventissi, 2017-10-04:
When I record it with dumpv2, which code should I use? The one given as "Code :" or "uint32_t address" or the uint32_t command or uint64_t data? I know the unt16_t is for raw but what if it is encoding as samsung?

Laventissi, 2017-10-04:
I have tried using uint64_t data with the code from there and changed uint32 to 64, also tried using uint32_t address and command using uint32 for samsung. I always get the same random code D97 something, is there something wrong with my code of sending the IR code?

crankyoldgit, 2017-10-04:
For your purposes, you want to use the _Code/data_ part.

Using your above decode, to re-send that same command, you would simply invoke it by:
`irsend.sendSAMSUNG(0xE0E0F00F);`

You can technically call `sendSAMSUNG()` with anything from a uint8_t to a uint64_t for the first argument. However, 0xE0E0F00F is 32 bits long, so you need _at least_ a uint32_t to store it.
e.g.
```
uint32_t data = 0xE0E0F00F;

irsend.sendSAMSUNG(data);
```

Laventissi, 2017-10-04:
Thank you very much! I have tried doing uin32 address and added the code with it like: "0x7E0E0F00F"
This is working for me, is it correct/will it work or should I change it? 

Also I am still not sure how to use the commit you made, do I just have to download the files and update them from the homepage of this github? Sorry :3

Laventissi, 2017-10-04:
OK I have figured it out and downloaded the files again and replaced them.

I have tried using the Nikai remote to record codes and pressed the power on button only a few times and I get different results, I am not sure what to do after this, should I use the raw data from one of them to check it out or irsend.sendNikai that you made?

```
Encoding  : NIKAI
Code      : D5F2A (24 bits)
Timing[51]: 
   +  3982, -  3998,    +   508, -  2004,    +   508, -  2006,    +   506, -  2004, 
   +   508, -  2004,    +   508, -  1050,    +   508, -  1050,    +   506, -  2006, 
   +   506, -  1050,    +   508, -  2004,    +   508, -  1050,    +   508, -  2004, 
   +   508, -  1050,    +   508, -  1052,    +   506, -  1050,    +   508, -  1050, 
   +   508, -  1050,    +   508, -  2004,    +   508, -  2004,    +   508, -  1050, 
   +   508, -  2004,    +   508, -  1050,    +   508, -  2004,    +   508, -  1050, 
   +   510, -  2004,    +   456
uint16_t rawData[51] = {3982, 3998,  508, 2004,  508, 2006,  506, 2004,  508, 2004,  508, 1050,  508, 1050,  506, 2006,  506, 1050,  508, 2004,  508, 1050,  508, 2004,  508, 1050,  508, 1052,  506, 1050,  508, 1050,  508, 1050,  508, 2004,  508, 2004,  508, 1050,  508, 2004,  508, 1050,  508, 2004,  508, 1050,  510, 2004,  456};  // NIKAI D5F2A
uint64_t data = 0xD5F2A;

Encoding  : NIKAI
Code      : D5F2A (24 bits)
Timing[103]: 
   +  3982, -  3998,    +   508, -  2004,    +   508, -  2006,    +   506, -  2004, 
   +   508, -  2004,    +   508, -  1050,    +   508, -  1050,    +   508, -  2004, 
   +   508, -  1050,    +   508, -  2004,    +   508, -  1050,    +   508, -  1980, 
   +   530, -  1050,    +   508, -  1050,    +   508, -  1050,    +   508, -  1050, 
   +   508, -  1050,    +   508, -  2004,    +   508, -  2004,    +   508, -  1050, 
   +   508, -  2006,    +   506, -  1052,    +   506, -  2006,    +   506, -  1052, 
   +   506, -  2004,    +   456, -  8448,    +  3958, -  3996,    +   510, -  2004, 
   +   508, -  2004,    +   508, -  2004,    +   508, -  2004,    +   508, -  1050, 
   +   508, -  1050,    +   506, -  2004,    +   510, -  1050,    +   508, -  2004, 
   +   508, -  1050,    +   508, -  2004,    +   508, -  1050,    +   508, -  1050, 
   +   508, -  1050,    +   508, -  1050,    +   508, -  1050,    +   506, -  2004, 
   +   508, -  2004,    +   508, -  1050,    +   508, -  2004,    +   508, -  1026, 
   +   506, -  2030,    +   508, -  1050,    +   508, -  2004,    +   456
uint16_t rawData[103] = {3982, 3998,  508, 2004,  508, 2006,  506, 2004,  508, 2004,  508, 1050,  508, 1050,  508, 2004,  508, 1050,  508, 2004,  508, 1050,  508, 1980,  530, 1050,  508, 1050,  508, 1050,  508, 1050,  508, 1050,  508, 2004,  508, 2004,  508, 1050,  508, 2006,  506, 1052,  506, 2006,  506, 1052,  506, 2004,  456, 8448,  3958, 3996,  510, 2004,  508, 2004,  508, 2004,  508, 2004,  508, 1050,  508, 1050,  506, 2004,  510, 1050,  508, 2004,  508, 1050,  508, 2004,  508, 1050,  508, 1050,  508, 1050,  508, 1050,  508, 1050,  506, 2004,  508, 2004,  508, 1050,  508, 2004,  508, 1026,  506, 2030,  508, 1050,  508, 2004,  456};  // NIKAI D5F2A
uint64_t data = 0xD5F2A;

Encoding  : NIKAI
Code      : D5F2A (24 bits)
Timing[103]: 
   +  3982, -  3998,    +   508, -  2004,    +   506, -  2004,    +   508, -  2004, 
   +   508, -  2002,    +   508, -  1050,    +   508, -  1050,    +   506, -  2004, 
   +   508, -  1050,    +   508, -  2004,    +   510, -  1050,    +   506, -  2004, 
   +   508, -  1050,    +   508, -  1050,    +   508, -  1050,    +   508, -  1050, 
   +   508, -  1050,    +   506, -  2006,    +   506, -  2004,    +   484, -  1074, 
   +   508, -  2004,    +   508, -  1050,    +   508, -  2004,    +   508, -  1050, 
   +   508, -  2004,    +   456, -  8448,    +  3958, -  3998,    +   508, -  2004, 
   +   508, -  2004,    +   508, -  2004,    +   510, -  2006,    +   504, -  1050, 
   +   508, -  1050,    +   508, -  2004,    +   508, -  1050,    +   508, -  2004, 
   +   506, -  1050,    +   508, -  2004,    +   508, -  1050,    +   508, -  1050, 
   +   508, -  1050,    +   508, -  1052,    +   506, -  1050,    +   508, -  2004, 
   +   508, -  2004,    +   508, -  1050,    +   508, -  2004,    +   510, -  1050, 
   +   506, -  2006,    +   506, -  1050,    +   508, -  2004,    +   456
uint16_t rawData[103] = {3982, 3998,  508, 2004,  506, 2004,  508, 2004,  508, 2002,  508, 1050,  508, 1050,  506, 2004,  508, 1050,  508, 2004,  510, 1050,  506, 2004,  508, 1050,  508, 1050,  508, 1050,  508, 1050,  508, 1050,  506, 2006,  506, 2004,  484, 1074,  508, 2004,  508, 1050,  508, 2004,  508, 1050,  508, 2004,  456, 8448,  3958, 3998,  508, 2004,  508, 2004,  508, 2004,  510, 2006,  504, 1050,  508, 1050,  508, 2004,  508, 1050,  508, 2004,  506, 1050,  508, 2004,  508, 1050,  508, 1050,  508, 1050,  508, 1052,  506, 1050,  508, 2004,  508, 2004,  508, 1050,  508, 2004,  510, 1050,  506, 2006,  506, 1050,  508, 2004,  456};  // NIKAI D5F2A
uint64_t data = 0xD5F2A;

Encoding  : NIKAI
Code      : D5F2A (24 bits)
Timing[51]: 
   +  3982, -  3998,    +   508, -  2004,    +   508, -  2004,    +   508, -  2004, 
   +   508, -  2004,    +   506, -  1050,    +   508, -  1050,    +   508, -  2004, 
   +   508, -  1048,    +   508, -  2006,    +   506, -  1050,    +   506, -  2006, 
   +   508, -  1050,    +   510, -  1050,    +   508, -  1052,    +   506, -  1050, 
   +   508, -  1050,    +   508, -  2006,    +   508, -  2004,    +   508, -  1052, 
[... cut ...] 0,  508, 1050,  508, 1050,  508, 1050,  506, 1982,  530, 2006,  506, 1050,  508, 2006,  508, 1050,  508, 2004,  506, 1052,  508, 2006,  454, 8450,  3958, 3974,  530, 2006,  508, 2004,  506, 2006,  508, 2006,  506, 1050,  508, 1050,  508, 1978,  508, 1052,  512, 2024,  508, 1050,  508, 2004,  508, 1050,  508, 1050,  508, 1052,  482, 1078,  506, 1052,  508, 2004,  508, 2004,  508, 1050,  506, 2004,  508, 1050,  508, 2004,  508, 1050,  508, 2004,  456};  // NIKAI D5F2A
uint64_t data = 0xD5F2A;

```

Laventissi, 2017-10-04:
I think there is something wrong with the IRServer code I made, since I tried the codes I found with dumpv2 and sendsamsung with the example of IRsend and all of them worked but when I try them with the IRServer edited one I made they all get the same result: 
```
Encoding  : UNKNOWN
Code      : D7BAA9C4 (34 bits)
Timing[67]: 
   +  4436, -  4570,    +   504, -   624,    +   504, -   618,    +   530, -   592, 
   +   532, -   592,    +   530, -   594,    +   530, -   594,    +   558, -   566, 
   +   504, -   618,    +   530, -   594,    +   530, -   592,    +   530, -   594, 
   +   532, -   590,    +   530, -   594,    +   532, -   594,    +   530, -   592, 
   +   504, -   618,    +   530, -   592,    +   504, -   620,    +   530, -   594, 
   +   530, -   594,    +   530, -   594,    +   532, -   592,    +   532, -   592, 
   +   528, -   594,    +   532, -   592,    +   506, -   618,    +   532, -   594, 
   +   530, -   592,    +   532, -   592,    +   530, -   592,    +   530, -   594, 
   +   532, -   590,    +   558
uint16_t rawData[67] = {4436, 4570,  504, 624,  504, 618,  530, 592,  532, 592,  530, 594,  530, 594,  558, 566,  504, 618,  530, 594,  530, 592,  530, 594,  532, 590,  530, 594,  532, 594,  530, 592,  504, 618,  530, 592,  504, 620,  530, 594,  530, 594,  530, 594,  532, 592,  532, 592,  528, 594,  532, 592,  506, 618,  532, 594,  530, 592,  532, 592,  530, 592,  530, 594,  532, 590,  558};  // UNKNOWN D7BAA9C4

```

any idea why this is happening?

Laventissi, 2017-10-04:
Bad luck :( They return the same code when checked with IR receiver but when I try them on the actual device it doesn't work :'(

Laventissi, 2017-10-04:
this is the new code

```
/*
 * IRremoteESP8266: IRServer - demonstrates sending IR codes controlled from a webserver
 * Version 0.2 June, 2017
 * Copyright 2015 Mark Szabo
 *
 * An IR LED circuit *MUST* be connected to ESP8266 pin 4 (D2).
 *
 * TL;DR: The IR LED needs to be driven by a transistor for a good result.
 *
 * Suggested circuit:
 *     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
 *
 * Common mistakes & tips:
 *   * Don't just connect the IR LED directly to the pin, it won't
 *     have enough current to drive the IR LED effectively.
 *   * Make sure you have the IR LED polarity correct.
 *     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
 *   * Typical digital camera/phones can be used to see if the IR LED is flashed.
 *     Replace the IR LED with a normal LED if you don't have a digital camera
 *     when debugging.
 *   * Avoid using the following pins unless you really know what you are doing:
 *     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
 *     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
 *     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
 *   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
 *     for your first time. e.g. ESP-12 etc.
 */
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <WiFiClient.h>

const char* ssid = "._. 2ghz";
const char* password = "rtkl12345";
MDNSResponder mdns;

ESP8266WebServer server(80);

IRsend irsend(4);  // An IR LED is controlled by GPIO pin 4 (D2)

void handleRoot() {
  server.send(200, "text/html",
              "<html>" 
                "<head><title>IR Control</title></head>" 
                "<h1>Nikai TV</h1>" 
                "<body>" 
                  "<p><a href=\"ir?nikai=0xD5F2A\">ON/OFF</a></p>" 
                  "<p><a href=\"ir?nikai=0xC0F3F\">Mute</a></p>" 
                  "<p><a href=\"ir?nikai=0xD0F2F\">Vol +</a></p>" 
                  "<p><a href=\"ir?nikai=0xD1F2E\">Vol -</a></p>" 
                "</body>"
                "<h1>Receiver</h1>" 
                "<body>" 
                  "<p><a href=\"ir?code=0x1807F00FF\">ON/OFF</a></p>" 
                  "<p><a href=\"ir?code=0x1807FC03F\">Mute</a></p>" 
                  "<p><a href=\"ir?code=0x1807FE817\">Vol +</a></p>" 
                  "<p><a href=\"ir?code=0x1807F28D7\">Vol -</a></p>" 
                  "<p><a href=\"ir?code=0x1807FA857\">Ch +</a></p>" 
                  "<p><a href=\"ir?code=0x1807F6897\">Ch -</a></p>" 
                "</body>" 
                "<h1>Samsung TV</h1>" \
                "<body>" \
[... 80 lines cut ...]
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);
  server.on("/ir", handleIr);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}

```

crankyoldgit, 2017-10-04:
I agree, the last dump you reported looks 'corrupted'. It seems to be sending the same for each mark/space pair and the same for every bit of the code/IR message.
Answering your earlier question, yes, you should almost always use the sendBlah() version rather than sendRaw() if you can get away with it.

crankyoldgit, 2017-10-04:
Your code(s) for Samsung seem to have a leading '7' which probably shouldn't be there.
e.g.
0x7E0E040BF should *probably* be 0xE0E040BF
as 0x7E0E040BF is larger than 32 bits (technically 35, but probably 36 bits)
Samsung codes are typically 32 bits long, thus 0xE0E040BF fits.

crankyoldgit, 2017-10-04:
Question: Is your TV correctly responding to the codes generated by the library? That is, is the new sending code I've added for you working as expected?

crankyoldgit, 2017-10-04:
Also, re-check your NEC(Receiver) codes in your code, they also appear to be 1 hex digit too long.

crankyoldgit, 2017-10-04:
Regarding your different dump lengths for the Nikai (D5F2A) dumps,
What is happening there is this (Nikai) protocol can have a quite short gap between repeat(ed) messages.
i.e. approx 8.5 milliseconds. By default, the dumpv2 program waits for at least 15 milliseconds of silence before it considers a capture is complete. Hence sometimes it is capturing multiple messages in a single capture. If you want to see for your self, look in capture dump in the raw data. See the ~8400-odd value, that's the gap between messages. You'll kind of see that the set of data to left of that value is then repeated approximately again, after that value.

Trusting the 'decode' value is probably safest.

Laventissi, 2017-10-04:
Yes the irsend.sendNikai seems to be working but the code is not and it deosn't work with the TV. I have also tried removing the 7's from the samsung codes and done same for receiver codes since I added extra 1 there as per the uint32 address from the reading but I still get same results with some random DC---- code and it is in 34 bits.. I will try taking the codes again and using them.

Laventissi, 2017-10-04:
I have tried again, now it is still the same thing with Samsung codes when I check with the IR receiver, the nikai tv and set top box(receiver) codes come as 0. I don't know what is going on :s

Laventissi, 2017-10-04:
I have tried making a new code with the example code...
Here is the code:

```
/*
 * IRremoteESP8266: IRServer - demonstrates sending IR codes controlled from a webserver
 * Version 0.2 June, 2017
 * Copyright 2015 Mark Szabo
 *
 * An IR LED circuit *MUST* be connected to ESP8266 pin 4 (D2).
 *
 * TL;DR: The IR LED needs to be driven by a transistor for a good result.
 *
 * Suggested circuit:
 *     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
 *
 * Common mistakes & tips:
 *   * Don't just connect the IR LED directly to the pin, it won't
 *     have enough current to drive the IR LED effectively.
 *   * Make sure you have the IR LED polarity correct.
 *     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
 *   * Typical digital camera/phones can be used to see if the IR LED is flashed.
 *     Replace the IR LED with a normal LED if you don't have a digital camera
 *     when debugging.
 *   * Avoid using the following pins unless you really know what you are doing:
 *     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
 *     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
 *     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
 *   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
 *     for your first time. e.g. ESP-12 etc.
 */
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <WiFiClient.h>

const char* ssid = "._. 2ghz";
const char* password = "rtkl12345";
MDNSResponder mdns;

ESP8266WebServer server(80);

IRsend irsend(4);  // An IR LED is controlled by GPIO pin 4 (D2)

void handleRoot() {
  server.send(200, "text/html",
              "<html>" \
                "<head><title>ESP8266 Demo</title></head>" \
                "<body>" \
                  "<h1>Nikai TV</h1>" \
                  "<p><a href=\"ir?nikai=0xD5F2A\">ON/OFF</a></p>" \
                  "<p><a href=\"ir?nikai=0xC0F3F\">Mute</a></p>" \
                  "<p><a href=\"ir?nikai=0xD0F2F\">Vol Up</a></p>" \
                  "<p><a href=\"ir?nikai=0xD1F2E\">Vol Down</a></p>" \
                  "<h1>StarTrack</h1>" \
                  "<p><a href=\"ir?code=0x807F00FF\">ON/OFF</a></p>" \
                  "<p><a href=\"ir?code=0x807FC03F\">Mute</a></p>" \
                  "<p><a href=\"ir?code=0x807FE817\">Vol Up</a></p>" \
                  "<p><a href=\"ir?code=0x807F28D7\">Vol Down</a></p>" \
                  "<p><a href=\"ir?code=0x807FA857\">Ch Up</a></p>" \
                  "<p><a href=\"ir?code=0x807F6897\">Ch Down</a></p>" \
                  "<h1>Samsung</h1>" \
                  "<p><a href=\"ir?samsung=0xE0E040BF\">ON/OFF</a></p>" \
                  "<p><a href=\"ir?samsung=0xE0E0F00F\">Mute</a></p>" \
                  "<p><a href=\"ir?samsung=0xE0E0E01F\">Vol Up</a></p>" \
[... 404 lines cut ...]
   +   558, -   566,    +   556, -   568,    +   558, -   568,    +   556, -   568, 
   +   558, -   566,    +   556, -   566,    +   558, -   564,    +   558, -   566, 
   +   558, -   564,    +   556, -   566,    +   560, -   566,    +   558, -   568, 
   +   558, -   564,    +   558, -   566,    +   558, -   566,    +   558, -   564, 
   +   558, -   566,    +   556, -   566,    +   556, -   566,    +   558, -   568, 
   +   558, -   566,    +   556
uint16_t rawData[67] = {4488, 4524,  558, 568,  558, 566,  556, 566,  558, 566,  558, 544,  578, 566,  558, 566,  558, 566,  560, 566,  558, 564,  560, 564,  558, 566,  556, 568,  558, 568,  556, 568,  558, 566,  556, 566,  558, 564,  558, 566,  558, 564,  556, 566,  560, 566,  558, 568,  558, 564,  558, 566,  558, 566,  558, 564,  558, 566,  556, 566,  556, 566,  558, 568,  558, 566,  556};  // UNKNOWN D7BAA9C4

Encoding  : UNKNOWN
Code      : D7BAA9C4 (34 bits)
Timing[67]: 
   +  4492, -  4520,    +   558, -   570,    +   558, -   566,    +   558, -   564, 
   +   558, -   564,    +   558, -   566,    +   556, -   568,    +   558, -   566, 
   +   556, -   568,    +   558, -   566,    +   558, -   566,    +   556, -   566, 
   +   558, -   566,    +   558, -   566,    +   558, -   568,    +   558, -   564, 
   +   558, -   566,    +   558, -   566,    +   558, -   564,    +   558, -   566, 
   +   556, -   566,    +   558, -   566,    +   558, -   568,    +   556, -   566, 
   +   558, -   568,    +   558, -   566,    +   558, -   566,    +   556, -   566, 
   +   560, -   564,    +   558, -   566,    +   558, -   566,    +   558, -   566, 
   +   556, -   568,    +   558
uint16_t rawData[67] = {4492, 4520,  558, 570,  558, 566,  558, 564,  558, 564,  558, 566,  556, 568,  558, 566,  556, 568,  558, 566,  558, 566,  556, 566,  558, 566,  558, 566,  558, 568,  558, 564,  558, 566,  558, 566,  558, 564,  558, 566,  556, 566,  558, 566,  558, 568,  556, 566,  558, 568,  558, 566,  558, 566,  556, 566,  560, 564,  558, 566,  558, 566,  558, 566,  556, 568,  558};  // UNKNOWN D7BAA9C4

```

Something seems to be wrong with my edit of the code? Or is it something wrong with the example code itself? I highly doubt something is wrong with the example code as I am inexperienced and I have probably messed up but I haven't really changed anything except for the web page design thing and added the "else if" arguments for the "code" "samsung" and "nikai" variables.

crankyoldgit, 2017-10-04:
Yes, there are multiple things wrong with _your_ 'code' and how you are using it.
I don't use the server code much so I didn't notice it immediately.

First things first, the codes reported by the dumpv2 program are in [Hexadecimal](https://en.wikipedia.org/wiki/Hexadecimal), the original 'server' example code expects them in [Decimal](https://en.wikipedia.org/wiki/Decimal).

You have some options, either convert the hex codes to dec (plenty of online calculators availablee tc if you don't know how to do it) and use them in your existing code,
e.g.
  `"<p><a href=\"ir?samsung=3772793023\">ON/OFF</a></p>"  // Note: 3772793023 (dec) == E0E040BF (hex) == 0xE0E040BF (C-style hex literal)`

**or:**

Change your code to look like:
  `"<p><a href=\"ir?samsung=E0E040BF\">ON/OFF</a></p>"  // Note: No leading "0x". That prefix is typically only for c/c++ literals.`
  &
  `uint32_t samsung = strtoul(server.arg(i).c_str(), NULL, 16);  // Note: Base 16 (hex), not Base 10 (dec)`

**also:**
You are specifying the wrong bit size for Nikai. It is a 24 bit protocol.
`      irsend.sendNikai(nikai, 24);  // Note: 24 bits, not 32`
or better still, remove the **bit size** for all the sendProtocol() routines and just use the default, like the following:
`      irsend.sendNikai(nikai);`
The sendProtocol() routines with bit sizes (not array sizes) _mostly_ default to the correct bit size for the protocol in use.

I'm going to **strongly** recommend you use start small & simple and use the IRSendDemo example code first to confirm if your captured codes are working as expected and have the desired effect.
You should be able to plug in the hex codes from the dumpv2 program straight into that.
e.g.
`sendSAMSUNG(0xE0E040BF);` or `sendNikai(0xD5F2A);` etc.

Once you've got that sorted, you can work your way to getting it working on the 'server' code.

Laventissi, 2017-10-05:
Thanks a lot @crankyoldgit ! That was the only issue so far with other things not working that I didn't change the default value for decimal..

Also thanks a lot for making that nikai encoding system for me! I have successfully made what I wanted to and it all works perfect, this issue post can be closed now :)

Thanks a lot once again! 👍

crankyoldgit, 2017-10-05:
So, can you please confirm that both sending and receiving nikai codes
works? I.e. I can then mark it as working/tested in the code.

closed 2017-10-07 by crankyoldgit

Laventissi, 2017-10-07:
I am sorry for the late reply but yes the nikai codes work.

mohammadowais3701, 2020-04-17:
**no matching function for call to 'IRsend::sendRaw(unsigned int [149], unsigned int, int&)'
**
Kindly help me for this to solve this error
`
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_SEND_PIN 4
IRsend irsend(IR_SEND_PIN);
int khz=38;
unsigned int ACoff[]={3026, 3044, 3032, 4304, 578, 1692, 550, 582, 550, 1706, 552, 712, 552, 580, 552, 1704, 552, 582, 552, 1838, 550, 582, 580, 566, 550, 582, 550, 714, 552, 582, 550, 582, 552, 582, 550, 714, 550, 582, 552, 582, 578, 1690, 552, 712, 552, 582, 550, 582, 550, 582, 550, 714, 550, 582, 552, 582, 550, 582, 578, 700, 550, 582, 550, 582, 550, 582, 550, 1838, 550, 582, 552, 580, 552, 582, 552, 712, 578, 568, 550, 582, 550, 582, 550, 714, 550, 1704, 552, 1706, 550, 582, 550, 714, 550, 582, 578, 566, 552, 582, 552, 712, 550, 582, 552, 580, 552, 1704, 552, 714, 550, 582, 552, 582, 578, 568, 550, 714, 552, 582, 552, 582, 550, 582, 550, 714, 550, 582, 552, 582, 552, 582, 578, 722, 552, 1706, 550, 582, 552, 1704, 552, 736, 552, 582, 550, 1704, 552, 1704, 552, 582, 580}; // UNKNOWN 9BB5B2B5
unsigned int ACon[]=  {3032, 3046, 3028, 4304, 576, 1692, 552, 582, 550, 1706, 550, 714, 550, 582, 552, 1706, 550, 582, 552, 1836, 552, 582, 578, 568, 552, 580, 552, 714, 550, 582, 550, 582, 552, 582, 550, 1836, 552, 582, 550, 582, 578, 1692, 550, 712, 552, 582, 552, 582, 550, 582, 550, 714, 552, 582, 550, 582, 552, 582, 578, 700, 550, 582, 550, 582, 550, 582, 552, 1838, 550, 582, 552, 582, 552, 582, 550, 714, 578, 568, 550, 582, 550, 582, 550, 714, 552, 1706, 550, 1704, 550, 582, 552, 714, 550, 582, 578, 568, 550, 582, 550, 714, 552, 582, 550, 582, 550, 1706, 550, 714, 550, 582, 550, 582, 578, 568, 550, 714, 550, 582, 550, 582, 552, 582, 550, 714, 550, 582, 552, 582, 552, 582, 578, 722, 552, 1704, 552, 582, 550, 1706, 552, 736, 550, 584, 550, 1706, 550, 1706, 550, 1704, 578};// UNKNOWN 9BB5B2B5
unsigned int Temp16[]={3030, 3042, 3032, 4304, 576, 1694, 550, 582, 552, 1706, 550, 712, 552, 580, 552, 1706, 550, 582, 550, 1836, 550, 582, 578, 568, 552, 582, 552, 1834, 552, 582, 550, 1706, 552, 1706, 548, 714, 552, 580, 552, 582, 578, 1690, 552, 714, 550, 582, 552, 582, 550, 582, 552, 712, 550, 582, 550, 582, 552, 582, 578, 700, 550, 582, 552, 582, 552, 582, 550, 1836, 552, 582, 550, 584, 550, 582, 550, 712, 580, 568, 550, 582, 550, 584, 550, 712, 552, 1704, 552, 1704, 552, 580, 552, 714, 550, 582, 580, 566, 552, 580, 552, 712, 552, 582, 550, 582, 552, 1704, 552, 712, 552, 582, 550, 582, 580, 566, 552, 712, 552, 582, 552, 582, 550, 582, 550, 714, 550, 582, 550, 582, 550, 582, 578, 722, 550, 1706, 552, 582, 550, 1704, 552, 1860, 552, 1704, 552, 1704, 552, 582, 550, 582, 580}; // UNKNOWN F35ACDF0
unsigned int Temp17[] ={3030, 3042, 3032, 4306, 574, 1692, 550, 582, 550, 1704, 552, 712, 552, 582, 552, 1704, 550, 582, 550, 1836, 552, 582, 578, 568, 552, 1704, 552, 714, 550, 582, 550, 1706, 550, 1706, 552, 712, 550, 582, 552, 580, 578, 1688, 554, 712, 552, 582, 550, 582, 552, 582, 550, 714, 550, 582, 552, 582, 550, 582, 580, 698, 550, 582, 552, 582, 550, 582, 550, 1838, 552, 580, 550, 582, 550, 582, 552, 712, 580, 568, 550, 582, 550, 582, 552, 714, 550, 1704, 552, 1706, 552, 580, 552, 712, 550, 582, 578, 568, 550, 582, 550, 714, 550, 582, 552, 582, 550, 1704, 552, 714, 552, 582, 550, 582, 578, 568, 550, 714, 550, 582, 552, 582, 550, 582, 550, 714, 552, 582, 550, 582, 550, 582, 578, 722, 552, 1704, 552, 1706, 552, 582, 550, 738, 552, 1706, 552, 1706, 550, 582, 552, 582, 578}; // UNKNOWN 4046F041
unsigned int Temp18[] = {3030, 3070, 3004, 4304, 576, 1694, 552, 580, 550, 1706, 550, 714, 550, 582, 552, 1706, 550, 580, 552, 1836, 552, 582, 578, 568, 550, 1706, 550, 1838, 550, 580, 552, 1704, 552, 1706, 550, 712, 550, 582, 552, 582, 578, 1692, 552, 714, 550, 582, 550, 582, 550, 582, 552, 714, 550, 582, 550, 582, 550, 582, 578, 698, 552, 582, 550, 582, 550, 582, 552, 1836, 552, 580, 550, 584, 550, 582, 552, 712, 578, 568, 552, 582, 550, 582, 550, 712, 552, 1706, 554, 1702, 552, 580, 550, 714, 552, 582, 578, 568, 550, 582, 552, 712, 550, 582, 550, 582, 552, 1704, 554, 710, 552, 582, 552, 582, 578, 568, 552, 712, 552, 582, 550, 582, 552, 582, 550, 714, 550, 582, 550, 582, 550, 582, 580, 722, 550, 1704, 552, 1704, 552, 582, 550, 1862, 550, 1704, 552, 1704, 552, 582, 550, 582, 578}; // UNKNOWN 1575D184
unsigned int Temp19[] = {3030, 3046, 3028, 4306, 574, 1692, 552, 582, 550, 1706, 550, 714, 552, 582, 550, 1706, 552, 580, 552, 1836, 552, 582, 578, 1690, 552, 582, 550, 714, 550, 582, 550, 1706, 550, 1704, 552, 714, 550, 582, 550, 582, 580, 1690, 552, 712, 552, 582, 550, 582, 550, 582, 550, 714, 550, 582, 550, 582, 552, 582, 578, 698, 552, 582, 552, 582, 552, 582, 550, 1836, 552, 580, 552, 582, 552, 582, 550, 714, 578, 568, 550, 582, 550, 582, 550, 714, 550, 1706, 550, 1706, 550, 582, 552, 714, 550, 582, 580, 568, 550, 584, 550, 714, 550, 582, 550, 582, 550, 1704, 552, 712, 550, 582, 550, 582, 578, 568, 550, 714, 552, 582, 552, 582, 550, 582, 550, 714, 550, 582, 550, 582, 552, 582, 578, 724, 550, 1708, 550, 1704, 552, 1704, 552, 736, 552, 1704, 552, 1706, 550, 582, 550, 582, 578};//Change it for your remote
unsigned int Temp20[] = {3030, 3044, 3030, 4304, 578, 1694, 550, 582, 552, 1708, 548, 714, 550, 582, 550, 1704, 552, 582, 550, 1838, 550, 582, 578, 1692, 550, 582, 550, 1836, 552, 582, 550, 1706, 550, 1704, 552, 712, 550, 582, 550, 582, 580, 1690, 552, 712, 552, 582, 552, 582, 550, 582, 552, 714, 550, 582, 550, 582, 550, 582, 578, 698, 552, 582, 550, 582, 550, 582, 552, 1836, 552 [... cut ...]
   irsend.sendRaw(Temp28, sizeof(Temp28) / sizeof(Temp28[0]), khz); //Send Signal to Turn On the AC 
  delay(2000);
   irsend.sendRaw(Temp29, sizeof(Temp29) / sizeof(Temp29[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);
   irsend.sendRaw(Temp30, sizeof(Temp30) / sizeof(Temp30[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);
   irsend.sendRaw(Temp16, sizeof(Temp16) / sizeof(Temp16[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);
     irsend.sendRaw(ACoff, sizeof(ACoff) / sizeof(ACoff[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);

    

> `

crankyoldgit, 2020-04-18:
@mohammadowais3701 Please don't hi-jack an existing/closed issue.

> int khz=38;

`uint16_t khz = 38;`

Also you're verging on https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#i-want-to-change-the-example-code-to-make-it-do-something-different-but-i-dont-know-how-please-help-me