crankyoldgit/IRremoteESP8266 #319: Change pin in void setup()

renat2985, 2017-10-12:
Last version. Please tell us how can change the pin in void setup().

darshkpatel, 2017-10-12:
Can you provide us with more information?
Maybe a Code snippet?

renat2985, 2017-10-12:
Example:
```
IRrecv myIR = irrecv();
void setup() {
  myIR.enableIRrecv(0);
}
```

crankyoldgit, 2017-10-12:
That's an incorrect usage example.

You need to do it using the class constructor.
See https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L87

If you want to do it in (and only in) setup(), then you are going to need to have a global pointer to the class. Initialise the class in setup and set the pointer to the class, so you can use the class (via the pointer) in the main loop etc.

The alternative approaches are:
a) Set it only once during compilation in the 'global' section. e.g. how almost every example of the code does it.
or
b) Run it only once (left as an example for you to work out how to do that, in the main loop() procedure.
e.g. something like:
```
bool first = true;
if (first) {
  IRrecv myIR = irrecv(pin_i_want_to_use);
  first = false;
}
```

NOTE: You can only have one ir receiver object. The code doesn't support multiple instances of the class.

renat2985, 2017-10-13:
```
IRrecv irrecv();
decode_results results;

bool first_run = true;

void loop() {
  if (first_run) {
    IRrecv myIR = irrecv(pin_i_want_to_use);
    first_run = false;
  }
  /*
   * rest of code etc
   */
}
```

crankyoldgit, 2017-10-13:
Try the following (completely untested):
```
IRrecv myIR;
decode_results results;

bool first_run = true;

void setup() {  
 Serial.begin(115200); 
}

void loop() {
  // Moving 'IRrecv myIR;' here may work too.
  if (first_run) {
    myIR = irrecv(4);
    irrecv.enableIRIn();  // Start the receiver
    first_run = false;
  }
  /*
   * rest of code etc
   */
}
```

I think your problem is with (I think) programming language [variable](https://www.arduino.cc/en/Tutorial/Variables) [scoping](http://en.cppreference.com/w/cpp/language/scope). Even I got the [scope](https://www.arduino.cc/en/Reference/Scope) wrong in the previous example code I suggested.

In short, if you **only** want to access a [variable](https://www.arduino.cc/en/Tutorial/Variables) in `setup()` it's fine to declare it in setup()'s [scope](https://www.arduino.cc/en/Reference/Scope). If you want to access it in both `setup()` & `loop()`, you **need** to declare it at a [scope](http://en.cppreference.com/w/cpp/language/scope) above the two, in that case it needs to be a [global](https://forum.arduino.cc/index.php?topic=370764.0). My mistake was only declaring it in the [scope](https://www.arduino.cc/en/Reference/Scope) of the `if (first_run) {}` statement, thus it wouldn't be available outside of it.

I know this might sound harsh and sorry if it does, but per our wiki: https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#things-we-are-unlikely-to-help-you-with this is falling into the category _"Teaching you how to write Arduino C/C++ code."_
You seem to be having more general programming issues, not with the library per se.

renat2985, 2017-10-13:
You can show a working example with a change pin in setup() or loop()?

crankyoldgit, 2017-10-13:
What is it that you are actually trying to do/achieve?

crankyoldgit, 2017-10-13:
Here are a couple of examples of other people using the library and setting the pin at run-time.
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P016_IR.ino#L57
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_ir_send.ino#L54

crankyoldgit, 2017-10-13:
Using those as examples, which both use the pointer method I initially described, something like this would work: (Again, I'm not testing this code, just giving you a outline etc.)

```
IRrecv *irReceiver;  // Pointer to a IRrecv object/class instance.
decode_results results;

void setup() {
  irReceiver = new IRrecv(4);  // Create a new IRrecv object. Change to what ever pin you need etc.
  irReceiver->enableIRIn(); // Start the receiver
}

void loop() {
  if (irReceiver->decode(&results)) {
    // Do stuff with the decoded results.
    irReceiver->resume();  // Continue looking for IR codes after your finished dealing with the data.
  }
}
```

If you can't work out how to do it from these examples, then you probably don't have the skills to use C++ classes and thus this library properly.

crankyoldgit, 2017-10-18:
Friendly chase up. Did you get it working?

renat2985, 2017-10-18:
Yes. Thanks.

closed 2017-10-18 by renat2985