crankyoldgit/IRremoteESP8266 #320: Encoding/decoding type of AC and some other remoted

waghekapil, 2017-10-13:
I'm using the this library to get the Panasonic AC remote IR code. It returns the different-different code for Same On/Off button with encoding protocal as UNKNOWN. I also get the Time line data, which is raw data.
**Not Working**

```
Encoding  : UNKNOWN
Code      : 88916564 (18 bits)
Timing[35]: 
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, -1300
     + 500, -1300     + 500, -1300     + 500, - 450     + 500, - 450
     + 500, - 450     + 500
uint16_t  rawData[35] = {500,450,500,450,500,450,500,450,500,450,500,450,500,450,500,450,500,450,500,450,500,450,500,1300,500,1300,500,1300,500,450,500,450,500,450,500
};  // UNKNOWN 88916564

Encoding  : UNKNOWN
Code      : 6FB84EC6 (4 bits)
Timing[7]: 
     + 150, - 950     + 200, - 350     + 150, -3300     + 150
uint16_t  rawData[7] = {150,950,200,350,150,3300,150
};  // UNKNOWN 6FB84EC6
```

**Can I get the raw data from IR code?**

For some other remotes I am getting full resul data.
**Working**
```
Encoding  : SANYO_LC7461
Code      : 152156F38C7 (42 bits)
Timing[87]: 
     +9100, -4500     + 650, - 550     + 650, -1700     + 600, - 550
     + 650, -1700     + 650, - 550     + 650, -1700     + 600, - 550
     + 650, - 550     + 650, -1700     + 600, - 550     + 650, - 550
     + 650, - 550     + 650, - 550     + 650, -1700     + 600, - 550
     + 650, -1700     + 600, - 550     + 650, -1700     + 600, - 550
     + 650, -1700     + 600, -1700     + 600, - 550     + 650, -1700
     + 600, -1700     + 600, -1700     + 600, -1700     + 650, - 550
     + 650, - 550     + 650, -1700     + 600, -1700     + 650, -1700
     + 600, - 550     + 650, - 550     + 650, - 550     + 650, -1700
     + 600, -1700     + 600, - 550     + 650, - 550     + 650, - 550
     + 650, -1700     + 600, -1700     + 600, -1700     + 600
uint16_t  rawData[87] = {9100,4500,650,550,650,1700,600,550,650,1700,650,550,650,1700,600,550,650,550,650,1700,600,550,650,550,650,550,650,550,650,1700,600,550,650,1700,600,550,650,1700,600,550,650,1700,600,1700,600,550,650,1700,600,1700,600,1700,600,1700,650,550,650,550,650,1700,600,1700,650,1700,600,550,650,550,650,550,650,1700,600,1700,600,550,650,550,650,550,650,1700,600,1700,600,1700,600
};  // SANYO_LC7461 152156F38C7
uint32_t  address = 0xA90;
uint32_t  command = 0x38;
uint64_t  data = 0x152156F38C7;

```

I'm using MQTT for backend communication with hardware device. So, I want to send the as much as less data to the server. :)

crankyoldgit, 2017-10-13:
Hi, 
 This library does not support decoding of Panasonic A/C remote signals. It also doesn't support/understand sending of them either. A/C unit remotes are [very complex](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#things-we-are-unlikely-to-help-you-with). So, you will probably be limited to replaying (via sendRaw() etc) any codes, unless you find or care to write a module that understands them.

Some hints on how to get a good capture of the commands can be found [here](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#problems-decoding) and in the IRrecvDumpV2 [code](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L30).

Looking at your Panasonic captures, they don't look good TBH. I would play with the mentioned settings to see if you can get something that looks vaguely repeatable. i.e. Pressing the button gives you a _single_ capture that has similar values over and over again for each press.

FYI, we recently added some example code for MQTT, it may be of use to you:
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino

You are also using an old version of the library to do the captures you've done so far, I strongly suggest you update to the latest.

Happy hunting.

crankyoldgit, 2017-10-13:
Oh, I missed a question. In short, if it is marked as UNKNOWN, no, you can't work out the 'rawData' from the _code_ value.

From the troubleshooting guide:

> When it detects it as Unknown, the NN-bit code value reported is useless for sending purposes. In fact, it's pretty poor as a unique value for decoding too. In short, don't rely on it for anything.

crankyoldgit, 2017-10-14:
I've created a FAQ item on this topic which may be of help to you: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code

crankyoldgit, 2017-10-18:
@waghekapil How did it go?

crankyoldgit, 2017-10-25:
@waghekapil Friendly ping. If I don't hear anything with in another week, I'll assume you're all sorted now.

waghekapil, 2017-10-30:
Hi @crankyoldgit ,

Thanks for your reply. Due to festivals and switched to another project I didn't get any change to check this.

My apologies.

Here is the update from my side:

1) I updated the library.

2) Ran IRrecvDumpV2. Got the encoding as UNKNOWN and got the rawdata of Panasonic AC remote.

3) I ran IRsendDemo example with Panasonic AC remote raw data. It worked fine :)

Now what I noticed is that each time of same, remote same button I'm getting different raw data. Is there any problem or it is ok?

Thanks! again

crankyoldgit, 2017-10-30:
Like with everything, it depends. You are unlikely to get the exact same values each time. However, you should see the same pattern of numbers. e.g. "9024, 4488, 512, 1008, 524, 1586, 508, ..." is effectively the same as "8912, 4602, 480, 998, 498, 1602, 502, ...". That is, if the numbers are with in +/- 20% then it is probably the "same". It's the sequence of similar values, and their .. um .. up & down-ness (for want of a better description) that actually matters. I hope that answers your question.

However, you've got a raw-code that works for you, then huzzah! That's kind of all that matters in the end.

closed 2017-10-30 by crankyoldgit