crankyoldgit/IRremoteESP8266 #321: how to send unknow coding 32 bit

aleale99, 2017-10-14:
i need to send an unknow format 32 bit code

i do not find a specific function, i have only the hex codes  es. 4B3AB9D1
in docs you writed:   IRsend.sendFoobar(0xdeadbeef);
but during compiling i got         'class IRsend' has no member named 'sendFoobar'

which is the right mode?
obvliusly I don't have the raw string

crankyoldgit, 2017-10-14:
You can't send an **unknown** code command.
See point #2 in https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#problems-decoding

The sendFoobar() thing is just an example/placeholder of the function you would have been using in your own code. e.g. sendSAMSUNG() etc. There is no sendFoobar()

The only way to send an unknown command is to send the raw string using the `sendRaw()` method. See the example code for how to use `sendRaw()`.
You can get the raw string by using the IRrecvDumpV2 program.

crankyoldgit, 2017-10-14:
FYI, I just added a [FAQ answer](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code) based on this issue.

closed 2017-10-14 by crankyoldgit