crankyoldgit/IRremoteESP8266 #325: Using more than one IR LED

leobel96, 2017-11-01:
Hello, I'd like to use this library to create something similar to Logitech Harmony: a hub that can be put in the center of a room and that can control every device using MQTT server. To make it possible I think I should use more than only one IR sender to make sure the signal will arrive to every device in all directions. Is it a good idea to connect more than one IR led or there is the risk of interferences and distorted information? If yes, should I connect all the LEDs in parallel to the same transistor or I should use a transistor for every led? Have you got suggestions on how many LEDs should I use to cover every direction? Thank you!!!

crankyoldgit, 2017-11-01:
I do almost exactly what you are asking about.
I have a D1 mini with 4 IR LEDs connected to transistors, which are triggered off a single GPIO. So, yes. It's do-able. I seem to have no problem with interference, however I have them all facing at least 90 degrees to each other. However, multiple in the same directions shouldn't matter if they are triggered at the same time, and are the same type.

I went with a single transistor per IR LED, however, in principle you should be able to do it in parallel (I am not an electrical engineer, so don't take my word for it) but your mileage may vary.

I remember having problems driving 3 IR LEDs with a single transistor, but 2 seemed to work okay. I had plenty of single NPN transistors, so one per was just easy. I bought a pack of 100, so I had plenty to spare.

I use a small box with a D1 mini, and 4 LEDs, one per side (Top, Front, Left, Right). As the unit is sitting on top of a cabinet against a wall, it doesn't need one for the rear as it is facing a wall. I have an A/C unit sitting above it, hence the 'Top' one.

As for the code, I'm using the exact code in the [IR MQTT Server examples](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino) for the device. Works like a charm for me.

crankyoldgit, 2017-11-01:
FYI, the specs for your IR LED should tell you the coverage angle(s), so if you need full coverage, work off that as a start.

crankyoldgit, 2017-11-01:
It really depends on the orientation and distance(s) involved in your setup. I've got mine up high, and the front and side(s) lined up roughly with the devices, plus I have power on the walls, not in the middle of the room. Having it as far as practical yet still in range means the coverage angle results in a fairly wide area. (Trigonometry FTW!) I'd suggest trialling it with a breadboard to work out what angles you needs, and how many IR LEDs etc. You can also get IR LEDs with wider angles too.

On the plus side, if you need multiple devices, at least IR LEDs, transistors, and ESP8266s are very cheap. ;-) Note: Unlike multiple IR LEDs on a single ESP8266, having multiple IR-enabled ESP devices **will** potentially cause you interference if you have them respond to the same MQTT topic/command(s) as they are likely to get the MQTT messages at slightly different times, hence overlap. If you are going to have multiple IR transmitter devices in the same _room_, I'd recommend having separate topics and control them individually etc.

crankyoldgit, 2017-11-01:
If you have plenty of spare IR LEDs, you can try to widen the beam by doing something like this: https://www.instructables.com/answers/how-do-I-give-an-LED-a-wider-beam-spread/

leobel96, 2017-11-01:
Perfect, I'll make some tests. Thank you!!!

closed 2017-11-01 by leobel96