crankyoldgit/IRremoteESP8266 #326: Controlling a 315Mhz Radio Transmitter with this Library

NorthernMan54, 2017-11-05:
Mark,

This is not an issue, but just a placeholder to share with everyone my usage of your library to control 315Mhz Radio transmitters rather than LED's, and ultimately control my Fireplace.  I was able to directly use your library and Michael Higgins ESP8266-HTTP-IR-Blaster package.  After determining what the bit pattern was for the radio signal, I just used the raw mode of your library to control the transmitter, and transmit the same bit pattern.  The only minor tweak I needed to make, was using a frequency of 500Khz which had the side effect of not modulating the LED/Transmitter.

My integration efforts are documented here.

https://www.instructables.com/id/Control-Your-Fireplace-With-HomeKit-and-Alexa/

As this is not truly as issue, but an FYI for the community it can closed.

crankyoldgit, 2017-11-06:
/me blinks. Wow. That's really cool to know and think about.
At 500KHz, that's a period of 2ms, or 1ms on, 1ms off. I'm kind of speechless that our library works at that level/speed. It certainly wasn't designed to go down to that level.

Surely there is a proper Arduino Library for the 315MHz transmitter module, no? e.g. Virtual Wire or RadioHead.

NorthernMan54, 2017-11-06:
David,

What I found was that at 500khz, the duty was very long and not short, probably due to a math issue in the period calculation.  I needed the gpio pin held high, for the total length of the bit transmission.  I had started down the path of a pull request for 0 kHz to be unlimited duty, but found that quirk with 500 and didn’t need to.

As to why your library and not others, I was using this wrapper, https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster, for other things so it made coding easier.

crankyoldgit, 2017-11-06:
Yes, I can see how it could breakdown at around the 1uSec level.
Fyi, you can set the [duty cycle](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L80)  but yes, for backward compatibility sendRaw uses a 50% duty cycle, and which IIRC the duty cycle calcs do err on the 'on' side more than the 'off' when it comes to rounding.

No worries about the library usage, I was just curious as to why not a more specialised library. Again I'm kind of thrilled my code works (if only sort of) for that kind of use.

Thanks for the feedback. Certainly makes a change from the typical 'issues'. ;-)

closed 2017-11-06 by crankyoldgit

markszabo, 2017-11-06:
Wow, great! Thank you for sharing your work with us, it is truly amazing :)