crankyoldgit/IRremoteESP8266 #327: Send ir commands to AC system s

leobel96, 2017-11-06:
Hi. Is it possible to send it commands using MQTT server to AC systems different from Kelvinator such as Fujitsu, Daikin,...? An update of the MQTT server example adding other AC systems would be great!! Thank you!

crankyoldgit, 2017-11-06:
Yes. However, you'll have to add the code to the supplied example code. It should be fairly similar to the Kelvinator example.

leobel96, 2017-11-06:
The problem is that I also should add a function similar to "parseStringAndSendKelv(code_str)" and add it inside the "switch(ir_type)", it's not enough to modify the MQTT request example already done...

crankyoldgit, 2017-11-06:
Yes, that's what I mean. You should be able to copy and convert the existing method `parseStringAndSendKelv()` fairly easily. e.g. Using the correct final call for your A/C unit, and setting the correct sizes/values etc for your A/C unit. Most of the hard work is done for you. If you have problems when you implement it, feel free to send us a link to your github repository/branch and I can offer advice.

leobel96, 2017-11-06:
Ok, I'll try and ask you if I'll encounter problems. Thanks!

closed 2017-11-06 by leobel96