crankyoldgit/IRremoteESP8266 #328: [Request] Add Toshiba AC support

leobel96, 2017-11-06:
Hello, sorry for the too many requests I'm opening but I find great potentialities in this project also because we are entering in the era of IoT and this library can be easily used to control our "stupid and old" devices using internet and automations. However I opened this issue because I noted that the support for Toshiba AC (the one I'va got) is missing but I've found a library that should make it easy to add support for it. I've seen that you have already used it some time ago so I think it shouldn't be hard. https://github.com/r45635/HVAC-IR-Control/pull/7

crankyoldgit, 2017-11-06:
That looks fairly straight forward. I can probably knock that over in a few days of effort, assuming you're willing to do some testing for me etc. Have you tried out their code yet to see if it works on your Toshiba A/C unit?

leobel96, 2017-11-06:
That would be great! Yes, I can make some testing for you and no, I've not tested it yet.

crankyoldgit, 2017-11-06:
Can you please test their code to confirm it works with your device? Otherwise there is little point etc. :-)

Also can you detail what specific model number of Toshiba A/C you have, and what the model number is for the remote too please.

leobel96, 2017-11-06:
I'll come back home this evening, then I'll look for model numbers. I don't understand in their examples what are DHT pin and Reset pin, what should I connect to them and what are the mandatory fields in the function sendHvacToshiba()

crankyoldgit, 2017-11-06:
DHT is typically a Digital Humidity & Temperature sensor. Skimming over [their code](https://github.com/r45635/HVAC-IR-Control/blob/master/HVAC_ESP8266/HVAC_ESP8266T.ino) it looks like it isn't used. You can just ignore it.
As for the reset(RST) pin, if you change the line `ESP.deepSleep(...);` to a suitably long `delay();`, you can ignore it for testing purposes. The reset pin gets hooked to a GPIO (I forget which at the moment) which is triggered by a timer, which wakes/reboots the ESP module.

Their code assumes a working IR LED circuit triggered by the D0 pin.

leobel96, 2017-11-06:
Perfect, I'll try it, Thanks

leobel96, 2017-11-06:
I've just tested it and it seems to work!

crankyoldgit, 2017-11-07:
Seeing as their code seemed to work, I've constructed a branch that seems to work from the superficial tests I can do without the actual device.

Branch is [here](https://github.com/markszabo/IRremoteESP8266/tree/toshiba_ac).
Example code (untested) is [here](https://github.com/markszabo/IRremoteESP8266/tree/toshiba_ac/examples/TurnOnToshibaAC).

Can you please try it out **AND** tell me the model numbers of the A/C Unit and the Remote so we can record what it is known to work on etc?

leobel96, 2017-11-07:
Fantastic, I'll try it in many hours. Anyway the unit model is toshiba ras-b13n3kv2. I can't find the remote model neither written on it nor on the manual but I have seen it's the same for all the family "akita evo II" to which the AC belongs. Thank you, great work!!

leobel96, 2017-11-07:
I've tested the example and it works fine, anyway if I try modifying the example in [this](https://pastebin.com/TKv8uN79) way, it should switch off but it doesn't switch off

crankyoldgit, 2017-11-07:
I thought I saw an optimisation w.r.t how on/off state is done compared to the reference code. Looks like I was wrong.
I've made another [commit](https://github.com/markszabo/IRremoteESP8266/commit/324985398b9f3abc796284f6e8d940ebff89d1c3) to the branch. Please try it out. If that doesn't work, we are probably going to have to create & turn on some special debugging features.

leobel96, 2017-11-07:
Now verything works fine. Very good work!!!

crankyoldgit, 2017-11-08:
Good to hear. I'll start the process of merging it into the next version of the library (v2.3.0)

closed 2017-11-10 by leobel96