crankyoldgit/IRremoteESP8266 #331: Integration with Home Assistant

leobel96, 2017-11-07:
Home Assistant (https://home-assistant.io) is an open source project that lets you make your home smarter connecting your devices, making automations and allowing you to handle everything from a nice UI and very few code lines (https://home-assistant.io/getting-started/configuration/). I think IRremoteESP8266 could have its maximum expression inside this project using the already supported [MQTT connection](https://home-assistant.io/components/mqtt/) as part of the already available [895 components](https://home-assistant.io/components/) . Home Assistant's developers can be found in [this](https://discord.gg/c5DvZ4e) chat. Thank you.

crankyoldgit, 2017-11-08:
You can already use [Home Assistant](https://home-assistant.io) via (as you pointed out) the [MQTT component](https://home-assistant.io/components/mqtt/) if you use the [MQTT example code](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino). Therefore, there is really no need to do anything further for integration IMHO.

e.g. I've been [using](https://github.com/crankyoldgit/Home-AssistantConfig/blob/master/configuration.yaml#L99) Home Assistant with it for about 6 months.

closed 2017-11-08 by crankyoldgit

leobel96, 2017-11-08:
I mean something more integrated, with an interface that lets you create a frontend UI similar to a remote where you can select your TV model and the remote magically appears on the homepage, where also the voice extensions such as "dialogflow" understand what is a "switch off" command without creating a flow for every button. I don't know if I made myself clear...

crankyoldgit, 2017-11-08:
You can build an interface out of existing Home Assistant tools & features, plus even if you couldn't, requesting such would be something you should do on the Home Assistant feature request forums. Not here.

As for voice control. I already do that with my HA system and IRremoteESP8266 using Amazon Echo/Alexa & Google Assistant/Home.

I'm _not_ using the HA built-in MQTT server. I'm using [mosquitto](https://mosquitto.org/) running on the same SBC that I have HA running on. I had a MQTT server before I was running HA, otherwise I'd probably have used the built-in one.

leobel96, 2017-11-08:
I've already requested it in feature request forum but Home assistant's developers are Very busy and said that can't implement this feature because they have a very long list of features to be implemented and bugs to be Untangled. Anyway I'll try using mosquitto. Is it normal that until a stable mqtt connection is established I can't access to my esp8266 server interface using the MQTTserver example? I mean it connects to WiFi but connection results unavailable when I put my esp8266 IP inside URL field in my browser

crankyoldgit, 2017-11-08:
Yes. The code does tend to lock up a bit while it is trying to establish an MQTT session. Normally, your MQTT session from the ESP to the MQTT server should set up within a second or two of start up, and stay connected for hours/days. The example code is designed for good/ideal conditions. If you've got a poor wifi connection and trouble talking to your MQTT server, the web interface will be quite unresponsive at times.

leobel96, 2017-11-08:
Good, I'll try to fix the MQTT problems hoping it'll also fix the problems inside the browser. Thank you.

crankyoldgit, 2017-11-08:
For reference, my ESP8266 D1 mini setup seems to lose its session to the MQTT server about once a day on average. It quickly reestablishes that session in probably less than a second or so. Out side of those times, MQTT messages and HTTP messages/responses all seem to be sub-second affairs for me.

leobel96, 2017-11-08:
First time I configured mqtt on HA and I flashed the MQTT server example, everything worked fine. Then I flashed other scripts (such as the Toshiba library I tested) and from when I reflashed the old script, it gives me problems. Tried reflashing it many and many times but nothing changes. I'll try using mosquitto.

crankyoldgit, 2017-11-08:
Hmm. You may need to reset the SPIFFs. Not sure how. Or did you remember to set/increase MQTT_MAX_PACKET_SIZE correctly?

There should be plenty of debug info on the serial port to tell what is going on with the MQTT connection if it is dropping out.

leobel96, 2017-11-08:
Yes, I Remembered it. In debug I obtain error RC:-2 or RC:-4 for MQTT

crankyoldgit, 2017-11-09:
I think we have moved beyond the scope of the original issue. Please create a new issue for this. Also, please include the full output from the serial port as that will be helpful.

leobel96, 2017-11-09:
Yes, you are right, also because deeper I go and harder and stranger it becomes. When I'll have 10 minutes I'll open a new issue explaining everything

crankyoldgit, 2017-11-09:
FYI,
 those 'rc=' values are the results of the PubSubClient's connect() function. They appear to be defined at:
https://github.com/knolleary/pubsubclient/blob/v2.6/src/PubSubClient.h#L45
```
#define MQTT_CONNECTION_TIMEOUT     -4
#define MQTT_CONNECT_FAILED         -2
```
Just checking the obvious, you've double checked the IP address and port number (and password etc if you are using it) of the MQTT parameters in the example code? e.g. Did you overwrite the example code when you used the other branch?

leobel96, 2017-11-09:
I tried using the example code and modifying only the line 
#define MQTT_SERVER "test.mosquitto.org" with no success. Anyway I did a lot of attempts with different controllers obtaining different results. I have a lot to say. I think it's better to close this issue and talk about it in another issue. I have seen there are 9 hours of time zone between me and you. Here it's night. Tomorrow I'll create the issue and I'll try to explain it the clearly. Bye

leobel96, 2017-11-09:
It was a problem of credentials. Now everything is OK.