crankyoldgit/IRremoteESP8266 #341: How to filter garbage rawdata?

waghekapil, 2017-11-16:
### Version/revison of the library used
_ v2.2.1

### Expected behavior
_ Installed updated library and run the IRrecvDumpV2. 

### Actual behavior
_Without pressing any remote button it is learning some garbage rawdata like:

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_

Encoding  : UNKNOWN
Code      : 4CB0FADC (3 bits)
Timing[5]: 
   +   394, -    96,    +   196, - 11552,    +   160
uint16_t rawData[5] = {394, 96,  196, 11552,  160};  // UNKNOWN 4CB0FADC

Encoding  : UNKNOWN
Code      : A362B57A (7 bits)
Timing[13]: 
   +   180, -  3880,    +   130, -  1180,    +   130, -  3120,    +    92, -  2412, 
   +   176, -  1100,    +   146, -  1180,    +   332
uint16_t rawData[13] = {180, 3880,  130, 1180,  130, 3120,  92, 2412,  176, 1100,  146, 1180,  332};  // UNKNOWN A362B57A

Encoding  : UNKNOWN
Code      : E2CBE30C (9 bits)
Timing[17]: 
   +    96, -   214,    +   452, -   518,    +   158, -   170,    +   216, -   756, 
   +   436, -   892,    +   108, -  1206,    +   304, -  3626,    +   400, -  4644, 
   +   122
uint16_t rawData[17] = {96, 214,  452, 518,  158, 170,  216, 756,  436, 892,  108, 1206,  304, 3626,  400, 4644,  122};  // UNKNOWN E2CBE30C

Encoding  : UNKNOWN
Code      : AFEFBB42 (8 bits)
Timing[15]: 
   +   132, -   410,    +   158, -  4914,    +   134, -  2504,    +   176, -   936, 
   +   270, -   256,    +   164, -  2002,    +   144, -   432,    +    90
uint16_t rawData[15] = {132, 410,  158, 4914,  134, 2504,  176, 936,  270, 256,  164, 2002,  144, 432,  90};  // UNKNOWN AFEFBB42

Encoding  : UNKNOWN
Code      : E1110F58 (5 bits)
Timing[9]: 
   +   160, -   744,    +   154, -  1070,    +   240, -  1518,    +   108, -  6642, 
   +   148
uint16_t rawData[9] = {160, 744,  154, 1070,  240, 1518,  108, 6642,  148};  // UNKNOWN E1110F58

Encoding  : UNKNOWN
Code      : 2D004F7C (8 bits)
Timing[15]: 
   +   210, -   982,    +   224, -  2360,    +   124, -  1170,    +   384, -   880, 
   +   774, -   532,    +   164, -  1130,    +   326, -    92,    +   108
uint16_t rawData[15] = {210, 982,  224, 2360,  124, 1170,  384, 880,  774, 532,  164, 1130,  326, 92,  108};  // UNKNOWN 2D004F7C

### Steps to reproduce the behavior
_ Just run IRrecvDumpV2.ino

#### Circuit diagram and hardware used (if applicable)
_TSOP1738 IR Receiver

Can you please tell me how to filter these garbage rawdata?

crankyoldgit, 2017-11-16:
I think you are getting noise. That just happens depending on your environment, circuit, and sensor etc.

The example code (dumpV2) errs on the side of reporting everything it sees as _some_ people are expecting weird signals. The software can't tell what it is you are looking for in advance in all cases.

You've got some options to minimise it though:

1. You can change the code to ignore UNKNOWN decodes.
i.e. Only display it if the type is **not** `UNKNOWN`.
Downside: if the library can't decode it in to a known type, you may miss what ever signal you want to get.

2. Change the value [here](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L559) from `6` to something higher. e.g. 20, as the highest noise value you've seen was 17 in length.
Downside: If the signal you want is UNKNOWN  *and* shorter than the value chosen, you won't get any idea it happened, PLUS you may still get noise reported **IF** it's longer than the value you chose.

Hope that helps.

waghekapil, 2017-11-16:
@crankyoldgit thanks for reply. Yes, it helped :)

I'll go with second option as I'm also learning AC remotes and their decodes are UNKNOWN.

I'm closing this now.

closed 2017-11-16 by waghekapil