crankyoldgit/IRremoteESP8266 #342: I want to control Ac of GREEE and KENWOOD

Najab, 2017-11-16:
When i read ir codes(raw data) from Kenwood Ac remote it should be different array size every time like this:
Encoding  : UNKNOWN
Code      : 37E14664 (152 bits)
uint16_t rawData[303] = {9018, 4526,  570, 1676,  580, 1670,  582, 548,  584, 552,  578, 562,  574, 566,  584, 562,  580, 1676,  578, 546,  584, 1668,  578, 1674,  588, 548,  584, 554,  582, 560,  594, 552,  588, 546,  580, 1668,  580, 1668,  580, 554,  580, 554,  584, 554,  584, 560,  580, 566,  582, 556,  578, 546,  576, 1674,  572, 560,  580, 556,  580, 1680,  578, 564,  586, 560,  572, 1684,  598, 526,  574, 556,  572, 560,  574, 562,  576, 562,  584, 560,  610, 536,  596, 540,  590, 560,  558, 544,  582, 552,  582, 554,  578, 560,  570, 574,  548, 598,  588, 530,  554, 7992,  564, 560,  568, 1684,  580, 1674,  574, 562,  580, 1680,  584, 558,  578, 568,  584, 1676,  580, 1668,  584, 544,  586, 548,  576, 560,  578, 560,  582, 1682,  576, 570,  580, 556,  592, 534,  578, 550,  578, 552,  588, 548,  582, 558,  584, 560,  580, 566,  576, 560,  584, 538,  578, 552,  582, 550,  580, 556,  576, 562,  582, 560,  580, 566,  578, 558,  580, 546,  576, 552,  580, 552,  582, 556,  576, 562,  586, 556,  576, 570,  574, 560,  594, 532,  578, 552,  578, 554,  582, 554,  582, 556,  578, 566,  578, 568,  578, 558,  582, 544,  578, 552,  580, 552,  578, 558,  582, 558,  580, 564,  580, 566,  582, 554,  578, 546,  578, 1674,  578, 1676,  576, 560,  578, 562,  584, 1682,  578, 568,  580, 538,  578, 7970,  578, 546,  574, 554,  582, 550,  586, 550,  574, 564,  576, 566,  580, 566,  580, 554,  580, 544,  578, 1672,  578, 554,  578, 554,  588, 550,  580, 562,  582, 564,  576, 558,  572, 554,  576, 550,  582, 550,  584, 550,  582, 556,  592, 550,  576, 568,  578, 558,  586, 538,  586, 542,  584, 548,  574, 564,  580, 558,  584, 558,  576, 570,  580, 556,  576, 550,  582, 546,  584, 550,  582, 1676,  580};  // UNKNOWN 37E14664

Encoding  : UNKNOWN
Code      : 2968E045 (153 bits)
uint16_t rawData[305] = {9016, 4526,  584, 1664,  578, 1674,  574, 558,  566, 570,  570, 568,  576, 566,  578, 570,  574, 1682,  578, 550,  572, 1676,  574, 1678,  576, 562,  570, 568,  570, 572,  582, 564,  574, 560,  574, 1672,  574, 1678,  576, 558,  582, 554,  576, 564,  580, 562,  576, 570,  578, 558,  574, 552,  574, 1674,  578, 556,  572, 564,  578, 1682,  584, 560,  584, 562,  578, 1678,  582, 546,  580, 550,  584, 550,  572, 564,  574, 564,  576, 568,  576, 568,  580, 556,  574, 550,  582, 548,  576, 556,  578, 558,  574, 566,  574, 566,  578, 568,  578, 540,  576, 7968,  586, 546,  578, 1668,  584, 1672,  584, 554,  578, 1680,  582, 564,  584, 560,  580, 1674,  586, 1664,  584, 546,  580, 552,  576, 560,  582, 558,  574, 1688,  584, 564,  582, 554,  568, 558,  574, 554,  582, 550,  582, 554,  576, 564,  588, 554,  576, 570,  570, 566,  584, 540,  574, 556,  582, 548,  580, 558,  574, 566,  580, 562,  574, 574,  578, 558,  570, 554,  574, 554,  574, 558,  578, 558,  582, 558,  584, 558,  578, 568,  574, 562,  582, 544,  576, 552,  572, 560,  578, 558,  582, 558,  578, 564,  578, 568,  582, 554,  580, 544,  576, 554,  576, 558,  586, 550,  580, 562,  570, 572,  582, 564,  572, 564,  578, 548,  570, 1680,  582, 1674,  580, 558,  578, 562,  574, 1690,  578, 568,  584, 536,  574, 7972,  586, 546,  580, 548,  576, 556,  578, 558,  578, 560,  578, 564,  580, 566,  580, 558,  580, 546,  578, 1670,  582, 552,  578, 558,  580, 562,  572, 570,  578, 568,  576, 562,  578, 548,  574, 556,  580, 552,  576, 560,  574, 564,  584, 560,  578, 570,  576, 558,  582, 544,  578, 606,  450, 628,  582, 554,  582, 558,  574, 568,  582, 562,  582, 554,  570, 556,  572, 554,  578, 554,  576, 1678,  578, 1684,  190};  // UNKNOWN 2968E045

When i read code from GREE ac remote it gives me different arrays like this:

Encoding  : UNKNOWN
Code      : DF5AE5D4 (37 bits)
Timing[73]: 

uint16_t rawData[73] = {9050, 4462,  670, 536,  676, 1630,  676, 528,  676, 1632,  672, 1636,  682, 524,  670, 536,  678, 528,  674, 530,  678, 1630,  674, 1632,  682, 522,  680, 526,  682, 524,  680, 524,  678, 1630,  678, 1628,  680, 524,  676, 1632,  682, 524,  682, 524,  680, 1626,  678, 1628,  682, 524,  674, 532,  676, 528,  676, 530,  678, 528,  674, 1634,  682, 522,  674, 1634,  676, 528,  682, 524,  686, 1620,  686, 520,  680};  // UNKNOWN DF5AE5D4

Encoding  : UNKNOWN
Code      : 9501ACEA (66 bits)
uint16_t rawData[132] = {534, 674,  532, 680,  524, 682,  524, 678,  528, 684,  1622, 680,  526, 678,  528, 680,  526, 678,  526, 678,  528, 676,  1630, 682,  522, 682,  526, 672,  532, 678,  528, 682,  522, 684,  522, 676,  528, 674,  532, 678,  528, 676,  528, 676,  530, 676,  530, 684,  522, 668,  1638, 680,  1626, 678,  528, 678,  528, 680,  39964, 9014,  4468, 686,  520, 682,  1624, 678,  528, 680,  1628, 680,  1628, 680,  526, 678,  526, 682,  524, 678,  526, 678,  1628, 678,  1630, 684,  522, 670,  538, 656,  544, 680,  526, 688,  1620, 680,  1626, 678,  530, 670,  1636, 674,  532, 676,  530, 674,  1632, 680,  1626, 676,  530, 678,  528, 676,  530, 680,  524, 674,  532, 682,  522, 674,  1632, 682,  1626, 682,  524, 676,  530, 674,  1634, 674,  530, 682 };  // UNKNOWN 9501ACEA

Encoding  : UNKNOWN
Code      : 3D5214B1 (30 bits)
uint16_t rawData[59] = {676, 530,  680, 526,  678, 526,  684, 522,  682, 524,  678, 526,  678, 528,  682, 524,  682, 1626,  678, 1628,  684, 1624,  682, 524,  680, 1626,  680, 1628,  682, 1624,  676, 528,  678, 530,  680, 1626,  686, 520,  680, 526,  674, 532,  678, 1628,  682, 524,  684, 522,  684, 522,  686, 1620,  672, 1634,  674, 532,  676, 1630,  680};  // UNKNOWN 3D5214B1

When i read code from SINGER ac remote it gives me array of 227 size it's Ok for controlling and i can control Ac with this code:

Encoding  : UNKNOWN
Code      : CED21653 (114 bits)
uint16_t rawData[227] = {3270, 1426,  642, 952,  618, 978,  596, 222,  578, 240,  566, 280,  542, 1018,  556, 286,  526, 294,  524, 1058,  520, 1062,  526, 310,  524, 1044,  544, 300,  518, 300,  518, 1066,  520, 1064,  524, 296,  520, 1064,  520, 1062,  526, 306,  532, 298,  522, 1062,  526, 292,  524, 312,  524, 1048,  544, 294,  526, 292,  528, 290,  546, 296,  530, 288,  530, 288,  548, 294,  526, 294,  524, 312,  534, 294,  528, 290,  526, 314,  526, 296,  522, 296,  526, 318,  528, 290,  524, 296,  526, 1056,  520, 312,  538, 292,  526, 1058,  542, 278,  502, 340,  514, 1064,  514, 1052,  544, 296,  524, 294,  524, 294,  550, 292,  534, 284,  524, 320,  524, 1056,  530, 294,  524, 1060,  528, 1054,  526, 296,  526, 292,  528, 310,  532, 292,  530, 296,  522, 1056,  530, 312,  524, 294,  528, 292,  532, 306,  528, 294,  530, 294,  552, 286,  526, 292,  526, 296,  548, 294,  524, 294,  528, 292,  548, 294,  532, 288,  532, 312,  522, 296,  532, 286,  520, 312,  534, 296,  528, 290,  524, 312,  536, 290,  520, 306,  540, 298,  530, 288,  528, 292,  550, 292,  526, 292,  526, 294,  538, 306,  526, 290,  526, 314,  532, 292,  520, 296,  520, 306,  552, 286,  522, 296,  526, 318,  526, 1058,  528, 1048,  534, 294,  522, 1060,  528, 292,  524, 298,  522, 1058,  530, 308,  532};  // UNKNOWN CED21653

But i can't control both Ac(GREE, KENWOOD) from these codes somebody help me for this issue Please.
Thank You in advance.

crankyoldgit, 2017-11-16:
I'm going to point you to the FAQ for some things to try to let you help yourself.
i.e. https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#the-libraryexample-code-doesnt-understand-my-air-conditioner-remote-

Looking at the data you have provided, I'm fairly certain the GREE one(s) will benefit from the increased `TIMEOUT` value. Possibly/probably even the Kenwood one too. For example: The low value of `TIMEOUT` you are using is causing the single message to be split into multiple messages. If you are relying on the sendRaw() method for it, you want it as a single _big_ message.

I'm glad the `sendRaw()` method works for your Singer AC as quickly looking at that data, it isn't a simple protocol to reproduce.

There is some development work going on for decoding (some) A/C units in the [v2.3.0-dev branch](https://github.com/markszabo/IRremoteESP8266/tree/v2.3.0-dev) which may be of help. But it is still in development, it may crash/do odd things etc. Use at your own risk etc etc.

crankyoldgit, 2017-11-16:
What value of `TIMEOUT` are you using?

Najab, 2017-11-16:
147 for SANYO, 227 for SINGER, 211 for DAIKOOL, 229 for HAIER

crankyoldgit, 2017-11-16:
That is the length of the raw data, not the `TIMEOUT` value. See the FAQ for this.

Najab, 2017-11-16:
Sorry TIMEOUT is 15U. I'm using your library V2

crankyoldgit, 2017-11-16:
What does the FAQ suggest you should do with respect to the `TIMEOUT` value?
Hint: The line underneath it in the code also suggests what to do.

Najab, 2017-11-16:
When i change TIMEOUT 15U to TIMEOUT 90U and read rawData from Singer Ac remote it gives me same data and i can also control ac with this code:
Encoding  : UNKNOWN
Code      : 994C0003 (114 bits)
Timing[227]: 
   +  3200, -  1494,    +   580, -   998,    +   570, -  1026,    +   548, -   274, 
   +   544, -   300,    +   526, -   294,    +   518, -  1064,    +   524, -   296, 
   +   522, -   324,    +   522, -  1054,    +   534, -  1032,    +   548, -   294, 
   +   522, -  1058,    +   528, -   292,    +   528, -   290,    +   520, -  1062, 
   +   522, -  1064,    +   552, -   294,    +   522, -  1038,    +   552, -  1030, 
   +   550, -   294,    +   528, -   290,    +   522, -  1060,    +   530, -   290, 
   +   550, -   294,    +   528, -  1052,    +   530, -   290,    +   534, -   284, 
   +   556, -   288,    +   524, -   294,    +   524, -   292,    +   552, -   294, 
   +   528, -   290,    +   526, -   292,    +   540, -   304,    +   520, -   296, 
   +   524, -   294,    +   554, -   290,    +   532, -   286,    +   534, -   284, 
   +   552, -   292,    +   526, -   290,    +   556, -   290,    +   528, -  1032, 
   +   552, -   290,    +   522, -   296,    +   524, -  1058,    +   530, -   290, 
   +   554, -   290,    +   534, -  1048,    +   530, -  1054,    +   532, -   290, 
   +   520, -   296,    +   556, -   288,    +   518, -   300,    +   528, -   288, 
   +   554, -   292,    +   530, -  1050,    +   524, -  1060,    +   532, -   288, 
   +   526, -  1054,    +   532, -   290,    +   538, -   280,    +   554, -   290, 
   +   524, -   292,    +   550, -   294,    +   524, -  1058,    +   528, -   292, 
   +   528, -   290,    +   538, -   306,    +   530, -   288,    +   526, -   290, 
   +   552, -   322,    +   454, -   334,    +   528, -   290,    +   548, -   296, 
   +   524, -   294,    +   530, -   290,    +   554, -   290,    +   522, -   296, 
   +   528, -   316,    +   524, -   294,    +   524, -   294,    +   542, -   276, 
   +   552, -   292,    +   528, -   290,    +   542, -   276,    +   544, -   300, 
   +   522, -   296,    +   554, -   290,    +   530, -   288,    +   524, -   292, 
   +   552, -   292,    +   524, -   294,    +   524, -   294,    +   556, -   288, 
   +   524, -   294,    +   536, -   282,    +   554, -   288,    +   532, -   286, 
   +   528, -   290,    +   546, -   298,    +   516, -   302,    +   526, -   292, 
   +   550, -   294,    +   532, -  1050,    +   522, -   298,    +   540, -   278, 
   +   556, -  1028,    +   546, -   298,    +   520, -   298,    +   520, -  1060, 
   +   526, -   294,    +   562
uint16_t rawData[227] = {3200, 1494,  580, 998,  570, 1026,  548, 274,  544, 300,  526, 294,  518, 1064,  524, 296,  522, 324,  522, 1054,  534, 1032,  548, 294,  522, 1058,  528, 292,  528, 290,  520, 1062,  522, 1064,  552, 294,  522, 1038,  552, 1030,  550, 294,  528, 290,  522, 1060,  530, 290,  550, 294,  528, 1052,  530, 290,  534, 284,  556, 288,  524, 294,  524, 292,  552, 294,  528, 290,  526, 292,  540, 304,  520, 296,  524, 294,  554, 290,  532, 286,  534, 284,  552, 292,  526, 290,  556, 290,  528, 1032,  552, 290,  522, 296,  524, 1058,  530, 290,  554, 290,  534, 1048,  530, 1054,  532, 290,  520, 296,  556, 288,  518, 300,  528, 288,  554, 292,  530, 1050,  524, 1060,  532, 288,  526, 1054,  532, 290,  538, 280,  554, 290,  524, 292,  550, 294,  524, 1058,  528, 292,  528, 290,  538, 306,  530, 288,  526, 290,  552, 322,  454, 334,  528, 290,  548, 296,  524, 294,  530, 290,  554, 290,  522, 296,  528, 316,  524, 294,  524, 294,  542, 276,  552, 292,  528, 290,  542, 276,  544, 300,  522, 296,  554, 290,  530, 288,  524, 292,  552, 292,  524, 294,  524, 294,  556, 288,  524, 294,  536, 282,  554, 288,  532, 286,  528, 290,  546, 298,  516, 302,  526, 292,  550, 294,  532, 1050,  522, 298,  540, 278,  556, 1028,  546, 298,  520, 298,  520, 1060,  526, 294,  562};  // UNKNOWN 994C0003

Najab, 2017-11-16:
After change the TIMEOUT 15U to TIMEOUT 90U for GREE is remain same

crankyoldgit, 2017-11-16:
I'm very surprised by that. The GREE protocol has a gap of 19ms between data chunks. Changing TIMEOUT to 90 (ms) should more than cover it. Before, a value of 15 (ms) would have been too short.

How did it go for the Kenwood A/C Remote?

What exact version of the library are you using? i.e. Look for the string `_IRREMOTEESP8266_VERSION_` in your local copy of _IRremoteESP866.h_, it should say "2.2.1". If it doesn't then you should update the library.

crankyoldgit, 2017-11-18:
FYI, I had another look at some of your captures.
The Singer one is obviously good. The very first Kenwood one looks okay as far as I can tell. The second Kenwood one (in the top of this issue) is just a bad capture. It doesn't look good at all.

The GREE ones are all parts of a much bigger signal, that setting the TIMEOUT value to a better size should fix. The Gree messages I have seen previously are made up of smaller messages, they all have to be sent in order to control the device. Hence, you need to capture it as a single message to get the timing/gaps right, if you want to send via sendRaw().

Najab, 2017-11-20:
I'm using "2.2.1" version of the library.
Only problem occurs for new 2017 model AC with all different brands(KENWOOD, GREE, DAWLANCE)

crankyoldgit, 2017-11-20:
I'm out of obvious simple ideas except for trying other transition frequencies. i.e. something other than 38kHz. Potentially these new devices use a different frequency.
Another common issue some protocols have is that they require a repeated signal. Typically that happens around 20-120ms after the first one. The gap varies across brand/protocol.
e.g.
```
// message
sendRaw(message_array, number_of_entries, 38);
delay(120);
// repeat of the message
sendRaw(message_array, number_of_entries, 38);
delay(120);
```

If you want a very rough guide of if a capture you've made is any good, you can try putting the rawdata through this script (requires unix/linux etc): https://github.com/markszabo/IRremoteESP8266/blob/auto_analyse_tool/tools/AutoAnalyseRawData.sh

e.g. put the `uint16_t rawData[303] = {9018, 4526, 570, 1676, ... };` line into a file and do:
```
./AutoAnalyseRawData.sh < filename_containing_rawData
```

If you try the first rawData you reported here, and the Singer ones, they come out fairly clean (i.e. they are probably easy/simple/correct etc)

Another angle to try is to look at LIRC on a linux machine if you can. They may have better decoding/capture abilities. or try some other tools/setups:
e.g. http://www.harctoolbox.org/IrScrutinizer.html
https://www.analysir.com/blog/
etc.

Najab, 2017-11-20:
is there any other way for read ir codes and then transmitt?

crankyoldgit, 2017-11-20:
Yes. If you have an Arduino board, you can try the original IRremote library, or similar to capture the code. The ESP8266 should be able to send it.
Or you could try the Global Cache IR database (https://irdb.globalcache.com/) or try to find Pronto codes for the devices. We support those sending formats too.

You can always to capturing it with an oscilloscope, that'd be the most accurate. Once analysed and converted, you should be able to send via the ESP8266.

Najab, 2017-11-20:
is there any other way for read ir codes and transmit except raw meth?

closed 2017-11-20 by Najab

adnandogar, 2020-08-05:
@Najab Did you solved?
In my case led blynks and emit rawdata but i m not sure whats the problem so far?

crankyoldgit, 2020-08-06:
@adnandogar Have you gone through all the steps of the [Troubleshooting guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide)?

P.S. Rather than tap on a years closed old issue, you should probably open a new one.

adnandogar, 2020-08-06:
@crankyoldgit after following detailed troubleshooting this issue resolved.
Thanks

crankyoldgit, 2020-08-07:
Can you supply us with the details of the device & remote (and which mitsubishi protocol) so we can add it to: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md