crankyoldgit/IRremoteESP8266 #351: Problems decoding Sharp TV code.

jorgecis, 2017-11-21:
### Lib Version
Lib Version 2.2

### Expected behavior
Decode the infrared code as Sharp, 

### Actual behavior
The code is detected as unknown, 

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

   Response from the Power Toggle 

Encoding  : UNKNOWN
Code      : 6D5D1DB3 (16 bits)
Timing[31]:
     + 300, -1850     + 300, - 800     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, - 800     + 300, -1850     + 300, -1850
     + 300, - 800     + 300, -1850     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, -1850     + 300, - 800     + 300
uint16_t  rawData[31] = {300,1850, 300,800, 300,800, 300,800, 300,800, 300,800, 300,1850, 300,1850, 300,800, 300,1850, 300,800, 300,800, 3
00,800, 300,1850, 300,800, 300};  // UNKNOWN 6D5D1DB3

Encoding  : UNKNOWN
Code      : 6322900E (16 bits)
Timing[31]:
     + 300, -1850     + 300, - 800     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, -1850     + 300, - 800     + 300, - 800

     + 300, -1850     + 300, - 800     + 300, -1850     + 300, -1850
     + 300, -1850     + 300, - 800     + 300, -1850     + 300
uint16_t  rawData[31] = {300,1850, 300,800, 300,800, 300,800, 300,800, 300,1850, 300,800, 300,800, 300,1850, 300,800, 300,1850, 300,1850,
300,1850, 300,800, 300,1850, 300};  // UNKNOWN 6322900E

Encoding  : UNKNOWN
Code      : 6D5D1DB3 (16 bits)
Timing[31]:
     + 300, -1850     + 300, - 800     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, - 800     + 300, -1850     + 300, -1850
     + 300, - 800     + 300, -1850     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, -1850     + 300, - 800     + 300
uint16_t  rawData[31] = {300,1850, 300,800, 300,800, 300,800, 300,800, 300,800, 300,1850, 300,1850, 300,800, 300,1850, 300,800, 300,800, 3
00,800, 300,1850, 300,800, 300};  // UNKNOWN 6D5D1DB3

Encoding  : UNKNOWN
Code      : 6322900E (16 bits)
Timing[31]:
     + 300, -1850     + 300, - 800     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, -1850     + 300, - 800     + 300, - 800

     + 300, -1850     + 300, - 800     + 300, -1850     + 300, -1850
     + 300, -1850     + 300, - 800     + 300, -1850     + 300
uint16_t  rawData[31] = {300,1850, 300,800, 300,800, 300,800, 300,800, 300,1850, 300,800, 300,800, 300,1850, 300,800, 300,1850, 300,1850,
300,1850, 300,800, 300,1850, 300};  // UNKNOWN 6322900E

Encoding  : UNKNOWN
Code      : 6D5D1DB3 (16 bits)
Timing[31]:
     + 300, -1850     + 300, - 800     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, - 800     + 300, -1850     + 300, -1850
     + 300, - 800     + 300, -1850     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, -1850     + 300, - 800     + 300
uint16_t  rawData[31] = {300,1850, 300,800, 300,800, 300,800, 300,800, 300,800, 300,1850, 300,1850, 300,800, 300,1850, 300,800, 300,800, 3
00,800, 300,1850, 300,800, 300};  // UNKNOWN 6D5D1DB3

Encoding  : UNKNOWN
Code      : 6322900E (16 bits)
Timing[31]:
     + 300, -1850     + 300, - 800     + 300, - 800     + 300, - 800
     + 300, - 800     + 300, -1850     + 300, - 800     + 300, - 800

     + 300, -1850     + 300, - 800     + 300, -1850     + 300, -1850
     + 300, -1850     + 300, - 800     + 300, -1850     + 300
uint16_t  rawData[31] = {300,1850, 300,800, 300,800, 300,800, 300,800, 300,1850, 300,800, 300,800, 300,1850, 300,800, 300,1850, 300,1850,
300,1850, 300,800, 300,1850, 300};  // UNKNOWN 6322900E

### Steps to reproduce the behavior
   To repeat the issue you will use a real sharp remote control and the example.  Looks like the code is there, becasue I got 3 times the same 2 raw codes.

crankyoldgit, 2017-11-21:
Hi @jorgecis 
Thanks for the detailed report.

Just a quick observation, while you say it's using v2.2 of the library, I think the compiled code you are running is older than that. Versions v2.2 and newer don't have timing values rounded to the nearest 50 anymore. You might want to check you have the latest version (currently v2.2.1) and recompile/upload the dumpV2 example program.

Can you please try doing that, and we can see where we are at after that?

In the meantime I'll try to simulate what I can based on the data you've provided.

jorgecis, 2017-11-21:
Hi,

   Sorry, I was a victim of platformio,  I have to force the update, anyway, using the last version looks a lot better.

I got 3 times the following codes

Encoding  : SHARP
Code      : 41A2 (15 bits)
Timing[31]:
   +   286, -  1846,    +   258, -   822,    +   234, -   818,    +   240, -   758,
   +   288, -   764,    +   292, -   760,    +   286, -  1818,    +   286, -  1846,
   +   268, -   812,    +   234, -  1844,    +   260, -   818,    +   238, -   816,
   +   232, -   794,    +   262, -  1842,    +   262, -   818,    +   240
uint16_t rawData[31] = {286, 1846,  258, 822,  234, 818,  240, 758,  288, 764,  292, 760,  286, 1818,  286, 1846,  268, 812,  234, 1844,
260, 818,  238, 816,  232, 794,  262, 1842,  262, 818,  240};  // SHARP 41A2
uint32_t address = 0x1;
uint32_t command = 0x16;
uint64_t data = 0x41A2;

Encoding  : UNKNOWN

Code      : 6322900E (16 bits)
Timing[31]:
   +   292, -  1812,    +   292, -   788,    +   258, -   766,    +   290, -   762,
   +   294, -   786,    +   240, -  1838,    +   286, -   792,    +   264, -   760,
   +   264, -  1840,    +   294, -   784,    +   262, -  1816,    +   288, -  1818,
   +   286, -  1846,    +   258, -   794,    +   262, -  1842,    +   262
uint16_t rawData[31] = {292, 1812,  292, 788,  258, 766,  290, 762,  294, 786,  240, 1838,  286, 792,  264, 760,  264, 1840,  294, 784,  2
62, 1816,  288, 1818,  286, 1846,  258, 794,  262, 1842,  262};  // UNKNOWN 6322900E

   The first one is detected as sharp but not the second one, also not sure why I got 3 pairs of codes every time.

crankyoldgit, 2017-11-21:
Phew. That looks a lot better.

The multiple codes are explained (in part) by the comments here: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Sharp.cpp#L63

As you have a Sharp remote, that matches this protocol, _can you please do me a favour and capture the entire output of the dumpV2 program **just** for a single button press?_
i.e. I'd like to make sure once and for all if it sends the message in three chunks. (normal, then inverted, then normal; or just normal, then inverted)

Oh, also what model remote control and TV it is would be a bonus too.

Don't worry so much that it isn't recognising the second 'inverted' message, that's expected/normal operation at this point. The code it gives you for the first "chunk" is what matters when it comes to using the appropriate send() routine.

jorgecis, 2017-11-23:
Yes sure, here is the full output for a Sharp POWER TOGGLE, the model tv is Sharp LC-52D62U
Also, I have another question the function is sendSharp or sendSharpraw? yes I read the comments but still is a little confuse 

Encoding  : SHARP
Code      : 41A2 (15 bits)
Timing[31]:
   +   286, -  1820,    +   294, -   758,    +   288, -   764,    +   294, -   758,
   +   286, -   766,    +   290, -   760,    +   284, -  1820,    +   294, -  1810,
   +   294, -   758,    +   286, -  1844,    +   260, -   794,    +   262, -   764,
   +   294, -   758,    +   288, -  1844,    +   260, -   766,    +   290
uint16_t rawData[31] = {286, 1820,  294, 758,  288, 764,  294, 758,  286, 766,  290, 760,  284, 1820,  294, 1810,  294, 758,  286, 1844,  260, 794,  262, 764,  294, 758,  288, 1844,  260, 766,  290};  // SHARP 41A2
uint32_t address = 0x1;
uint32_t command = 0x16;
uint64_t data = 0x41A2;

Encoding  : UNKNOWN
Code      : 6322900E (16 bits)
Timing[31]:
   +   292, -  1840,    +   294, -   784,    +   240, -   786,    +   260, -   792,
   +   266, -   786,    +   260, -  1844,    +   258, -   822,    +   236, -   790,
   +   266, -  1838,    +   266, -   814,    +   232, -  1846,    +   290, -  1816,
   +   288, -  1816,    +   286, -   766,    +   290, -  1814,    +   290
uint16_t rawData[31] = {292, 1840,  294, 784,  240, 786,  260, 792,  266, 786,  260, 1844,  258, 822,  236, 790,  266, 1838,  266, 814,  232, 1846,  290, 1816,  288, 1816,  286, 766,  290, 1814,  290};  // UNKNOWN 6322900E

Encoding  : SHARP
Code      : 41A2 (15 bits)
Timing[31]:
   +   292, -  1812,    +   292, -   762,    +   284, -   768,    +   288, -   764,
   +   294, -   758,    +   286, -   766,    +   292, -  1812,    +   290, -  1814,
   +   290, -   762,    +   294, -  1812,    +   292, -   760,    +   286, -   766,
   +   290, -   762,    +   294, -  1810,    +   294, -   758,    +   288
uint16_t rawData[31] = {292, 1812,  292, 762,  284, 768,  288, 764,  294, 758,  286, 766,  292, 1812,  290, 1814,  290, 762,  294, 1812,  292, 760,  286, 766,  290, 762,  294, 1810,  294, 758,  288};  // SHARP 41A2
uint32_t address = 0x1;
uint32_t command = 0x16;
uint64_t data = 0x41A2;

Encoding  : UNKNOWN
Code      : 43683F81 (16 bits)
Timing[31]:
   +   288, -  1842,    +   260, -   820,    +   238, -   788,    +   258, -   766,
   +   290, -   762,    +   294, -  1836,    +   266, -   814,    +   232, -   764,
   +   292, -  1840,    +   264, -   816,    +   240, -  1838,    +   266, -  1840,
   +   286, -  1820,    +   294, -   758,    +   288, -  1818,    +   286
uint16_t rawData[31] = {288, 1842,  260, 820,  238, 788,  258, 766,  290, 762,  294, 1836,  266, 814,  232, 764,  292, 1840,  264, 816,  240, 1838,  266, 1840,  286, 1820,  294, 758,  288, 1818,  286};  // UNKNOWN 43683F81

Encoding  : SHARP
Code      : 41A2 (15 bits)
Timing[31]:
   +   290, -  1814,    +   288, -   764,    +   294, -   758,    +   288, -   766,
   +   290, -   760,    +   264, -   788,    +   258, -  1846,    +   288, -  1816,
   +   288, -   764,    +   292, -  1814,    +   290, -   762,    +   294, -   758,
   +   288, -   764,    +   294, -  1812,    +   292, -   760,    +   286
uint16_t rawData[31] = {290, 1814,  288, 764,  294, 758,  288, 766,  290, 760,  264, 788,  258, 1846,  288, 1816,  288, 764,  292, 1814,  290, 762,  294, 758,  288, 764,  294, 1812,  292, 760,  286};  // SHARP 41A2
uint32_t address = 0x1;
uint32_t command = 0x16;
uint64_t data = 0x41A2;

Encoding  : UNKNOWN
Code      : 6322900E (16 bits)
Timing[31]:
   +   288, -  1844,    +   260, -   792,    +   264, -   760,    +   284, -   768,
   +   290, -   762,    +   294, -  1812,    +   292, -   760,    +   286, -   766,
   +   290, -  1814,    +   290, -   762,    +   294, -  1838,    +   266, -  1840,
   +   264, -  1842,    +   294, -   758,    +   288, -  1818,    +   286
uint16_t rawData[31] = {288, 1844,  260, 792,  264, 760,  284, 768,  290, 762,  294, 1812,  292, 760,  286, 766,  290, 1814,  290, 762,  294, 1838,  266, 1840,  264, 1842,  294, 758,  288, 1818,  286};  // UNKNOWN 6322900E

crankyoldgit, 2017-11-23:
Interesting. Thanks for that. It seems to send a pair of (normal + inverted) messages 3 times. I might need to change/set the repeats to 2. (initial pair + 2 more pairs)
You may need to call it via `sendSharpRaw(data, SHARP_BITS, 2);` Please let me know if it works without setting the 'repeat' argument, or if it does require it to be set to '2' in order to work.

Re: _sendSharp() vs. sendSharpRaw()_

Yep, I get that it's confusing. I'll try to explain it a different way, if you then understand it, please feel free to come up with wording that will make it easier to understand of others. Send it as a PR, or put it here and I'll update the comments etc.

Long story short, once upon a time, there was only `sendSharp()`, who ever wrote it got the bit order wrong. Later, the decode routine was written for the protocol and the error found. Technically it doesn't matter, the output value of the decode  is the value to plug into `sendSharpRaw()`. The order matters if you are using the Address & Command values or if you were looking them up from other non-IRremote-based reference sites etc.

Soooo ... `sendSharpRaw()` assumes/uses the **correct** bit order for the data, `sendSharp()` uses the wrong order, but which users of this (and the _original_ IRremote Arduino) library have been using for years. `sendSharp()` just changes the order of the bits, and calls `sendSharpRaw()` now-a-days.

**TL;DR:** If you are using our library to decode, the result should go into `sendSharpRaw()`, if you are using values from elsewhere, use `sendSharpRaw()`. Backward compatibility for a _bug/error_ is a cow! ;-)
(ref: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Sharp.cpp#L171)

And if you don't get what I mean by bit order. try [this](https://en.wikipedia.org/wiki/Most_significant_bit#Most_Significant_Bit_First_vs_Least_Significant_Bit_First)

crankyoldgit, 2017-11-23:
Oh @jorgecis Could you please edit your previous message and add code block quotes around the output you supplied? i.e. https://help.github.com/articles/creating-and-highlighting-code-blocks/

It will stop the output from being mangled.  Thanks in advance.

jorgecis, 2017-11-23:
Sure I already edit my comment, also in your comment you wrote:

TL;DR: If you are using our library to decode, the result should go into ```sendSharpRaw()```, if you are using values from elsewhere, use ```sendSharpRaw()```. Backward compatibility for a bug/error is a cow! ;-)

use the same function in both cases❓ hehehe

crankyoldgit, 2017-11-23:
Yes, I meant it. In short, `sendSharp()` is deprecated. Unless you were using it previously, don't use it. 
Always use `sendSharpRaw()` unless you were using `sendSharp()` before.

crankyoldgit, 2017-11-25:
Just following up. How did you go?

jorgecis, 2017-11-28:
Using the sendSharpRaw everything works fine. the code is decode correctly and the send function works perfect.
Please close this issue.

crankyoldgit, 2017-11-29:
I'm glad it's working for you. Feel free to spread the word. ;-)

closed 2017-11-29 by crankyoldgit

ipmungam, 2021-07-01:
In case useful for someone, I can confirm that in version 2.7.18 the command

`sendSharpRaw(data, SHARP_BITS, 2);`

 works for Sharp Aquos LE835X, with data in the format e.g.
 
` const uint64_t data = 0x41A2;     //Toggle power.`