crankyoldgit/IRremoteESP8266 #352: problem decoding KENWOOD AC code

Najab, 2017-11-21:
When i receive IR signals from remote it gives me different array size value after every 1 or 2 click and TIMEOUT is Both 15U or 90U. 

Encoding  : UNKNOWN
Code      : 42DF5AF7 (168 bits)
Timing[335]: 
   +   590, -   182,    +   200, -  8328,    +   140, - 11252,    +   300, -  9114, 
   +   270, -  5252,    +   218, -  2162,    +   250, -   152,    +   660, -  6550, 
   +   288, - 23072,    +   240, -   154,    +   644, -   602,    +   232, -   342, 
   +   884, - 56316,    +   326, - 61588,    +   492, - 12774,    +   246, -  6900, 
   +   252, - 10124,    +  1182, -  1880,    +   904, - 46032,    +   244, - 71180, 
   +   646, - 29760,    +   246, -   254,    +  1376, -  7202,    +   236, -  1698, 
   +   320, - 11328,    +   274, - 24748,    +   316, - 21458,    +  9234, -  4282, 
   +   790, -  1460,    +   762, -  1506,    +   710, -   428,    +   684, -   450, 
   +   658, -   488,    +   634, -   522,    +   608, -   542,    +   588, -  1670, 
   +   580, -   546,    +   574, -  1678,    +   576, -  1678,    +   586, -   550, 
   +   576, -   564,    +   570, -   572,    +   576, -   568,    +   578, -   558, 
   +   588, -  1660,    +   582, -   548,    +   584, -   550,    +   574, -   560, 
   +   586, -   554,    +   578, -   566,    +   578, -   568,    +   578, -   558, 
   +   578, -   546,    +   578, -  1674,    +   582, -   550,    +   580, -   556, 
   +   576, -   562,    +   576, -   568,    +   582, -  1686,    +   580, -   556, 
   +   584, -   542,    +   578, -   550,    +   576, -   556,    +   580, -   556, 
   +   576, -   562,    +   578, -   564,    +   578, -   568,    +   578, -   558, 
   +   578, -   548,    +   580, -   548,    +   582, -   550,    +   576, -   560, 
   +   584, -   556,    +   574, -   570,    +   582, -   562,    +   586, -   532, 
   +   580, -  7968,    +   584, -   542,    +   582, -  1668,    +   580, -   550, 
   +   570, -   564,    +   572, -  1688,    +   576, -   566,    +   580, -   568, 
   +   576, -  1680,    +   576, -  1670,    +   576, -   552,    +   574, -   558, 
   +   572, -   564,    +   586, -  1676,    +   576, -   566,    +   580, -   566, 
   +   580, -   556,    +   584, -   542,    +   582, -   548,    +   578, -   554, 
   +   580, -   556,    +   572, -   566,    +   578, -   564,    +   578, -   566, 
   +   580, -   556,    +   580, -   544,    +   576, -   554,    +   572, -   560, 
   +   580, -   556,    +   572, -   566,    +   576, -   566,    +   578, -   566, 
   +   578, -   558,    +   576, -   550,    +   576, -   550,    +   578, -   554, 
   +   580, -   556,    +   582, -   556,    +   582, -   562,    +   574, -   572, 
   +   576, -   560,    +   580, -   544,    +   578, -   550,    +   576, -   556, 
   +   572, -   564,    +   578, -   560,    +   582, -   562,    +   574, -   570, 
   +   580, -   556,    +   572, -   554,    +   576, -   554,    +   582, -   548, 
   +   576, -   562,    +   576, -   564,    +   578, -   564,    +   580, -   566, 
   +   586, -   550,    +   576, -   548,    +   578, -   550,    +   584, -   548, 
   +   580, -   556,    +   570, -   568,    +   576, -   566,    +   582, -  1686, 
   +   576, -  1664,    +   580, -  7966,    +   584, -   542,    +   578, -   550, 
   +   582, -   550,    +   576, -   558,    +   572, -   566,    +   576, -   566, 
   +   578, -   568,    +   578, -   558,    +   580, -   544,    +   574, -  1676, 
   +   568, -   564,    +   574, -   562,    +   578, -   560,    +   580, -   562, 
   +   580, -   566,    +   588, -   548,    +   576, -   550,    +   580, -   548, 
   +   586, -   546,    +   584, -   554,    +   576, -   560,    +   580, -   564, 
   +   580, -   564,    +   578, -   558,    +   576, -   550,    +   372
uint16_t rawData[337] = {590, 182,  200, 8328,  140, 11252,  300, 9114,  270, 5252,  218, 2162,  250, 152,  660, 6550,  288, 23072,  240, 154,  644, 602,  232, 342,  884, 56316,  326, 61588,  492, 12774,  246, 6900,  252, 10124,  1182, 1880,  904, 46032,  244, 65535, 05645,  646, 29760,  246, 254,  1376, 7202,  236, 1698,  320, 11328,  274, 24748,  316, 21458,  9234, 4282,  790, 1460,  762, 1506,  710, 428,  684, 450,  658, 488,  634, 522,  608, 542,  588, 1670,  580, 546,  574, 1678,  576, 1678,  586, 550,  576, 564,  570, 572,  576, 568,  578, 558,  588, 1660,  582, 548,  584, 550,  574, 560,  586, 554,  578, 566,  578, 568,  578, 558,  578, 546,  578, 1674,  582, 550,  580, 556,  576, 562,  576, 568,  582, 1686,  580, 556,  584, 542,  578, 550,  576, 556,  580, 556,  576, 562,  578, 564,  578, 568,  578, 558,  578, 548,  580, 548,  582, 550,  576, 560,  584, 556,  574, 570,  582, 562,  586, 532,  580, 7968,  584, 542,  582, 1668,  580, 550,  570, 564,  572, 1688,  576, 566,  580, 568,  576, 1680,  576, 1670,  576, 552,  574, 558,  572, 564,  586, 1676,  576, 566,  580, 566,  580, 556,  584, 542,  582, 548,  578, 554,  580, 556,  572, 566,  578, 564,  578, 566,  580, 556,  580, 544,  576, 554,  572, 560,  580, 556,  572, 566,  576, 566,  578, 566,  578, 558,  576, 550,  576, 550,  578, 554,  580, 556,  582, 556,  582, 562,  574, 572,  576, 560,  580, 544,  578, 550,  576, 556,  572, 564,  578, 560,  582, 562,  574, 570,  580, 556,  572, 554,  576, 554,  582, 548,  576, 562,  576, 564,  578, 564,  580, 566,  586, 550,  576, 548,  578, 550,  584, 548,  580, 556,  570, 568,  576, 566,  582, 1686,  576, 1664,  580, 7966,  584, 542,  578, 550,  582, 550,  576, 558,  572, 566,  576, 566,  578, 568,  578, 558,  580, 544,  574, 1676,  568, 564,  574, 562,  578, 560,  580, 562,  580, 566,  588, 548,  576, 550,  580, 548,  586, 546,  584, 554,  576, 560,  580, 564,  580, 564,  578, 558,  576, 550,  372};  // UNKNOWN 42DF5AF7

Encoding  : UNKNOWN
Code      : 45D3C31B (154 bits)
Timing[307]: 
   +  9010, -  4528,    +   576, -  1672,    +   566, -  1680,    +   578, -   556, 
   +   570, -   564,    +   570, -   568,    +   572, -   570,    +   584, -   562, 
   +   578, -  1678,    +   582, -   544,    +   572, -  1674,    +   578, -  1676, 
   +   576, -   560,    +   570, -   568,    +   576, -   566,    +   576, -   570, 
   +   570, -   564,    +   574, -  1676,    +   578, -   548,    +   574, -   558, 
   +   574, -   562,    +   576, -   560,    +   576, -   566,    +   578, -   568, 
   +   586, -   550,    +   578, -   546,    +   584, -  1666,    +   576, -   558, 
   +   576, -   560,    +   578, -   560,    +   582, -   560,    +   584, -  1680, 
   +   588, -   552,    +   570, -   556,    +   576, -   552,    +   576, -   556, 
   +   574, -   562,    +   576, -   560,    +   580, -   560,    +   580, -   566, 
   +   576, -   560,    +   584, -   540,    +   572, -   556,    +   580, -   550, 
   +   576, -   560,    +   578, -   560,    +   576, -   566,    +   576, -   572, 
   +   578, -   542,    +   578, -  7968,    +   576, -   550,    +   576, -  1674, 
   +   576, -   558,    +   584, -   552,    +   576, -  1678,    +   590, -   562, 
   +   574, -   570,    +   576, -  1680,    +   580, -  1664,    +   576, -   556, 
   +   578, -   554,    +   576, -   560,    +   582, -  1676,    +   584, -   560, 
   +   578, -   568,    +   576, -   560,    +   574, -   552,    +   580, -   548, 
   +   584, -   546,    +   578, -   560,    +   574, -   564,    +   574, -   568, 
[... 388 lines cut ...]
   +   568, -   574,    +   574, -  1684,    +   578, -  1666,    +   582, -   548, 
   +   576, -   554,    +   578, -   558,    +   578, -  1682,    +   578, -   568, 
   +   570, -   576,    +   578, -   556,    +   580, -   546,    +   576, -   552, 
   +   586, -   546,    +   578, -   558,    +   584, -   554,    +   580, -   562, 
   +   584, -   562,    +   580, -   556,    +   578, -   548,    +   574, -   554, 
   +   576, -   556,    +   578, -   556,    +   570, -   568,    +   580, -   562, 
   +   580, -   566,    +   576, -   558,    +   580, -   544,    +   574, -   554, 
   +   576, -   554,    +   578, -   556,    +   570, -   566,    +   576, -   566, 
   +   574, -   572,    +   576, -   558,    +   578, -   546,    +   578, -   550, 
   +   564, -   566,    +   574, -   562,    +   582, -   556,    +   580, -   562, 
   +   580, -   564,    +   584, -   552,    +   582, -   542,    +   580, -   548, 
   +   578, -   554,    +   574, -   560,    +   578, -   562,    +   576, -   564, 
   +   580, -   564,    +   574, -   562,    +   576, -   550,    +   578, -   552, 
[... cut ...] , 560,  578, 556,  578, 1682,  582, 560,  600, 546,  566, 550,  572, 7968,  582, 546,  576, 552,  582, 548,  574, 560,  578, 560,  576, 568,  578, 568,  576, 560,  570, 554,  580, 1668,  578, 556,  578, 558,  574, 564,  578, 564,  584, 562,  580, 556,  582, 542,  580, 548,  580, 552,  574, 562,  580, 558,  580, 564,  582, 562,  582, 556,  576, 548,  576, 552,  580, 552,  578, 556,  574, 564,  580, 560,  572, 574,  578, 556,  574, 550,  578, 550,  578, 554,  580, 1674,  578};  // UNKNOWN B5EF83D2

crankyoldgit, 2017-11-21:
The library doesn't support decoding of the Kenwood A/C commands.
As discussed (and in the FAQ/troubleshooting notes) working with A/C remotes is tricky & hard.
If you want the library to be able to decode it, you've got a fair bit of coding/work ahead of you.

Doing some analysis on your output:
The first capture has garbage at the beginning. Discard it.

Any capture with an EVEN number of elements in the `rawData[]` size is probably wrong. e.g. 304
You should almost never see that happening. It could be a bug in the interrupt routines, or something else. Not sure. Ignore those.

The codes with a TINY last value. e.g. `..., 44};` are also probably wrong/bad. That's to small a value normally.

If I was to guess, I'd go with the ones with a size of 303. They look the best/most correct to me.

A tip I remember from another user, make sure you are not holding the remote too close to the IR sensor. That often can confuse the sensor and make it play up.

Doing a deeper analysis on one of the [303] codes:

> uint16_t rawData[303] = {9014, 4522, 568, 1678, 578, 1672, 582, 552, 574, 562, 570, 568, 576, 566, 574, 572, 574, 1682, 586, 542, 582, 1668, 570, 1684, 578, 558, 572, 566, 578, 564, 578, 568, 570, [...]

gives:
```
Potential Mark Candidates (using a range of 200):
9014
588

Potential Space Candidates (using a range of 200):
7966
4522
1692
582

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK = 9014
BIT_MARK = 588
SPACE_GAP1 = 7966
HDR_SPACE = 4522
ONE_SPACE = 1692
ZERO_SPACE = 582

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+110000010110000010000000010010100000000000000000 GAP(7964)
  Bits: 48
  Hex:  0xC160804A0000 (MSB first)
        0x52010683 (LSB first)
  Dec:  212620213354496 (MSB first)
        1375798915 (LSB first)
  Bin:  110000010110000010000000010010100000000000000000 (MSB first)
        000000000000000001010010000000010000011010000011 (LSB first)
BIT_MARK(UNEXPECTED)0100100110001000000000000000000000000000000000000000000000001011 GAP(7966)
  Bits: 64
  Hex:  0x498800000000000B (MSB first)
        0xD000000000001192 (LSB first)
  Dec:  5298484961601388555 (MSB first)
        14987979559889015186 (LSB first)
  Bin:  0100100110001000000000000000000000000000000000000000000000001011 (MSB first)
        1101000000000000000000000000000000000000000000000001000110010010 (LSB first)
BIT_MARK(UNEXPECTED)000000000100000000000000000000000001
  Bits: 36
  Hex:  0x4000001 (MSB first)
        0x800000200 (LSB first)
  Dec:  67108865 (MSB first)
        34359738880 (LSB first)
  Bin:  000000000100000000000000000000000001 (MSB first)
        100000000000000000000000001000000000 (LSB first)
```

The last chunk is an odd size (36 bits isn't an even multiple of 8), so I'm not sure what is up with that.

As I've suggested elsewhere, you may need to look at another platform to try to capture these codes. Some combination of your environment/circuit/settings/ir demodulator/the code/board/chip/etc is having difficulties. It could be you are using, say, a 38kHz IR receiver module, and the signal is at 32, 33, 36, or 40kHz etc.

crankyoldgit, 2017-11-25:
Have you tried a different (non-38kHz receiver) yet? e.g. 36kHz etc.
https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#basic-1

crankyoldgit, 2018-03-16:
Marking this closed for now due to lack of response.

closed 2018-03-16 by crankyoldgit