crankyoldgit/IRremoteESP8266 #353: I want KENWOOD AC remote code

Najab, 2017-11-22:
If any one have KENWOOD AC remote code for ON/OFF and Change TEMPERATURES in the form of RAW Data please provide for me.
Thanks in advance.

crankyoldgit, 2017-11-22:
Suggestion: Have you tried this approach?
https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

Najab, 2017-11-22:
It gives me this type of Data

Raw: (18327) 8908, -4548, 536, -1700, 528, -1708, 532, -592, 532, -596, 524, -608, 532, -604, 528, -612, 520, -1724, 528, -588, 524, -1708, 532, -1708, 540, -588, 532, -600, 528, -608, 528, -608, 536, -596, 532, -1700, 532, -592, 536, -1704, 528, -596, 532, -604, 532, -604, 536, -600, 528, -604, 528, -588, 532, -1704, 536, -592, 540, -588, 532, -1716, 532, -1716, 544, -600, 528, -600, 536, -580, 536, -588, 528, -600, 524, -604, 524, -608, 528, -604, 528, -616, 528, -600, 532, -584, 536, -588, 532, -592, 528, -600, 532, -604, 532, -604, 532, -604, 528, -584, 532, -7952, 536, -1700, 528, -1708, 532, -1708, 532, -596, 532, -1712, 540, -596, 532, -608, 536, -1708, 528, -1704, 532, -588, 528, -1712, 532, -600, 524, -1724, 528, -608, 532, -604, 536, -592, 536, -584, 528, -596, 528, -596, 544, -584, 532, -600, 536, -600, 528, -612, 532, -596, 536, -584, 528, -592, 540, -588, 532, -596, 536, -600, 524, -612, 528, -608, 540, -592, 524, -592, 528, -596, 536, -588, 536, -592, 528, -600, 532, -604, 536, -604, 536, -592, 536, -584, 524, -596, 528, -600, 532, -596, 528, -604, 528, -608, 528, -612, 528, -600, 532, -588, 532, -588, 528, -600, 532, -596, 524, -608, 540, -596, 528, -612, 528, -596, 536, -1700, 528, -596, 532, -1708, 520, -608, 528, -1720, 528, -1720, 528, -612, 532, -1692, 532, -7960, 532, -588, 528, -596, 532, -592, 528, -600, 524, -612, 528, -604, 536, -604, 532, -596, 532, -1700, 528, -596, 524, -600, 532, -596, 528, -604, 536, -600, 532, -608, 528, -604, 528, -588, 532, -592, 536, -588, 524, -604, 532, -604, 532, -600, 528, -612, 528, -600, 528, -592, 528, -596, 528, -596, 528, -600, 532, -600, 528, -608, 532, -608, 528, -604, 528, -588, 528, -596, 532, -592, 524, -1720, 532, -604, 536, -1712, 532, -608, 536, -596, 532, -584, 532, -592, 532, -592, 528, -604, 528, -604, 532, -604, 532, -1720, 528, -604, 532, -1696, 528, -596, 532, -592, 528, -1716, 532, -600, 536, -1716, 532, -1724, 524, -588, 548, -13808, 8916, -4544, 528, -1704, 528, -1712, 532, -592, 532, -600, 524, -608, 532, -604, 532, -608, 532, -1712, 528, -592, 532, -1704, 532, -1712, 532, -596, 536, -596, 528, -608, 528, -612, 532, -600, 528, -1704, 524, -600, 528, -1712, 532, -596, 524, -608, 528, -608, 532, -612, 528, -600, 532, -588, 524, -1712, 532, -592, 532, -600, 528, -1720, 528, -1720, 532, -608, 528, -604, 532, -588, 532, -588, 528, -596, 532, -596, 532, -600, 540, -600, 528, -608, 532, -596, 532, -588, 524, -596, 532, -596, 524, -604, 536, -596, 532, -604, 528, -612, 532, -580, 528, -7964, 532, -1700, 532, -1704, 528, -1712, 528, -600, 528, -1720, 532, -604, 536, -604, 528, -1712, 536, -1696, 536, -588, 532, -1708, 536, -596, 524, -1720, 536, -600, 532, -608, 532, -596, 532, -588, 524, -596, 536, -588, 532, -600, 532, -600, 536, -604, 528, -612, 528, -600, 528, -592, 528, -592, 532, -592, 532, -600, 532, -600, 532, -604, 532, -608, 532, -596, 524, -592, 532, -592, 532, -592, 532, -600, 528, -604, 524, -608, 528, -612, 532, -596, 532, -588, 532, -592, 536, -588, 536, -592, 532, -600, 532, -604, 532, -608, 536, -592, 536, -584, 532, -588, 528, -596, 528, -600, 532, -600, 528, -608, 532, -608, 528, -604, 536, -1696, 536, -588, 532, -1708, 532, -596, 528, -1720, 528, -1724, 532, -608, 524, -1700, 536, -7960, 532, -588, 536, -588, 532, -592, 532, -596, 528, -604, 528, -608, 536, -604, 532, -596, 532, -1700, 536, -588, 528, -600, 524, -604, 528, -604, 532, -604, 532, -604, 532, -596, 532, -588, 536, -584, 536, -592, 528, -600, 532, -604, 528, -608, 528, -612, 528, -600, 524, -592, 536, -588, 528, -596, 528, -600, 532, -604, 528, -608, 532, -604, 536, -592, 536, -584, 528, -592, 540, -588, 536, -1708, 528, -1720, 532, -1716, 536, -604, 528, -604, 532, -584, 532, -588, 532, -592, 528, -600, 524, -612, 528, -604, 532, -1724, 532, -596, 532, -1700, 532, -592, 524, -600, 532, -1712, 532, -1716, 532, -1716, 528, -1728, 536, -576, 556, -43628, 8908, -4548, 524, -1708, 532, -1704, 524, -604, 520, -608, 528, -604, 528, -608, 524, -616, 524, -1720, 524, -592, 528, -1708, 528, -1712, 528, -600, 524, -612, 524, -612, 520, -616, 520, -608, 520, -1712, 516, -608, 524, -600, 536, -592, 528, -604, 528, -608, 532, -608, 528, -600, 532, -588, 528, -1704, 536, -592, 532, -596, 532, -1712, 532, -1720, 532, -604, 532, -596, 532, -584, 528, -596, 528, -596, 524, -604, 524, -608, 532, -604, 528, -608, 524, -604, 528, -592, 524, -596, 528, -596, 532, -600, 536, -596, 528, -608, 532, -604, 528, -584, 536, -7960, 528, -1708, 532, -1704, 532, -1712, 524, -604, 528, -1724, 532, -604, 47472, -65272, 1000, -64536, 0, -0, 197, -65535, 65532, -1, 1, -4, 65339, -14847, 65282, -65526, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 0, -0, 8192, -3120, 63999, -2544, 787, -6, 59123, -4609, 520, -63487, 2796, -24, 59636, -5633, 59903, -6409, 4613, -64765, 60667, -5633, 263, -65026, 60150, -4106, 1536, -62977, 233, -2325, 64766, -60671, 64263, -61173, 13322, -16629, 1599, -24787, 29111, -28248, 64549, -41519, 28076, -41649, 12784, -38762, 58456, -8995, 25225, -60301, 2533, -11506, 63840, -50077, 10189, -13695, 39095, -38887, 5954, -796, 28017, -24933, 14022, -3017, 57085, -6969, 22631, -1080, 16790, -28585, 33893, -39773, 19097, -24225, 53562, -15992, 28622, -21635, 31314, -44731, 53306, -58567, 47198, -59957, 61533, -25056, 39253, -19775, 61485, -17978, 51238, -47315, 39268, -48519, 58548, -50354, 48551, -46521, 6844, -48672, 29256, -21193, 8979, -47947, 38347, -6182, 29334, -39330, 29808, [... cut ...] , -0, 0, -0, 0, -0, 0, -0, 0, -0, 11264, -288, 65512, -65301, 64764, -60923, 5633, -64776, 65019, -61180, 5130, -513, 62973, -236, 790, -4609, 64521, -64515, 2535, -65300, 64510, -61178, 5633, -65287, 65533, -60155, 5382, -6, 0, -240, 1810, -63486, 30448, -5856, 59107, -19993, 4612, -9745, 55537, -65534, 2560, -62977, 1279, -64512, 36872, -65436, 6336, -40768, 39753, -40, 32, -0, 65504, -7, 65529, -47288, 0, -0, 0, -65535, 1, -45489, 65535, -65280, 0, -85, 683, -49155, 43005, -184, 0, -0, 12472,

crankyoldgit, 2017-11-22:
wow. All of that for a single button press?!
You'll notice, similar to the dumpV2 output, in that this has positive and negative values.
Positive values indicate a "mark" signal (IR led is transmitting) and negative indicates a "space" (IR led is NOT transmitting). The values will be in micro-seconds.
To convert it to the rawData syntax, simply remove the '-' signs, and count up how many entries you want to send and put that into the array initialisation etc.

It seems to report that there are 18327 entries in total. That's just too big to be a single real message obviously.

The message header (start) mark looks like it's around the 9000 mark. (We also saw that in your other captures)
I'd truncate the rest of the message at/or around the "47472, -65272, 1000, -64536, 0, -0, 197, -65535, 65532" area, as that looks like it's reading just noise, and not real signals. i.e. I've never seen a "mark" signal that is 47000+ micro-seconds long.
So try doing that and see how it goes.

Basically, large spaces (gaps) are not uncommon. Large marks (i.e. the positive ones in this capture) rarely exceed about 10,000. If you are seeing that big a mark, it's probably garbage and probably a good place to snip the data at.

crankyoldgit, 2017-11-22:
Following the above, try these possible values:

```
uint16_t rawbuf[687] = {8908, 4548, 536, 1700, 528, 1708, 532, 592, 532, 596, 524, 608, 532, 604, 528, 612, 520, 1724, 528, 588, 524, 1708, 532, 1708, 540, 588, 532, 600, 528, 608, 528, 608, 536, 596, 532, 1700, 532, 592, 536, 1704, 528, 596, 532, 604, 532, 604, 536, 600, 528, 604, 528, 588, 532, 1704, 536, 592, 540, 588, 532, 1716, 532, 1716, 544, 600, 528, 600, 536, 580, 536, 588, 528, 600, 524, 604, 524, 608, 528, 604, 528, 616, 528, 600, 532, 584, 536, 588, 532, 592, 528, 600, 532, 604, 532, 604, 532, 604, 528, 584, 532, 7952, 536, 1700, 528, 1708, 532, 1708, 532, 596, 532, 1712, 540, 596, 532, 608, 536, 1708, 528, 1704, 532, 588, 528, 1712, 532, 600, 524, 1724, 528, 608, 532, 604, 536, 592, 536, 584, 528, 596, 528, 596, 544, 584, 532, 600, 536, 600, 528, 612, 532, 596, 536, 584, 528, 592, 540, 588, 532, 596, 536, 600, 524, 612, 528, 608, 540, 592, 524, 592, 528, 596, 536, 588, 536, 592, 528, 600, 532, 604, 536, 604, 536, 592, 536, 584, 524, 596, 528, 600, 532, 596, 528, 604, 528, 608, 528, 612, 528, 600, 532, 588, 532, 588, 528, 600, 532, 596, 524, 608, 540, 596, 528, 612, 528, 596, 536, 1700, 528, 596, 532, 1708, 520, 608, 528, 1720, 528, 1720, 528, 612, 532, 1692, 532, 7960, 532, 588, 528, 596, 532, 592, 528, 600, 524, 612, 528, 604, 536, 604, 532, 596, 532, 1700, 528, 596, 524, 600, 532, 596, 528, 604, 536, 600, 532, 608, 528, 604, 528, 588, 532, 592, 536, 588, 524, 604, 532, 604, 532, 600, 528, 612, 528, 600, 528, 592, 528, 596, 528, 596, 528, 600, 532, 600, 528, 608, 532, 608, 528, 604, 528, 588, 528, 596, 532, 592, 524, 1720, 532, 604, 536, 1712, 532, 608, 536, 596, 532, 584, 532, 592, 532, 592, 528, 604, 528, 604, 532, 604, 532, 1720, 528, 604, 532, 1696, 528, 596, 532, 592, 528, 1716, 532, 600, 536, 1716, 532, 1724, 524, 588, 548, 13808, 8916, 4544, 528, 1704, 528, 1712, 532, 592, 532, 600, 524, 608, 532, 604, 532, 608, 532, 1712, 528, 592, 532, 1704, 532, 1712, 532, 596, 536, 596, 528, 608, 528, 612, 532, 600, 528, 1704, 524, 600, 528, 1712, 532, 596, 524, 608, 528, 608, 532, 612, 528, 600, 532, 588, 524, 1712, 532, 592, 532, 600, 528, 1720, 528, 1720, 532, 608, 528, 604, 532, 588, 532, 588, 528, 596, 532, 596, 532, 600, 540, 600, 528, 608, 532, 596, 532, 588, 524, 596, 532, 596, 524, 604, 536, 596, 532, 604, 528, 612, 532, 580, 528, 7964, 532, 1700, 532, 1704, 528, 1712, 528, 600, 528, 1720, 532, 604, 536, 604, 528, 1712, 536, 1696, 536, 588, 532, 1708, 536, 596, 524, 1720, 536, 600, 532, 608, 532, 596, 532, 588, 524, 596, 536, 588, 532, 600, 532, 600, 536, 604, 528, 612, 528, 600, 528, 592, 528, 592, 532, 592, 532, 600, 532, 600, 532, 604, 532, 608, 532, 596, 524, 592, 532, 592, 532, 592, 532, 600, 528, 604, 524, 608, 528, 612, 532, 596, 532, 588, 532, 592, 536, 588, 536, 592, 532, 600, 532, 604, 532, 608, 536, 592, 536, 584, 532, 588, 528, 596, 528, 600, 532, 600, 528, 608, 532, 608, 528, 604, 536, 1696, 536, 588, 532, 1708, 532, 596, 528, 1720, 528, 1724, 532, 608, 524, 1700, 536, 7960, 532, 588, 536, 588, 532, 592, 532, 596, 528, 604, 528, 608, 536, 604, 532, 596, 532, 1700, 536, 588, 528, 600, 524, 604, 528, 604, 532, 604, 532, 604, 532, 596, 532, 588, 536, 584, 536, 592, 528, 600, 532, 604, 528, 608, 528, 612, 528, 600, 524, 592, 536, 588, 528, 596, 528, 600, 532, 604, 528, 608, 532, 604, 536, 592, 536, 584, 528, 592, 540, 588, 536, 1708, 528, 1720, 532, 1716, 536, 604, 528, 604, 532, 584, 532, 588, 532, 592, 528, 600, 524, 612, 528, 604, 532, 1724, 532, 596, 532, 1700, 532, 592, 524, 600, 532, 1712, 532, 1716, 532, 1716, 528, 1728, 536, 576, 556};
```
or
```
uint16_t rawbuf[800] = {8908, 4548, 536, 1700, 528, 1708, 532, 592, 532, 596, 524, 608, 532, 604, 528, 612, 520, 1724, 528, 588, 524, 1708, 532, 1708, 540, 588, 532, 600, 528, 608, 528, 608, 536, 596, 532, 1700, 532, 592, 536, 1704, 528, 596, 532, 604, 532, 604, 536, 600, 528, 604, 528, 588, 532, 1704, 536, 592, 540, 588, 532, 1716, 532, 1716, 544, 600, 528, 600, 536, 580, 536, 588, 528, 600, 524, 604, 524, 608, 528, 604, 528, 616, 528, 600, 532, 584, 536, 588, 532, 592, 528, 600, 532, 604, 532, 604, 532, 604, 528, 584, 532, 7952, 536, 1700, 528, 1708, 532, 1708, 532, 596, 532, 1712, 540, 596, 532, 608, 536, 1708, 528, 1704, 532, 588, 528, 1712, 532, 600, 524, 1724, 528, 608, 532, 604, 536, 592, 536, 584, 528, 596, 528, 596, 544, 584, 532, 600, 536, 600, 528, 612, 532, 596, 536, 584, 528, 592, 540, 588, 532, 596, 536, 600, 524, 612, 528, 608, 540, 592, 524, 592, 528, 596, 536, 588, 536, 592, 528, 600, 532, 604, 536, 604, 536, 592, 536, 584, 524, 596, 528, 600, 532, 596, 528, 604, 528, 608, 528, 612, 528, 600, 532, 588, 532, 588, 528, 600, 532, 596, 524, 608, 540, 596, 528, 612, 528, 596, 536, 1700, 528, 596, 532, 1708, 520, 608, 528, 1720, 528, 1720, 528, 612, 532, 1692, 532, 7960, 532, 588, 528, 596, 532, 592, 528, 600, 524, 612, 528, 604, 536, 604, 532, 596, 532, 1700, 528, 596, 524, 600, 532, 596, 528, 604, 536, 600, 532, 608, 528, 604, 528, 588, 532, 592, 536, 588, 524, 604, 532, 604, 532, 600, 528, 612, 528, 600, 528, 592, 528, 596, 528, 596, 528, 600, 532, 600, 528, 608, 532, 608, 528, 604, 528, 588, 528, 596, 532, 592, 524, 1720, 532, 604, 536, 1712, 532, 608, 536, 596, 532, 584, 532, 592, 532, 592, 528, 604, 528, 604, 532, 604, 532, 1720, 528, 604, 532, 1696, 528, 596, 532, 592, 528, 1716, 532, 600, 536, 1716, 532, 1724, 524, 588, 548, 13808, 8916, 4544, 528, 1704, 528, 1712, 532, 592, 532, 600, 524, 608, 532, 604, 532, 608, 532, 1712, 528, 592, 532, 1704, 532,  [... cut ...] 16, 524, 1720, 524, 592, 528, 1708, 528, 1712, 528, 600, 524, 612, 524, 612, 520, 616, 520, 608, 520, 1712, 516, 608, 524, 600, 536, 592, 528, 604, 528, 608, 532, 608, 528, 600, 532, 588, 528, 1704, 536, 592, 532, 596, 532, 1712, 532, 1720, 532, 604, 532, 596, 532, 584, 528, 596, 528, 596, 524, 604, 524, 608, 532, 604, 528, 608, 524, 604, 528, 592, 524, 596, 528, 596, 532, 600, 536, 596, 528, 608, 532, 604, 528, 584, 536, 7960, 528, 1708, 532, 1704, 532, 1712, 524, 604, 528, 1724, 532, 604};
```

crankyoldgit, 2017-11-25:
How did you go?

Also, have you tried a different frequency IR receiver module with this remote?
https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#basic-1

crankyoldgit, 2018-03-16:
Marking this closed due to lack of response. Feel free to re-open etc.

closed 2018-03-16 by crankyoldgit