crankyoldgit/IRremoteESP8266 #356: Problem with Panasonic repeat

glasruit, 2017-11-23:
Hi,

The repeat signal function for the Panasonic does not work. The single signals are all fine :-)
Might be the gap length between two pulses, which should be around 100 ms.

See issue #222

crankyoldgit, 2017-11-24:
Indeed, I had some incorrect math when I converted the IRP notation for the protocol.
If you change 300 to 378 in src/ir_Panasonic.cpp
e.g.
`#define PANASONIC_MIN_COMMAND_LENGTH_TICKS 378UL`

I believe that should do the trick.

This effectively changes the minimum gap between messages from 41040uSecs to 74736uSecs

Can you please try that out and confirm if it works?

glasruit, 2017-11-24:
Thanks, it works flawless now.

crankyoldgit, 2017-11-24:
Excellent. Thanks for confirming. PR #358 is now in progress. It should be in the next release (2.3.0) due out soon.

closed 2017-11-25 by crankyoldgit