crankyoldgit/IRremoteESP8266 #360: [REQ] Magiquest IR Wand

coolacid, 2017-11-24:
### Version/revison of the library used
Master

### Expected behavior
I've started a fork, and hoping to make this happen on my own, but just can't seem to get the timing down right. This protocol has been implemented a couple times in the Arduino lib version:
- http://openschemes.com/2013/02/27/mq-widget-pic12f508-controlled-magiquest-toy/
- https://github.com/kitlaan/Arduino-IRremote/blob/master/ir_Magiquest.cpp
- https://github.com/mpflaga/Arduino-IRremote

When trying to port the code, the wands will send very random results for Raw Length, and time lengths. 

### Actual behavior
Feature request and I just can't quite get it working. 

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
```
Encoding  : UNKNOWN
Code      : B392C681 (57 bits)
Timing[113]: 
   +   316, -   112,    +    14, -    14,    +   608, -   336,    +   810, -   298, 
   +   816, -   272,    +   832, -   280,    +   824, -   404,    +   678, -   398, 
   +   700, -   334,    +    12, -   770,    +   410, -   694,    +   708, -   458, 
   +   404, -   772,    +   582, -   594,    +   242, -   856,    +   650, -   504, 
   +   606, -   596,    +   260, -   894,    +   212, -   878,    +   224, -   890, 
   +   228, -   880,    +   202, -   910,    +   620, -   500,    +   626, -   580, 
   +   538, -   646,    +   586, -   540,    +   326, -   762,    +   622, -   616, 
   +   248, -   844,    +   248, -   898,    +   208, -   908,    +   188, -   914, 
   +   196, -   910,    +   200, -   888,    +   208, -   894,    +   350, -   770, 
   +   218, -   872,    +   344, -   768,    +   380, -   720,    +   362, -   748, 
   +   622, -   612,    +   248, -   882,    +   220, -   868,    +   236, -   894, 
   +   210, -   888,    +   204, -  2018,    +   210, -   898,    +   208, -   890, 
   +   636, -   500,    +   352, -   750,    +   648, -   568,    +   282, -   850, 
   +   514, -   672,    +   578, -   538,    +   232, -   872,    +   338, -   782, 
   +   324
uint16_t rawData[113] = {316, 112,  14, 14,  608, 336,  810, 298,  816, 272,  832, 280,  824, 404,  678, 398,  700, 334,  12, 770,  410, 694,  708, 458,  404, 772,  582, 594,  242, 856,  650, 504,  606, 596,  260, 894,  212, 878,  224, 890,  228, 880,  202, 910,  620, 500,  626, 580,  538, 646,  586, 540,  326, 762,  622, 616,  248, 844,  248, 898,  208, 908,  188, 914,  196, 910,  200, 888,  208, 894,  350, 770,  218, 872,  344, 768,  380, 720,  362, 748,  622, 612,  248, 882,  220, 868,  236, 894,  210, 888,  204, 2018,  210, 898,  208, 890,  636, 500,  352, 750,  648, 568,  282, 850,  514, 672,  578, 538,  232, 872,  338, 782,  324};  // UNKNOWN B392C681
```
```
Encoding  : UNKNOWN
Code      : 73F3D55 (55 bits)
Timing[109]: 
   +   266, -   848,    +   392, -   702,    +   290, -   810,    +   418, -   688, 
   +   438, -   682,    +   404, -   680,    +   458, -   684,    +   410, -   684, 
   +   412, -   770,    +   586, -   590,    +   254, -   838,    +   656, -   494, 
   +   408, -   692,    +   656, -   600,    +   520, -   636,    +   208, -   868, 
   +   378, -   738,    +   350, -   756,    +   376, -   718,    +   396, -   732, 
   +   608, -   578,    +   548, -   656,    +   586, -   548,    +   602, -   586, 
   +   284, -   798,    +   334, -  2000,    +   188, -   912,    +   192, -   926, 
   +   210, -   896,    +   208, -   902,    +   198, -   922,    +   228, -   860, 
   +   222, -   896,    +   292, -   808,    +   372, -   730,    +   342, -   760, 
   +   382, -   738,    +   630, -   542,    +   308, -   834,    +   258, -   866, 
   +   240, -   854,    +   258, -   876,    +   212, -   900,    +   204, -   884, 
   +   244, -   864,    +   618, -   526,    +   604, -   614,    +   246, -   872, 
   +   506, -   658,    +   214, -   884,    +   590, -   530,    +   352, -   784, 
   +   570, -   650,    +   502, -   638,    +   284
uint16_t rawData[109] = {266, 848,  392, 702,  290, 810,  418, 688,  438, 682,  404, 680,  458, 684,  410, 684,  412, 770,  586, 590,  254, 838,  656, 494,  408, 692,  656, 600,  520, 636,  208, 868,  378, 738,  350, 756,  376, 718,  396, 732,  608, 578,  548, 656,  586, 548,  602, 586,  284, 798,  334, 2000,  188, 912,  192, 926,  210, 896,  208, 902,  198, 922,  228, 860,  222, 896,  292, 808,  372, 730,  342, 760,  382, 738,  630, 542,  308, 834,  258, 866,  240, 854,  258, 876,  212, 900,  204, 884,  244, 864,  618, 526,  604, 614,  246, 872,  506, 658,  214, 884,  590, 530,  352, 784,  570, 650,  502, 638,  284};  // UNKNOWN 73F3D55
```
```
Encoding  : UNKNOWN
Code      : 3DD68F0A (57 bits)
Timing[113]: 
   +   416, -   688,    +   422, -   736,    +   344, -   784,    +   338, -   796, 
   +   292, -   816,    +   298, -   800,    +   284, -   818,    +   292, -    90, 
   +    12, -    12,    +   688, -   316,    +   778, -   732,    +   402, -   460, 
   +   706, -   640,    +   602, -   226,    +   872, -   588,    +   514, -   636, 
   +   576, -   290,    +   824, -   292,    +   810, -   302,    +   842, -   260, 
   +   864, -   220,    +   886, -   624,    +   494, -   634,    +   508, -   608, 
   +   648, -   548,    +   570, -   246,    +    12, -   864,    +   228, -  2020, 
   +   322, -   820,    +   290, -   808,    +   326, -   780,    +   326, -   814, 
   +   278, -   830,    +   304, -   848,    +   252, -   814,    +   280, -   852, 
   +   260, -   886,    +   198, -   896,    +   224, -   870,    +   616, -   518, 
   +   360, -   746,    +   364, -   764,    +   344, -   760,    +   334, -   772, 
   +   334, -   808,    +   302, -   810,    +   552, -   632,    +   214, -   886, 
   +   240, -   866,    +   364, -   736,    +   628, -   520,    +   332, -   784, 
   +   598, -   634,    +   190, -   922,    +   600, -   522,    +   330, -   770, 
   +   608
uint16_t rawData[113] = {416, 688,  422, 736,  344, 784,  338, 796,  292, 816,  298, 800,  284, 818,  292, 90,  12, 12,  688, 316,  778, 732,  402, 460,  706, 640,  602, 226,  872, 588,  514, 636,  576, 290,  824, 292,  810, 302,  842, 260,  864, 220,  886, 624,  494, 634,  508, 608,  648, 548,  570, 246,  12, 864,  228, 2020,  322, 820,  290, 808,  326, 780,  326, 814,  278, 830,  304, 848,  252, 814,  280, 852,  260, 886,  198, 896,  224, 870,  616, 518,  360, 746,  364, 764,  344, 760,  334, 772,  334, 808,  302, 810,  552, 632,  214, 886,  240, 866,  364, 736,  628, 520,  332, 784,  598, 634,  190, 922,  600, 522,  330, 770,  608};  // UNKNOWN 3DD68F0A
```

### Other useful information
I don't quite understand why the rawData seems to be inconsistent. I've tested the receiver with other known working remotes and was able to decode NEC and SAMSUNG messages. 

You can find my start of the hack on my fork.

crankyoldgit, 2017-11-25:
Reading the links you provided, it seems very likely the "wand" is transmitting at 36kHz, and if you have a typical IR receiver module, it is most-commonly a 38kHz one. This would mean you'd be borderline for getting a good signal. (possible, but not super reliable and far from optimal)

That _may_ be what is going on here. It seems to be seeing roughly the right number of pulses. However, there are some quite unexpected values. e.g. the ones ~12us, and the odd ~2000us ones.
Those are completely out-of-whack for how the protocol is defined by other example code and by-hand analysis in the article. Hence a decoder would skip the message.

If you've checked the "wand"s batteries are good, and it works with what ever it is supposed to work with, my _guess_ it is a reception problem. Be it something environmental (noise, too close/too far, etc etc) or more likely the IR module you are using just doesn't like it. If I recall correctly, NEC and Samsung both use 38kHz, so they'd be right down the middle of the acceptable freq. band for the module.

I'd suggest you invest the few dollars in a 36kHz IR de-modulator from ebay/aliexpress/local electronics shop etc and see if that works any better.

TL;DR: Even if we wrote a decode routine for it, with those rawdata capture values as a reference, they wouldn't detect the signal in it. You've got an 'electronics/hardware' reception problem in addition to (lack of) software one. 
https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#receiving-an-ir-signal

coolacid, 2017-11-25:
Yeah, let's belay this issue. I forgot to update this last night. I ended up going back to my "working" setup which is probably a few years old and couldn't get good results with hardware that was working at the time. I've ordered a bunch of new sensors already to see if I can get a good solid result in the old library first. 

I was reading the docs and it suggested the 38kHz should pickup enough of the signal, but given all the weird bits I was seeing with the modules I had, I wanted some new better ones anyway. 

Thanks for taking a look and will report back as soon as I get the new modules and test them.

crankyoldgit, 2017-11-25:
Cool, and good luck. In the meantime, I'll think about adding limited support for the device, but without good sample captures, I (and you) can't be sure it's going to work.

coolacid, 2017-11-28:
Good News Everyone!

So, yeah, even though a lot of the documentation said it would work with the wrong IR Receiver, getting the correct 36kHz one gave me super results. 

I'll close this ticket, clean up my code and submit a PR. 

Thanks for knocking me over the head and saying what I knew it was ;)

closed 2017-11-28 by coolacid

crankyoldgit, 2017-11-29:
Excellent. Thanks for getting back to us with the results. It's always great to get confirmation that it **isn't** a library issue for this sort of thing.