crankyoldgit/IRremoteESP8266 #362: IRrevrDumpv2 Problem/Serial Monitor Output doesnt work

MarleyRobert, 2017-11-25:
Hey there!

I have a problem with the output. I tried the IRrecvDumpV2 on Mac and Windows and every time I get a output like this.

[image]

I use a TSOP38238(tried 3 of them) and everything is connected as in the instruction. I also tried 3.3V and 5V, but it doesnt change anything. I used a second NodeMCU Board, but unfortunately I got the same result :/

crankyoldgit, 2017-11-25:
You're using the wrong baud rate. (9600 Baud, see the bottom right hand corner of your screenshot)
Use **115200** baud instead.

As described in the example code:
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L40

MarleyRobert, 2017-11-26:
Thank you for your quick answer!

closed 2017-11-26 by MarleyRobert