crankyoldgit/IRremoteESP8266 #372: Help Decoding WOFI Lighting Remote

drschlaumeier, 2017-12-10:
Hallo,
I have WOFI Lighting Remote for my Living Room. I use IRemoteESP8266 to record raw data and replay the data. It works perfect. However, its unknown protocol and I would like to decode.
Here are some raw data:

Light ON button:
uint16_t rawData[131] = {3534, 1620,  378, 464,  376, 1332,  376, 462,  378, 1332,  378, 464,  454, 1254,  454, 384,  378, 1334,  454, 386,  378, 462,  378, 464,  378, 1332,  454, 384,  458, 1254,  454, 1254,  454, 384,  456, 1256,  454, 382,  378, 464,  376, 462,  456, 1254,  376, 1334,  454, 1254,  456, 382,  378, 1332,  456, 384,  378, 1334,  376, 462,  376, 1336,  454, 1254,  456, 384,  456, 384,  378, 462,  378, 462,  378, 462,  378, 462,  458, 384,  438, 402,  376, 464,  378, 462,  378, 1332,  376, 462,  378, 1332,  454, 386,  378, 1334,  452, 1254,  456, 1254,  376, 464,  376, 1334,  454, 384,  456, 384,  378, 464,  376, 464,  454, 1256,  376, 462,  456, 382,  378, 1334,  454, 384,  378, 464,  376, 462,  378, 1330,  458, 382,  456, 1254,  456, 384,  456};
// UNKNOWN 28D702CA

Light OFF button:
uint16_t rawData[131] = {3506, 1652,  424, 420,  420, 1288,  422, 420,  420, 1286,  422, 420,  420, 1286,  422, 420,  372, 1336,  420, 420,  374, 468,  420, 420,  420, 1288,  422, 418,  422, 1286,  426, 1284,  422, 422,  422, 1282,  426, 418,  422, 416,  424, 416,  424, 1282,  426, 1282,  426, 1284,  426, 416,  424, 1282,  426, 418,  422, 1282,  426, 418,  422, 1284,  426, 1284,  424, 416,  424, 416,  424, 416,  424, 416,  422, 416,  424, 418,  422, 416,  422, 418,  422, 416,  424, 416,  424, 418,  422, 1284,  426, 418,  424, 416,  422, 418,  420, 420,  422, 418,  424, 418,  422, 1284,  426, 416,  422, 418,  422, 420,  422, 418,  422, 1284,  424, 418,  422, 420,  422, 1286,  422, 420,  422, 418,  420, 420,  420, 1288,  422, 1288,  422, 418,  372, 1336,  422};
// UNKNOWN 2F31FF69

I already use the tool "AutoAnalyseRawData.sh" and it decodes something but I cannot understand. I hope the experts can help ... e.g. HEX:
ON:  0x55168EAC00 AE 848 A
OFF: 0x55168EAC00 40 848 D
It seems that there are only 1 Byte in middle and the last 4 bit different. Could it be that last 4 bit is CRC?

Furthermore, is there any function to send the HEX data? I want to vary the bytes and see if some other or hidden functions e.g. 
sendHEX(0x55168EAC00AE848A, 64, 38) 

Many thanks in advance
DRS

some links:

> http://www.wofi.de/wofi/deckenleuchte-kiana-1flg-9540-01-06-0600.html?___store=english&___from_store=default

> https://www.bestlamp.pl/kiana-lampa-sufitowa.html

crankyoldgit, 2017-12-10:
Hey ya,
The `AutoAnalyseRawData.sh` tool is just to aid in trying to work out what the structure of a message might be, so that people can write a send/decode routine if they want. It's for quickly analysing raw data samples. There is no corresponding `sendBlah()` function for it.

You'll need to write your own.

In theory, for what the `AutoAnalyseRawData.sh` script has reported, that shouldn't be too hard.
If you take a look at https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Nikai.cpp for example, you should be able to use that as a template for this protocol.

[sendData()](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L214) is as close to your desired `sendHEX()` as you are going to get at the moment.
The reason a generic sendHEX() doesn't exist as it depends on the timings of each mark & space lengths and sending of a header etc and any trailing footer/gap between messages.

crankyoldgit, 2017-12-10:
As for the breakdown of the hex code, without a lot more data from other functions, it's not easy to tell.
The last 4 bits could be a checksum or similar. It might not be.
Assuming it is a complex protocol, you'll probably need to decode lots of different functions/buttons etc to see what bits change etc and try to work out a pattern. Many have found recording it in a spreadsheet or similar helpful to try to work it out. There is no magic bullet I'm afraid. It comes down to luck, intelligence, and a fair degree of work to reverse engineer some protocols.

drschlaumeier, 2017-12-10:
Hey, thanx for the example. It worked...I will make my own module ...
However, the timings are stange. I need to play a bit around with it to find the ticks multiplier :-) Unfortunately, I have no oscilloscope :-( 
The remote has only 10 button. I have code of all 10 buttons and its always the same schemata like in my first post.

Rgds & thanks for the help

```
 #define WOFIhdrmark 3500U //35530U
 #define WOFIhdrspace 1700U //1620U
 #define WOFIbitmark 378U //425U //460U
 #define WOFIonespace 1332 //1275U //1330U
 #define WOFIzerospace 462U //425U //460U

void sendWOFI(uint64_t data){
  irsend.enableIROut(38);
  //header
  irsend.mark(WOFIhdrmark);
  irsend.space(WOFIhdrspace);
  //data
  irsend.sendData(WOFIbitmark, WOFIonespace, WOFIbitmark, WOFIzerospace, data, 64, true);
  //footer
  irsend.mark(WOFIbitmark);
  delay(100);
}

```

crankyoldgit, 2017-12-11:
You don't need worry about using a tick multiplier, that's more an advance thing that can allow a decode routine to auto-sync with the IR signal. It's not required.

Once you get it all working and sorted, I look forward to the Pull Request. ;-)

drschlaumeier, 2017-12-11:
I found the origin manufacturer `http://www.dalen-tech.com/` of the lighting. Its sold under various brands. I also discovered that you can switch the channels but my lamp is accepting both. Interesting ...
Look at that if you can help and see te patterns or algorithms?
Thanks

0x55168EAC00 AE 848 A - ON - CH1
0x55168EAC00 6E 444 A - ON - CH2

0x55168EAC00 40 848 D - OFF - CH1
0x55168EAC00 40 448 1 - OFF - CH2

0x55168EAC00 A8 848 B - Night - CH1
0x55168EAC00 A8 448 7 - Night - CH2

0x55168EAC00 80 848 1 - Full - CH1
0x55168EAC00 80 448 D - Full - CH2

0x55168EAC00 98 848 8 - Bright - CH1
0x55168EAC00 98 448 4 - Bright - CH2

0x55168EAC00 84 848 5 - Dim - CH1
0x55168EAC00 84 448 9 - Dim - CH2

0x55168EAC00 A1 848 2 - Warm - CH1	
0x55168EAC00 A1 448 E - Warm - CH2

0x55168EAC00 A9 848 A - Cold - CH1
0x55168EAC00 A9 448 6 - Cold - CH2

0x55168EAC00 82 848 3 - Sleep - CH1
0x55168EAC00 42 444 3 - Sleep - CH2

0x55168EAC00 8C 848 E - GlareFree - CH1
0x55168EAC00 4C 444 E - GlareFree - CH2

crankyoldgit, 2017-12-12:
I've spent far to much time looking at it, and can sort of see a pattern, and every time I do, it turns out to be incorrect. :-(
I'd try looking at it in LSB first order, as the current 4th last digit (8 for Ch1 and 4 for Ch2) is probably in reverse order. i.e. in LSB order that digit would be 1 for Ch1 and 2 for Ch2.

drschlaumeier, 2017-12-12:
Interesting :-) Seems to be something new here.
Anyhow thanks for looking at it. I also thought about LSB but it makes no sence since 0x55 is quite often used to sync frequ between serial transmitter/receiver. So my assumtion is MSB....but I will look at LSB maybe its really different.

crankyoldgit, 2017-12-12:
The 0x55 (01010101) is very typical as you point out. That is definitely sent "first".
However, 0xAA is also "01010101" **IF** it is sent LSB first.

I tried a number combinations of XOR and simple sum of every 8 & 4 bits and some combos for the verification nibble (last four bits transmitted)

I noticed that **IF** there is a difference between CH1 and CH2 checksums, it's typically a difference of four between the two checksums. That kind of implies a simple sum, but XOR still might work too. But there are too many inconsistencies.
Given there are about 20 valid codes (so far), I'm not sure it's worth coding up a generic solution. That said, it is an enigma that I'd love to solve.

FYI, I'm kind-of implementing your suggested sendHEX() in PR #373

drschlaumeier, 2017-12-12:
Yeah, maybe chinese think different :-)
I tried LSB first and some mix of MSB/LSB bytes but w/o success:-(
So same for me, Its an enigma. Maybe if some people with same lighting reading this can help and send their codes. IF you look under distributors it seems that they sell there products all over the world with Dalen brand and many other bands. The remote control looks always "nearly" the same. Only brand name varies and my one has no CH1 / CH2 printed on it.

>   FYI, I'm kind-of implementing your suggested sendHEX() in PR #373
Great. Looking forward ...

Rgds

crankyoldgit, 2017-12-12:
I'm going to consider this closed from the libraries _technical support_ point of view, but leave it open for a month or so in case some checksum genius comes along and points out the _obvious_ solution.

Cheers.

drschlaumeier, 2017-12-12:
I opened the remote to have a look at the PCB. There is only one chip marked with "TTR01D 1711-NC". Fast search in internet showed this `http://www.tontek.com.tw/download4.asp?sn=58` ... I need to study but its far to late today :-)

crankyoldgit, 2017-12-12:
A quick skim didn't reveal anything fantastic, other than it's a 4-bit cpu, so the checksum is likely to be calculated using 4-bit/nibble values only. Which is what I was trying mostly.

drschlaumeier, 2017-12-15:
Hi, no news. I studied the documents and the sample .asm codes if there some hints...but nothing special.
The sample code operates in 8bit even if 4 bit cpu. Sample code is basic for NEC 3.64Mhz to demonstrate timing. Main func's are Keyscan, send LeadingHi & LeadingLow, send Data, send EndHi, EndingLow. There is no checksum code.
I also borrowed an oscilloscope to verify the signal is same as decoded by IRremoteESP8266. Its the same.
Hmm, I fear we need more valid codes to get an idea :-(

Rgds

crankyoldgit, 2017-12-15:
Probably.
It's the CH2 code that drive me mental. The last 3 digits (excluding checksum) are `448` except, occasionally it's `444` for no good reason that I can see. i.e. CH1 is always `848` ... Grrrrr.

drschlaumeier, 2017-12-15:
yes, ...and if 444 than the last nibble is same as for CH1...and the code in moddle is different ...very odd ...

leobel96, 2018-01-10:
Sorry for the intrusion but as last chance you can try this http://irdb.tk/capture/

crankyoldgit, 2018-01-10:
@leobel96 I converted one of the signals to that format.
e.g.
"+3506 -1652 +424 -420 +420 -1288 +422 -420 +420 -1286 +422 -420 +420 -1286 +422 -420 +372 -1336 +420 -420 +374 -468 +420 -420 +420 -1288 +422 -418 +422 -1286 +426 -1284 +422 -422 +422 -1282 +426 -418 +422 -416 +424 -416 +424 -1282 +426 -1282 +426 -1284 +426 -416 +424 -1282 +426 -418 +422 -1282 +426 -418 +422 -1284 +426 -1284 +424 -416 +424 -416 +424 -416 +424 -416 +422 -416 +424 -418 +422 -416 +422 -418 +422 -416 +424 -416 +424 -418 +422 -1284 +426 -418 +424 -416 +422 -418 +420 -420 +422 -418 +424 -418 +422 -1284 +426 -416 +422 -418 +422 -420 +422 -418 +422 -1284 +424 -418 +422 -420 +422 -1286 +422 -420 +422 -418 +420 -420 +420 -1288 +422 -1288 +422 -418 +372 -1336 +422 -40000" (I added the last '-40000' [an end gap] otherwise the page crashes with a 500.)

And it results in:
"Protocol Gap-426-1274-64?, device 170, subdevice 104, OBC 177, misc = AA.68.71.35.0.2.21.B1"

Which, from my initial review of it's output is a generic "I don't know what it is, but here are some of the obvious structures." i.e. similar to the auto analysis script I wrote. But, perhaps it might help.
Notably, it's gone for what appears to be an inverted (0->1, 1->0 etc) bit pattern but that's just cosmetic difference.

Hope it helps you @drschlaumeier

leobel96, 2018-01-10:
Uhm, I hoped it would be more useful... Anyway how have you converted the codes sent by drschlaumeier to the format required in the site I have linked? Where do the "+" and "-" signs come from?

crankyoldgit, 2018-01-10:
"+" are for Marks (i.e. LED On) and "-" for Spaces (i.e. LED Off)
The Raw format that drschlaumeier included is: Odd entries are Marks, and Even entries are Spaces.

Technically, this is what I did on a linux machine:
```
n=0
for i in 3506, 1652, 424, 420, 420, 1288, 422, 420, 420, 1286, 422, 420, 420, 1286, 422, 420, 372, 1336, 420, 420, 374, 468, 420, 420, 420, 1288, 422, 418, 422, 1286, 426, 1284, 422, 422, 422, 1282, 426, 418, 422, 416, 424, 416, 424, 1282, 426, 1282, 426, 1284, 426, 416, 424, 1282, 426, 418, 422, 1282, 426, 418, 422, 1284, 426, 1284, 424, 416, 424, 416, 424, 416, 424, 416, 422, 416, 424, 418, 422, 416, 422, 418, 422, 416, 424, 416, 424, 418, 422, 1284, 426, 418, 424, 416, 422, 418, 420, 420, 422, 418, 424, 418, 422, 1284, 426, 416, 422, 418, 422, 420, 422, 418, 422, 1284, 424, 418, 422, 420, 422, 1286, 422, 420, 422, 418, 420, 420, 420, 1288, 422, 1288, 422, 418, 372, 1336, 422
do
  n=$((n + 1))
  if [[ $((n % 2)) -eq 1 ]]  # Check if it is an odd position.
  then
    echo -n "+"
  else
    echo -n "-"
  fi
  echo -n "$i"
done | sed 's/,/ /g'
```

As stated before, I had to append ` -40000` to the end as that page/url expects the implicit trailing space/gap.

crankyoldgit, 2018-01-10:
@leobel96 It was a good suggestion. It seems to work for other protocols. It was totally worth trying.

leobel96, 2018-01-10:
@crankyoldgit I have understood now. Thank you for the explanation!!

leobel96, 2018-01-10:
Probably you already know it but I also find very useful for codes this great repository https://sourceforge.net/p/lirc-remotes/code/ci/master/tree/remotes/ that many times saved me from have to analyze all the buttons of my remote

crankyoldgit, 2018-03-16:
Per comment on 13 December, it's been a few months. I'm now going to mark this closed. Feel free to re-open if you have any progress,

closed 2018-03-16 by crankyoldgit