crankyoldgit/IRremoteESP8266 #380: How to use toggleRC5?

universam1, 2017-12-19:
### Version/revison of the library used
2.3.0

### Expected behavior
Trying to get RC5 working with toggle bit.

I searched the repository but can't find a hint how to use toggleRC5() function. I understand this should provide a proper way to make use of RC5 functions that do require to toggle the bit. 

Thank you!

universam1, 2017-12-19:
Guess I can answer it myself.
It is to be used in the calling application to bitwise XOR the data to toggle the 12bit and simulate another button push.
Feel free to close.

crankyoldgit, 2017-12-19:
Correct.

closed 2017-12-19 by crankyoldgit

arihantdaga, 2018-11-02:
@crankyoldgit Also can you help me understand what does (length = 13 or length = 12) means - since in RC5 we have 6 bit command and 5 bit address, then out of 13 bits and 12 bits what could be command and address.. ?
Thank you in advance..

crankyoldgit, 2018-11-02:
RC5 has several different lengths depending on which library implementation and version of RC5 you use.

Technically, the RC5 spec says there are 14 bits (`kRC5RawBits`) of information in RC5/RC-5X protocols
The original _IRremote_ library for Arduino, used a length(number of bits) of 12 (`kRC5Bits`). This library carries that tradition because it's meant to be compatible with it. IIRC, The difference is there are some "fixed" bits which don't change or are implied (e.g. button release) for RC5 hence they are not needed to be included in the data/message. They are just implied. In short, the lengths used are _historic_ and not easily connected to reality.

This library also implements the RC5-X protocol, which has a length of 13 (`kRC5XBits`)

`sendRC5()` will send a message as RC5 if the length is specified as less than 13, and as RC5-X if it >= 13.

I'm not going to sugar coat it, RC5 is a horrible protocol to understand with how IRremote/IRremoteESP8266 captures data. Just trust the code knows what it's doing.

If you use IRrecvDumpV2 it should decode the command & address correctly and report it for you.
The code for that decoding bit is https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_RC5_RC6.cpp#L421

See also the `encodeRC5()` and `encodeRC5X()`. In short, look at the code, and the included reference links in the code (`ir_RC5_RC5.cpp`) for how it works and why it does what it does.

crankyoldgit, 2018-11-02:
Also, I'm not sure if AnalysIR uses the same "decoding" as our library.

The code says:
```
//   Caller needs to take care of flipping the toggle bit.
//   That bit differentiates between key press & key release.
```
I'd recommend using the `toggleRC5()` function call to handle use of toggling.
see: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_RC5_RC6.cpp#L163

arihantdaga, 2018-11-03:
@crankyoldgit Thank you very much Indeed for helping. 
for toggleBit() command - I am sorry, I am still confused

`That bit differentiates between key press & key release`
In my case if ESP just recieve a code from a sever in mqtt message - lets say 0x145A.
Is this how i should call toggleRC5() function - 

```
irsend.sendRC5(code);
toggledCode = irsed.toggleRC5(code);
irsend.sendRC5(toggledCode); 

```

crankyoldgit, 2018-11-03:
Yes. I believe that should work. You may need to add a delay between the sends to simulate a more human response time though.