crankyoldgit/IRremoteESP8266 #382: Wrong identification?

aleonnet, 2017-12-22:
Hi,
I've perfectly compiled and tested IRMQTTSERVER on a Wemos d1 mini.
I tested it with Samsung TV Remote (protocol_num,hexcode  e.g. 7,E0E09966) and it works great!

I'm trying to use the same steps to control my A/C which was identified as a KELVINATOR by IRrecvDumpV2.ino as you can see below.

The problem is that when I use the recommended syntax for KELVINATOR A/C the Wemos no longer send IR codes.

*   ac_protocol_num,really_long_hexcode  e.g. 18,190B8050000000E0190B8070000010F0 or 18, 49876250072000E04987626000780731
*   Kelvinator (18) Air Con on, Low Fan, 25 deg etc.

I really can't figured out what I may be doing wrong. Is this a issue/bug?

### Version/revison of the library used
IRremoteESP8266, version: 2.3.0

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
Encoding  : KELVINATOR
Code      : 49876250072000E04987626000780731 (128 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 23C, Fan: 0 (AUTO), Turbo: Off, Quiet: Off, XFan: Off, IonFilter: On, Light: On, Swing (Horizontal): Off, Swing (Vertical): On
Library   : v2.3.0

Raw Timing[279]:
   +  9092, -  4412,    +   704, -  1604,    +   704, -   500,    +   730, -   472, 
   +   730, -  1608,    +   700, -   472,    +   708, -   498,    +   732, -  1580, 
   +   704, -   500,    +   730, -  1604,    +   678, -  1602,    +   704, -  1604, 
   +   726, -   472,    +   708, -   498,    +   732, -   472,    +   730, -   476, 
   +   732, -  1564,    +   718, -   498,    +   732, -  1582,    +   700, -   498, 
   +   710, -   498,    +   730, -   474,    +   708, -  1630,    +   700, -  1606, 
   +   700, -   472,    +   732, -   476,    +   732, -   472,    +   730, -   478, 
   +   704, -   498,    +   704, -  1608,    +   700, -   498,    +   730, -  1608, 
   +   700, -   472,    +   710, -   498,    +   732, -  1602,    +   704, -   472, 
   +   730, - 19906,    +   710, -  1628,    +   678, -  1590,    +   714, -  1606, 
   +   726, -   474,    +   708, -   498,    +   704, -   500,    +   730, -   476, 
   +   726, -   478,    +   704, -   498,    +   730, -   478,    +   730, -   472, 
   +   710, -   498,    +   704, -   498,    +   732, -  1608,    +   700, -   472, 
   +   708, -   500,    +   704, -   498,    +   730, -   478,    +   704, -   498, 
   +   732, -   472,    +   708, -   500,    +   704, -   502,    +   726, -   478, 
   +   730, -   472,    +   730, -   478,    +   730, -   472,    +   706, -   502, 
   +   704, -   500,    +   730, -   478,    +   704, -  1628,    +   674, -  1608, 
   +   700, -  1608,    +   700, - 39942,    +  9070, -  4412,    +   730, -  1608, 
   +   700, -   472,    +   708, -   500,    +   704, -  1628,    +   700, -   476, 
   +   732, -   472,    +   730, -  1582,    +   726, -   476,    +   704, -  1590, 
   +   740, -  1582,    +   726, -  1608,    +   674, -   498,    +   732, -   476, 
   +   730, -   474,    +   704, -   504,    +   726, -  1608,    +   700, -   456, 
   +   752, -  1604,    +   678, -   498,    +   730, -   478,    +   726, -   476, 
   +   730, -  1582,    +   726, -  1604,    +   704, -   472,    +   732, -   472, 
   +   708, -   498,    +   730, -   478,    +   726, -   476,    +   730, -   472, 
   +   710, -  1630,    +   700, -  1608,    +   678, -   498,    +   730, -   474, 
   +   730, -  1608,    +   700, -   478,    +   730, - 19910,    +   730, -   472, 
   +   730, -   478,    +   808, -   394,    +   730, -   478,    +   726, -   476, 
   +   730, -   472,    +   730, -   478,    +   704, -   498,    +   730, -   478, 
   +   726, -   478,    +   708, -   500,    +   730, -  1564,    +   744, -  1604, 
   +   700, -  1568,    +   740, -  1608,    +   700, -   478,    +   704, -  1630, 
   +   704, -  1602,    +   700, -  1568,    +   740, -   476,    +   730, -   472, 
   +   732, -   476,    +   726, -   476,    +   704, -   504,    +   726, -  1608, 
   +   700, -   478,    +   726, -   478,    +   730, -   472,    +   730, -  1568, 
   +   740, -  1608,    +   700, -   478,    +   730, -   472,    +   730

uint16_t rawData[279] = {9092, 4412,  704, 1604,  704, 500,  730, 472,  730, 1608,  700, 472,  708, 498,  732, 1580,  704, 500,  730, 1604,  678, 1602,  704, 1604,  726, 472,  708, 498,  732, 472,  730, 476,  732, 1564,  718, 498,  732, 1582,  700, 498,  710, 498,  730, 474,  708, 1630,  700, 1606,  700, 472,  732, 476,  732, 472,  730, 478,  704, 498,  704, 1608,  700, 498,  730, 1608,  700, 472,  710, 498,  732, 1602,  704, 472,  730, 19906,  710, 1628,  678, 1590,  714, 1606,  726, 474,  708, 498,  704, 500,  730, 476,  726, 478,  704, 498,  730, 478,  730, 472,  710, 498,  704, 498,  732, 1608,  700, 472,  708, 500,  704, 498,  730, 478,  704, 498,  732, 472,  708, 500,  704, 502,  726, 478,  730, 472,  730, 478,  730, 472,  706, 502,  704, 500,  730, 478,  704, 1628,  674, 1608,  700, 1608,  700, 39942,  9070, 4412,  730, 1608,  700, 472,  708, 500,  704, 1628,  700, 476,  732, 472,  730, 1582,  726, 476,  704, 1590,  740, 1582,  726, 1608,  674, 498,  732, 476,  730, 474,  704, 504,  726, 1608,  700, 456,  752, 1604,  678, 498,  730, 478,  726, 476,  730, 1582,  726, 1604,  704, 472,  732, 472,  708, 498,  730, 478,  726, 476,  730, 472,  710, 1630,  700, 1608,  678, 498,  730, 474,  730, 1608,  700, 478,  730, 19910,  730, 472,  730, 478,  808, 394,  730, 478,  726, 476,  730, 472,  730, 478,  704, 498,  730, 478,  726, 478,  708, 500,  730, 1564,  744, 1604,  700, 1568,  740, 1608,  700, 478,  704, 1630,  704, 1602,  700, 1568,  740, 476,  730, 472,  732, 476,  726, 476,  704, 504,  726, 1608,  700, 478,  726, 478,  730, 472,  730, 1568,  740, 1608,  700, 478,  730, 472,  730};  // KELVINATOR
uint8_t state[16] = {0x49, 0x87, 0x62, 0x50, 0x07, 0x20, 0x00, 0xE0, 0x49, 0x87, 0x62, 0x60, 0x00, 0x78, 0x07, 0x31};

crankyoldgit, 2017-12-22:
Hey @aleonnet 
I think I may know the problem, but only **if** you were tracking the _master_ branch (i.e. via a `git clone`) rather than the "official" [2.3.0 release](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.3.0).

i.e. _master_ branch had the bug, but the 2.3.0 release does not.

The bug (#381 ) manifests itself almost exactly as you described in the IRMQTTServer program. eg. You could send some codes fine, then sending a particular protocol will cause the device to appear to have locked up, and not respond to anything. What you described is almost _exactly_ how I found the bug.

In short, the solution for you should be a simple `git pull` and recompile/upload etc. You'll probably need to power off & then on the wemos D1 and NOT send any signals before uploading the firmware via http. (or just re-upload it via serial)

However, if you obtained the library via the Arduino Library Manager, and it really is running pure 2.3.0, then I'm not sure what it is and I need more detail on what you are doing etc.

The _master_ branch had the bug for about 3 days, until about 4 hours ago. If it still is a problem after you re-sync, please let me know asap.

aleonnet, 2017-12-22:
Hi Mark,
The bug is exactly the same.
I tried the 2.3.0 release and it worked nicely!
Thank you.

crankyoldgit, 2017-12-29:
FYI, v2.3.1 has been released and it includes the fix for the bug you encountered.

Marking this issue closed.

closed 2017-12-29 by crankyoldgit