crankyoldgit/IRremoteESP8266 #386: Different raw values when decoding with IRrecvDumpV2

paquechu, 2018-01-02:
Hello,
I'm a bit baffled trying to decode IR signals from an EKOKAI air conditioner using the IRrecvDumpV2 sketch (version 0.3 of November 2017, IRremoteESP8266 v2.3.0)
By pressing the same button on the remote control (always in the same conditions) I get a list of different values; However, the value of the Code field is always the same
Is this normal or am I doing something wrong?

On the other hand, I do not know if pressing any button on the remote control only sends the information of the button pressed or a complete combination of the configuration that is established at that moment on the remote control.

I include an example of what I am commenting: In this case I press the button to lower temperature from 27 to 26 ºC.

First capture:
Timestamp : 004328.544
Encoding  : UNKNOWN
Code      : 380F0D08 (70 bits)
Library   : v2.3.0

Raw Timing[139]:
   +  9008, -  4496,    +   644, -  1660,    +   676, -   530,    +   648, -   558, 
   +   672, -  1636,    +   646, -  1660,    +   644, -   556,    +   650, -   584, 
   +   626, -   560,    +   644, -   580,    +   628, -  1680,    +   624, -   560, 
   +   648, -  1662,    +   644, -   582,    +   648, -   536,    +   674, -   530, 
   +   646, -   580,    +   628, -   560,    +   670, -   532,    +   646, -   562, 
   +   644, -   556,    +   672, -   536,    +   648, -  1662,    +   646, -  1660, 
   +   652, -   554,    +   644, -   558,    +   672, -   538,    +   644, -   560, 
   +   668, -   560,    +   648, -  1638,    +   668, -   536,    +   644, -  1660, 
   +   668, -   532,    +   648, -   560,    +   648, -  1660,    +   674, -   554, 
   +   622, - 19990,    +   646, -   580,    +   624, -  1660,    +   648, -   556, 
   +   648, -   558,    +   674, -   556,    +   622, -   560,    +   644, -   564, 
   +   668, -   536,    +   646, -  1662,    +   646, -  1658,    +   672, -   534, 
   +   648, -   558,    +   644, -   562,    +   648, -  1662,    +   644, -   584, 
   +   622, -   558,    +   648, -   562,    +   668, -   534,    +   670, -   536, 
   +   670, -   532,    +   672, -   536,    +   646, -   560,    +   646, -   558, 
   +   648, -   558,    +   670, -   534,    +   650, -   558,    +   646, -   560, 
   +   646, -   560,    +   668, -  1638,    +   646, -  1662,    +   646, -  1660, 
   +   646, -  1660,    +   648

uint16_t rawData[139] = {9008, 4496,  644, 1660,  676, 530,  648, 558,  672, 1636,  646, 1660,  644, 556,  650, 584,  626, 560,  644, 580,  628, 1680,  624, 560,  648, 1662,  644, 582,  648, 536,  674, 530,  646, 580,  628, 560,  670, 532,  646, 562,  644, 556,  672, 536,  648, 1662,  646, 1660,  652, 554,  644, 558,  672, 538,  644, 560,  668, 560,  648, 1638,  668, 536,  644, 1660,  668, 532,  648, 560,  648, 1660,  674, 554,  622, 19990,  646, 580,  624, 1660,  648, 556,  648, 558,  674, 556,  622, 560,  644, 564,  668, 536,  646, 1662,  646, 1658,  672, 534,  648, 558,  644, 562,  648, 1662,  644, 584,  622, 558,  648, 562,  668, 534,  670, 536,  670, 532,  672, 536,  646, 560,  646, 558,  648, 558,  670, 534,  650, 558,  646, 560,  646, 560,  668, 1638,  646, 1662,  646, 1660,  646, 1660,  648};  // UNKNOWN 380F0D08

Second capture:
Timestamp : 004354.604
Encoding  : UNKNOWN
Code      : 380F0D08 (70 bits)
Library   : v2.3.0

Raw Timing[139]:
   +  9010, -  4468,    +   674, -  1632,    +   676, -   526,    +   678, -   532, 
   +   674, -  1632,    +   674, -  1630,    +   674, -   534,    +   678, -   528, 
   +   676, -   532,    +   674, -   528,    +   678, -  1632,    +   676, -   530, 
   +   676, -  1630,    +   676, -   532,    +   672, -   530,    +   676, -   528, 
   +   674, -   534,    +   672, -   530,    +   676, -   532,    +   650, -   554, 
   +   678, -   528,    +   678, -   530,    +   672, -  1632,    +   676, -  1632, 
   +   678, -   532,    +   672, -   528,    +   674, -   534,    +   670, -   536, 
   +   674, -   530,    +   650, -  1658,    +   678, -   528,    +   674, -  1632, 
   +   676, -   532,    +   674, -   532,    +   674, -  1632,    +   652, -   552, 
   +   676, - 19964,    +   674, -   530,    +   676, -  1632,    +   680, -   526, 
   +   676, -   528,    +   672, -   530,    +   652, -   556,    +   674, -   528, 
   +   678, -   528,    +   650, -  1656,    +   674, -  1632,    +   652, -   556, 
   +   648, -   552,    +   654, -   554,    +   680, -  1626,    +   672, -   532, 
   +   652, -   554,    +   680, -   524,    +   650, -   554,    +   676, -   532, 
   +   648, -   558,    +   672, -   532,    +   676, -   530,    +   654, -   548, 
   +   676, -   528,    +   680, -   528,    +   676, -   532,    +   674, -   530, 
   +   674, -   528,    +   652, -  1658,    +   680, -  1626,    +   674, -  1634, 
   +   674, -  1630,    +   678

uint16_t rawData[139] = {9010, 4468,  674, 1632,  676, 526,  678, 532,  674, 1632,  674, 1630,  674, 534,  678, 528,  676, 532,  674, 528,  678, 1632,  676, 530,  676, 1630,  676, 532,  672, 530,  676, 528,  674, 534,  672, 530,  676, 532,  650, 554,  678, 528,  678, 530,  672, 1632,  676, 1632,  678, 532,  672, 528,  674, 534,  670, 536,  674, 530,  650, 1658,  678, 528,  674, 1632,  676, 532,  674, 532,  674, 1632,  652, 552,  676, 19964,  674, 530,  676, 1632,  680, 526,  676, 528,  672, 530,  652, 556,  674, 528,  678, 528,  650, 1656,  674, 1632,  652, 556,  648, 552,  654, 554,  680, 1626,  672, 532,  652, 554,  680, 524,  650, 554,  676, 532,  648, 558,  672, 532,  676, 530,  654, 548,  676, 528,  680, 528,  676, 532,  674, 530,  674, 528,  652, 1658,  680, 1626,  674, 1634,  674, 1630,  678};  // UNKNOWN 380F0D08

third capture:
Timestamp : 004472.809
Encoding  : UNKNOWN
Code      : 151E56F4 (76 bits)
Library   : v2.3.0

Raw Timing[151]:
   +    90, - 19168,    +   134, - 19126,    +   134, -  2270,    +  9020, -  4464, 
   +   680, -  1632,    +   674, -   372,    +   830, -   526,    +   676, -  1634, 
   +   676, -  1630,    +   678, -   528,    +   680, -   526,    +   672, -   528, 
   +   678, -   530,    +   676, -  1630,    +   678, -   528,    +   682, -  1632, 
   +   648, -   554,    +   674, -   528,    +   674, -   528,    +   680, -   528, 
   +   656, -   552,    +   680, -   522,    +   678, -   528,    +   678, -   528, 
   +   676, -   526,    +   674, -  1632,    +   672, -  1634,    +   678, -   528, 
   +   682, -   530,    +   678, -   524,    +   674, -   526,    +   678, -   530, 
   +   680, -  1628,    +   650, -   556,    +   682, -  1630,    +   676, -   526, 
   +   678, -   524,    +   680, -  1632,    +   678, -   526,    +   678, -  7240, 
   +   132, - 12568,    +   648, -   554,    +   674, -  1630,    +   678, -   532, 
   +   678, -   530,    +   754, -   448,    +   650, -   554,    +   680, -   530, 
   +   674, -   528,    +   672, -  1636,    +   676, -  1628,    +   680, -   530, 
   +   678, -   532,    +   678, -   530,    +   672, -  1630,    +   676, -   534, 
   +   678, -   530,    +   652, -   554,    +   674, -   166,    +  1042, -   532, 
   +   674, -   528,    +   674, -   530,    +   678, -   532,    +   650, -   556, 
   +   672, -   528,    +   674, -   530,    +   678, -   532,    +   678, -   528, 
   +   674, -   532,    +   672, -  1632,    +   674, -  1630,    +   676, -  1636, 
   +   674, -  1626,    +   680, - 16256,    +   134, - 19132,    +   130

uint16_t rawData[151] = {90, 19168,  134, 19126,  134, 2270,  9020, 4464,  680, 1632,  674, 372,  830, 526,  676, 1634,  676, 1630,  678, 528,  680, 526,  672, 528,  678, 530,  676, 1630,  678, 528,  682, 1632,  648, 554,  674, 528,  674, 528,  680, 528,  656, 552,  680, 522,  678, 528,  678, 528,  676, 526,  674, 1632,  672, 1634,  678, 528,  682, 530,  678, 524,  674, 526,  678, 530,  680, 1628,  650, 556,  682, 1630,  676, 526,  678, 524,  680, 1632,  678, 526,  678, 7240,  132, 12568,  648, 554,  674, 1630,  678, 532,  678, 530,  754, 448,  650, 554,  680, 530,  674, 528,  672, 1636,  676, 1628,  680, 530,  678, 532,  678, 530,  672, 1630,  676, 534,  678, 530,  652, 554,  674, 166,  1042, 532,  674, 528,  674, 530,  678, 532,  650, 556,  672, 528,  674, 530,  678, 532,  678, 528,  674, 532,  672, 1632,  674, 1630,  676, 1636,  674, 1626,  680, 16256,  134, 19132,  130};  // UNKNOWN 151E56F4

crankyoldgit, 2018-01-03:
With respect to the UNKNOWN value, have a read of this: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code
In short, the "Code" for UNKNOWNs is only good for seeing if it is roughly the same as another UNKNOWN message. **Don't** rely on the "Code" in UNKNOWN messages for **anything**.

Of the three captures you provided, the first two are indeed effectively identical.
The last one looks like it includes some random noise i.e. a bad capture. I'd ignore that signal for now. It looks like the Start/Header Mark for your messages is approx **9000** ms. If it doesn't start with that, you've probably collected some garbage in the capture.

Unless you get some other consistent capture size, I'd also bank on the rawData size of **139** being a _good_ capture.

As you surmised, it is typical for A/C units to send their entire state (i.e. all the settings) in each message.
e.g. If at the end of your temp lowering button press the remote reads "26C" then the message will be every setting on the remote, including the new temp setting of "26C".

If you press it again, and it reads "25C" i'd expect a slightly different message.

FYI, you can run the rawData line through the https://github.com/markszabo/IRremoteESP8266/blob/master/tools/AutoAnalyseRawData.sh program, and it will give you a rough idea of what the internal data off the message is.

e.g. For your first capture the output is:
```
Potential Mark Candidates (using a range of 200):
9008
676

Potential Space Candidates (using a range of 200):
19990
4496
1680
584

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
HDR_MARK = 9008
BIT_MARK = 676
SPACE_GAP1 = 19990
HDR_SPACE = 4496
ONE_SPACE = 1680
ZERO_SPACE = 584

Decoding protocol based on analysis so far:

HDR_MARK+HDR_SPACE+10011000010100000000011000001010010 GAP(19990)
  Bits: 35
  Hex:  0x4C2803052 (MSB first)
        0x250600A19 (LSB first)
  Dec:  20443050066 (MSB first)
        9938405913 (LSB first)
  Bin:  10011000010100000000011000001010010 (MSB first)
        01001010000011000000000101000011001 (LSB first)
BIT_MARK(UNEXPECTED)01000000110001000000000000001111
  Bits: 32
  Hex:  0x40C4000F (MSB first)
        0xF0002302 (LSB first)
  Dec:  1086586895 (MSB first)
        4026540802 (LSB first)
  Bin:  01000000110001000000000000001111 (MSB first)
        11110000000000000010001100000010 (LSB first)
```

A/C codes _typically_ come in _multiplies of 8_ bit sizes. I'd wager the 35 bits of the first block is 32 bits of data plus a standard binary marker of "010" added at the end. I've seen this a lot in a number of A/C manufacturer codes.

P.S. You are also a minor revision of the library behind, but that won't affect anything here. ;-)

paquechu, 2018-01-04:
hello, thank you very much for your quick reply :-)
With respect to the UNKNOWN value ... I take note.

<< Of the three captures you provided, the first two are indeed effectively identical.

I do not understand this, the first two captures have the same code (380F0D08) but the RawData values ​​are different and this last is what I do not understand. After doing several more tests I have seen that although the RawData values ​​(for the same function of the remote control) in different captures although they have different values, the command is executed correctly.

<< Unless you get some other consistent capture size, I would also bank on the rawData size of 139 being a good capture.

More than 90% of the catches I have made have size 139.

<< As you surmised, it is typical for A / C units to send their entire state (i.e. all the settings) in each message.

It is understood. This complicates the programming a lot since it forces me to make many captures and store them in arrays. The tests that I've been doing I think are causing memory problems. I do not know if there is another way to solve this ....

On the other hand I tried to run the AutoAnalyseRawData.sh tool on my Mac, but I guess there must be some incompatibility timpa since I get errors. I do it in the following way:

I create a file with a line containing the captured RawData (the whole line) and then:

.\AutoAnalyseRawData.sh < archivo.txt

Thank you very much again

Regards :-)

paquechu, 2018-01-04:
I forgot to comment that in the sketch that I am testing I have a four-dimensional array to store all the IR commands but when I try to call the function irsend.sendraw I have compilation errors.

I do not know if it is possible to use this function with a variable of this type.

irsend.sendRaw (RawData [0] [0] [0], 139.38);

I have tried several ways but it does not work.

crankyoldgit, 2018-01-05:
An example of how to invoke `sendRaw()` can be found in https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino
e.g. Lines 39 & 65.

From https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.h#L37
the prototype for `sendRaw()` is `void sendRaw(uint16_t buf[], uint16_t len, uint16_t hz);`

As long as the element of your "four-dimensional array" that you pass to `sendRaw()` is an array (or pointer ) of uint16's (unsigned int) then you should be fine.
If you are getting compilation errors, it is most likely you are calling it with the wrong arguments or argument types.

Sorry, I don't have a Mac to test on anymore, so I can't check the Analyse script on it.
Even though the timing values on the 1st and 2nd samples slightly differ, they are roughly the same values(+/- 20%) in the exact same sequence. Hence they decode to the same "codes". The Analyse script would show you that if it worked for you.
I'll state it again, that 3rd capture is effectively garbage. Re-capture that message till you get one that is of the right size and starting values etc.

Yes, you are correct A/C messages are very complex and huge. A **LOT** of effort goes into decoding them to make them more manageable if you want to expose the entire functionality of the device. i.e. Every mode and every option of every mode etc.

That said, you probably don't need to store _every_ single possible message as a _raw_ message. Most people really only use a few messages for basic home automation. e.g. Off, On + Cool mode at a preset temp/fan/etc, On + Heat mode at a preset temp/fan/etc, On + fan mode, ...

paquechu, 2018-01-11:
[Decodificacion_IR2.zip](https://github.com/markszabo/IRremoteESP8266/files/1623877/Decodificacion_IR2.zip)
Hello again
During this time I have been investigating closely about the infrared signals and the protocols on which they are implemented and I already understand those differences that I appreciated and that made me post here.
I have also managed to run the AutoAnalyseRawData.sh script on another linux machine and it has been very helpful.
The problem I had to execute sendRaw was because of too much memory in the array, not syntax. For this reason I am trying to program the protocol, because I know better how it works, however I am stuck in the calculation part of the last byte that is sent in the burst that I understand is a kind of checksum that I am not able to calculate.
I do not want to abuse your time, but if you want, here I put the important data about what I discovered along with real data captured from the remote in an excel file in case you can help me at this point.

The first sheet contains the data captured from the remote control in decimal values ​​and below it is a map with the calculated structure that is sent in the burst. With the estimate I have made thanks to the AutoAnalyseRawData script info it seems to work correctly according to the tests I have done.

The second sheet 

contains an outline of this map where the binary codes are represented and the function that each of them has.

Thank you very much
Regards

crankyoldgit, 2018-01-11:
After a quick glance at the spreadsheet (Tip: It would be good to put it into Google Docs/Sheets so that if you change it, people can see it immediately, and don't need Excel to view it etc.), I'm convinced that the data is stored in [LSB first](https://en.wikipedia.org/wiki/Most_significant_bit#Most_Significant_Bit_First_vs_Least_Significant_Bit_First) order.
This is reasoned from the Temperature values being a simple incremental sequence when viewed in the reverse order in your spreadsheet, and ditto with the fan speed.

This is quite common with aircon remotes. e.g. Kelvinator transmits in LSB first order.

That may significantly help with calculating the checksum value.

paquechu, 2018-01-14:
Hello,
Thanks for the indications, I have been looking at the Kelvinator code but I am not able to find the solution :-(
As you indicated, here is a link to the spreadsheet file that I have filled out with more information, in case someone can use it:

https://docs.google.com/spreadsheets/d/1rzEABs_qillXmTapB4AeMjLu-hMK8AmjZFZN2X_3ejw/edit?usp=sharing

I do not give up, but I do not want to have this query open for a long time. I will continue investigating and if I find the solution I will reopen the query.

Regards ;-)

crankyoldgit, 2018-01-14:
When I had a look at your data, the checksum seemed to be only calculated from the first four(4) transmitted bits of each byte. Or the lower four least significant bits of each byte if you treat them as being transmitted in LSB first order. That is, the other half of each byte doesn't affect the checksum. i.e. When the fan speed changes. The checksum also increments in step with the temperature value too. That leads me to think it's a simple addition (mod 16) or an XOR of the lower half of each byte.

I was not expecting you to find a "solution" in the kelvinator code, merely that is a protocol that transmits in LSB first order. i.e. The `sendKelvinator()` routine sends in LSB order **and** has the same "010" marker at the end of the first message chunk.

crankyoldgit, 2018-01-14:
Interesting, in your latest data/spreadsheet, it appears the checksum doesn't change when the "Swing Vertical" data changes. That's best highlighted in the "Other Probes" section. So it's probably only a checksum of the first block the message. i.e. only the first half (transmitted) of the first four bytes.

paquechu, 2018-01-14:
That's right, I do not think the fan speed will intervene. Only the first four bytes (operation mode and on / off bit) since in all operating modes, regardless of the value of the rest of the bits, the checksum result is always the same (taking into account that only the temperature varies) . But in the final part of the sheet you can see that when the bit of on / of is 0 then the checksum no longer meets that pattern and it is different then I think that only the first four bytes intervene, but as I said I am not able to do the calculation.
I will continue investigating and if I find out the "key" I will post it.
Regards

crankyoldgit, 2018-01-14:
Have you written the code to transmit or decode the messages yet?

paquechu, 2018-01-14:
[EkokaiTest.zip](https://github.com/markszabo/IRremoteESP8266/files/1629835/EkokaiTest.zip)
I have tried to develop it according to IRremoteESP8266 libraries but it is more difficult for me. I tried to adapt the code to the HeatpumpIR libraries and I think I could adapt them because the tests I have done are working for me. What I have done is create several arrays with the checksum that corresponds in each case of operation mode until I can solve the problem of the calculation of the checksum ...
I include them here.

crankyoldgit, 2018-01-18:
@paquechu I was reminded by this on another issue/PR .. The checksum algorithm listed here looks like it might be what you need based on what I remember.

https://github.com/ToniA/arduino-heatpumpir/blob/master/GreeHeatpumpIR.cpp#L156

crankyoldgit, 2018-01-18:
Yes, it works, it's also the checksum algorithm used in Kelvinator.
The data is in LSB first order for each byte. Sum the lower 4 bits of the first 4 transmitted bytes, then add the sum of the upper 4 bits of the next 3 bytes, then add 10. Mod 16 and tada you have the checksum. Worked for about 6 very different values on your spreadsheet.

paquechu, 2018-01-18:
Wow, very good news, I'll try it as soon as possible and comment on the result.
Thank you very much for your help :-)
Regards

paquechu, 2018-01-21:
Hi
[Ekokai.zip](https://github.com/markszabo/IRremoteESP8266/files/1650129/Ekokai.zip)
,
I have updated the spreadsheet, including a new sheet with the calculation of all the checksums based on your indications and it works in all cases :-).

I have also adapted the code that I put in another of my messages with the implementation of the checksum. I have verified that it works, although I had to make a small modification. Finally the result of the checksum must be in the high positions of the byte. (EkokaiTemplate [7] = EkokaiTemplate [7] << 4)

I have compiled a project to do several tests and until now all have worked well for me. I will continue doing more tests just in case there is something that has been forgotten.

I leave the code in this message in case it is helpful.

Thanks for the mention :-)

crankyoldgit, 2018-01-25:
@paquechu That latest Ekokai.zip file doesn't seem to unzip properly for me. I can list the contents, but not actually extract the two files. As they are tiny. Perhaps just upload files individually?

paquechu, 2018-01-25:
Hello, I have updated the post including another copy in zip format, I hope this time it works :-)
Regards

crankyoldgit, 2018-01-26:
Yes. That worked much better.

crankyoldgit, 2018-01-29:
Support for decoding Gree/Ekokai has now been merged into the _master_ branch. It will be in the next official release. Turns out Gree and Ekokai use the same protocol.

Marking this issue closed as I think we are done here now.

closed 2018-01-29 by crankyoldgit

crankyoldgit, 2018-03-02:
This is now live in the v2.3.3 release

hemonu, 2018-04-21:
FYI: I have two daitsu AC units with a remote control type YK1F in my house in spain, for which I reverse engineered the codes. It looks like they are identical to these ones. The timings I measured are not exactly the same, but it should be in tolerance. From a measurement with my oscilloscope I'm quite sure, that the time is as in NEC (9000/4500 for header, 562.5 for MARK, 562,5 / 1687,5 for SPACE). I will test and verify, when I'm in spain next time, most  likely in May. 
I have another daitsu unit with totally different codes, so it is not for all the brand!